sipeed/NanoKVM #72: Reset actually triggers power on and power on does nothing after latest firmware update 2.0.9

sadnesskant, 2024-09-27:
I checked and rechecked the pinouts of both my motherboard and the KVM-B board and I'm pretty sure it is correct 
[image]
However I can only power on the PC by selecting Reset and if I select Power on nothing happens, regardless if the PC is running or not.

Z2Z-GuGu, 2024-10-14:
@sadnesskant Hi, Can you take a photo to check the wiring between KVM-B and the motherboard? It seems that there is still an issue with the PWR and RST being connected incorrectly. Also, is the power light status displayed correctly?

ScottESanDiego, 2024-11-16:
I have the same issue.  ASUS ProArt X870E motherboard.  "Reset" does a short press of power.  "Power" seems like it does nothing.

sadnesskant, 2024-11-17:
> I have the same issue. ASUS ProArt X870E motherboard. "Reset" does a short press of power. "Power" seems like it does nothing.

I used the dupont jumper wires to connect it to the front panel header and it works as it suppose to, which is quite weird, don't understand why they would make any difference.

ScottESanDiego, 2024-11-20:
> I used the dupont jumper wires to connect it to the front panel header and it works as it suppose to, which is quite weird, don't understand why they would make any difference.

I'm hesitant to do that since I don't have a PCIe bracket to stick that to.  Will see what I can rig up to test though. :-) Thanks for the feedback!

mrdanielm, 2025-01-25:
I get the same issue with the ASUS Prime z790m-plus d4