sipeed/NanoKVM #73: Support for multiport kvm over ip  ezcoo

yodatak, 2024-09-27:
https://docs.pikvm.org/ezcoo/#connections

i was using my pikvm but it die now i use nanokvm but i cannot switch source with nanokvm can the usb support source change for ezcoo kvm can be enabled ?

with 
/dev/ttyUSB0
more protocol information here 
https://docs.pikvm.org/ezcoo/#developer-info

some examples 
https://docs.pikvm.org/wiring_examples/#hdmi-csi

thanks !

Z2Z-GuGu, 2024-10-14:
NanoKVM only has one USB port, so KVM switches like ezcoo kvm that use USB management switching function may not be directly compatible. NanoKVM Full provides two external serial ports, which may be able to switch between serial ports and KVM through additional modifications

DjSni, 2024-11-09:
Unfortunately, the hotkey function does not work either.
It works by pressing the Ctrl key twice and then the port number (in my case 1-4), e.g.: Ctrl, Ctrl, 2.

I have updated to the current version 2.1.0.

Simon1511, 2024-11-24:
Unfortunately not working with the latest release either.

richy-66, 2025-05-26:
i found out that in the lycheerv-nano repo there is a package for xdotool so this would be a possibility to at least send the ctrl+ctrl+1234 to the ezcoo from the ssh console. 

problem is i am not able to install any package on the nanokvm nor do i find a binary of xdotool

anyone and idea?

the only thing i was able to install xdotool via pip install - but i have no pyhton skills... :-(