sipeed/NanoKVM #75: support@sipeed.com does not exist, unable to contact support for delivery status.

saildot4k, 2024-09-29:


Zepan, 2024-09-30:
hi, the mail will reply in 2 work day (10.1~10.7 is national holiday), what is your email address?

saildot4k, 2024-09-30:
I emailed weeks ago but again email bounced back. As expected from a fly by
night Chinese company. Paypal just refunded me today due to non
communication on your part.

patschi, 2024-09-30:
I sent them an email to exactly this address on 4th of September and received a helpful reply just 11 hours later. Can't complain. Looks like it's a problem on your end or from your mail provider then.

You could have used GitHub issue, their (granted, weird) forum or any other contact method on their wiki way earlier.

When briefly searching on the internet, it looks like Sipeed was funded around 2015-2018 and does have around 12 different products (and probably a smaller team of enthusiasts to manage them) - so not really "fly by night". But in any case, the company seems to be more mature than this insulting comment.

closed 2024-10-10 by wj-xiao