sipeed/NanoKVM #76: New NanoKVM - /dev/mmcblk0p3 - no filesytem

madworm, 2024-09-30:
Just letting you know that I've received a "fresh" NanoKVM today

* /dev/mmcblk0p3 had no filesystem at all - according to the wiki it should be exFAT (I could fix that with mkfs.exfat, now works with usb)
* /etc/fstab doesn't have a mount point for it pointing to /data (but it gets mounted anyway after creating the exfat fs)

polyzium, 2024-09-30:
This must be the data partition that is automatically created when the KVM boots. This happens whenever the firmware is reflashed, so I assume these little guys are mass-produced and were probably never turned on at the factory.

See https://github.com/sipeed/NanoKVM/issues/7#issuecomment-2288903849 for reflashing the firmware.

madworm, 2024-09-30:
Well, the partition was there, but no exFAT had been created. I dumped the partition with dd and it was all zeros.

After creating the filesystem myself it worked.

Version 2.0.9

neoKushan, 2024-09-30:
Yeah I got a new one a couple of days ago and had to re-image the included MicroSD card, as it wasn't working correctly.

madworm, 2024-10-09:
What method did you use for that?

I tried step #5 from this, but nothing happened. 

`https://wiki.sipeed.com/hardware/en/kvm/NanoKVM/user_guide.html`

I copied the decompressed image file to the root of /dev/mmcblk0p2

The wiki makes it sound like upon reboot something is supposed to happen with it, but it just stays there. Or do I need to write the image to the sd-card with dd ?

vodkapmp, 2024-10-09:
> The wiki makes it sound like upon reboot something is supposed to happen with it, but it just stays there. Or do I need to write the image to the sd-card with dd ?

Yeah, use dd to write it to the SD and you should be good

Z2Z-GuGu, 2024-10-14:
@madworm Hi, we have implemented partition formatting and data mounting in/etc/init.d/S01fs. Has this been helpful for your question?

NightowlKr, 2025-01-03:
"I hope my issue report #242 is helpful."

mr-manuel, 2025-12-29:
Very odd. I flashed 9 new SD cards, and the problem appeared only on the very first one...

dormancygrace, 2026-09-14:
Hi! I've included a fix for this in [NanoKVM OS](https://github.com/dormancygrace/NanoKVM-OS), my independent community firmware for SG2002-based NanoKVM Cube and PCIe.

Data provisioning now tracks an interrupted format separately from a ready filesystem, retries the eligible unfinished format and only marks the data partition ready after successful formatting/mounting. New data partitions are labeled exFAT.

This addresses the provisioning failure path. Fresh-card first boot of this exact release remains a physical qualification item; an unknown existing data partition is not blindly reformatted.

[Implementation / details](https://github.com/dormancygrace/NanoKVM-OS/blob/9f65f9554142db4a61ba866b4498363f4f21d711/kvmapp/system/init.d/S01fs).

If you're still experiencing this issue, you're welcome to try it and let me know whether it resolves the problem on your setup.

It's currently in beta. Development testing has been on PCIe/UXC; Cube testing is still pending. Installing from stock firmware requires flashing a full SD image, so please back up your card first.

[Latest release](https://github.com/dormancygrace/NanoKVM-OS/releases/latest) · [Installation guide](https://github.com/dormancygrace/NanoKVM-OS/blob/main/docs/INSTALL.md)

Please include your board revision and reproduction steps when reporting results. Thanks!