sipeed/NanoKVM #77: PC won't boot if PC-USB cable is connected to the PC.

mcury1985, 2024-10-01:
The title says it all, it seems to be a short, not sure.
As soon as I unplug the PC-USB cable from the PC usb port, it boots perfectly.

Anyone knows what to do ?

alexislefebvre, 2024-10-01:
See:

- https://github.com/sipeed/NanoKVM/issues/40#issuecomment-2336927509

mcury1985, 2024-10-01:
> Is it the same issue than this one?
> 
>     * [Power to host via 5V USB? #40 (comment)](https://github.com/sipeed/NanoKVM/issues/40#issuecomment-2336927509)

That is it.. exactly.
By killing USB ports in the desktop, it boots perfectly. So it confirms that there is a reverse current preventing the board to boot.

I'm reading that topic to see what is going to be my next action. Perhaps mod the USB cable will be easier for me..

mcury1985, 2024-10-01:
Just modded the USB cable, everything is working now with both cables connected, PC-USB and AUX power.
PC-USB PIN 1 with an electrical tape did the job.

mcury1985, 2024-10-03:
I'm closing this, thanks @alexislefebvre

closed 2024-10-03 by mcury1985