sipeed/NanoKVM #78: Why not release a high performance hardware version?

gailulun, 2024-10-03:
Given that hardware upgrades are often difficult, I think using higher-performance hardware is a direction worth considering. For example, pikvm recommends 2GB of memory, while nanokvm currently only has a 256MB version. Using only 1/8 of the memory may become a bottleneck for later software upgrades.

Jakdaw, 2024-10-03:
The clue is in the name; not sure whether this is the project for you.

gailulun, 2024-10-03:
> The clue is in the name; not sure whether this is the project for you.

I'm afraid it's not what you think. The developers intend to port the pikvm system, which means they don't define this project as just a "lightweight pikvm replacement with limited functions." This is a project with phased significance, so please don't put shackles on yourself.

polyzium, 2024-10-03:
The NanoKVM is based on the LicheeRV Nano, a tiny SBC. Many compromises had to be made to fit a complete system capable of running Linux into such a small size.
> The developers intend to port the pikvm system

I am starting to doubt it, this is probably some kind of vaporware to boost SEO results. If you want PiKVM, you are stuck with either the unreasonably priced PiKVM V4 Mini/Plus or a cheap clone. Sure, PiKVM is a much more mature project, but it wasn't built for such low-power hardware, especially RISC-V.

gailulun, 2024-10-03:
> The NanoKVM is based on the LicheeRV Nano, a tiny SBC. Many compromises had to be made to fit a complete system capable of running Linux into such a small size.
>
>
> I am starting to doubt it, this is [...]

I mean, we don't have to port the pikvm system. RISCV also has corresponding and developing Linux distributions and even BSD distributions available. Why do we have to limit ourselves to busybox?

gailulun, 2024-10-03:
Believe me, this is good for everyone: the development difficulty for developers will be reduced; the company will enrich its product line; users will have more choices; and the project will have more flexibility and development potential. . .

patschi, 2024-10-03:
> That's the point, so we need a higher performance version.I mean, we don't have to port PikVM.

The project name "NANOKvm" implies it should be reasonable sized and rather of a compact/small size. I think they do achieve the goal very well. Especially with the very competitive, attractive price.

> we don't necessarily need such low power consumption

For me, this is a plus. As such things intend to run 24/7, it might a small difference - especially in scale. Also, the fact you can power it just through almost every normal USB port is a very good plus.

> Why should we give up future scalability

Enterprise built-in KVMs (like Dell iDRAC, HPE iLO, etc) are even smaller and probably hardware-wise not much more powerful. They do all they need to do, just like NanoKVM. I think there's enough features you can add without hitting a resource limitation. 

> limit ourselves to busybox due to insufficient resources

I like minimalism. Less attack vectors. No need to put tons of bloatware into it.

> RISC-V also has corresponding and developing Linux distributions and even BSD distributions available.

If you're looking for a full-blown machine to put all kind of application and services on it, NanoKVM is probably wrong for your use-case. Here PiKVM with a Raspberry Pi might be a way better fit.

Overall: I think they achieve their goal of delivering a small NanoKVM at a competitive price very well. I'm planning to build NanoKVM inside some server chassis as a "external KVM module" (internal VGA header, power straight from PSU, etc) and it's the perfect fit.

patschi, 2024-10-03:
It is probably easier said than done. More powerful hardware means more costs, more electricity you might need to serve during peak times, and more heat in a relatively small unit. It's a trade off. Granted, I don't know the chip specifications, so there might be more powerful ones with pretty much unchanged factors mentioned above...

> PS: I am very interested in your last sentence. Using the interface inside the chassis as much as possible will greatly improve stability and neatness.

It's not much magic. I do have a 19" router chassis with a N305, which obviously brings no KVM. The goal running it in a datacenter makes it difficult to access if any problems occur, so NanoKVM is my plan to go: 
- PSU provides power (molex to USB adapter)
- PCIe-to-USB adapter for USB HID
- VGA-header to VGA adapter to get VGA port inside the chassis
- VGA-to-HDMI adapter (power from PSU too) to get it do HDMI
- The KVM-B for POWER/RESET switches also somewhere internally, there are header ports I could access
- LAN is one of the tricker bits and I will probably just put a LAN cable out of the back of the chassis...

gailulun, 2024-10-04:
>More powerful hardware means more costs, more electricity you might need to serve during peak times, and more heat in a relatively small unit. 
These are completely acceptable to me.

- The existence of such a high premium for pikvm.
- Various feature requests in the feature list.
- Not everyone has a computer room where all functions can be separated and implemented by different dedicated equipment.

The above shows that there are many people who need "fat" ipkvm.

gailulun, 2024-10-04:
> More powerful hardware means more costs, more electricity you might need to serve during peak times, and more heat in a relatively small unit.

These are completely acceptable to me.

* The existence of such a high premium for pikvm.
* Various feature requests in the feature list.
* Not everyone has a computer room where all functions can be separated and implemented by different dedicated equipment.

The above shows that there are many people who need "fat" ipkvm.

gailulun, 2024-10-04:
- The project has a high level of attention (look at the growth rate of stars!)
- Both hardware and software are open source
- Reasonable price and cost
These are all our advantages.

Jersh, 2024-10-08:
@gailulun Get deleloping then?
High performance version sounds good like you said.
I'd be interested to follow your progress...

Zepan, 2024-10-14:
Hi, we will extend product types first(NanoKVM-Cube/PCIe/USB/DP/...), after gather enough suggestion, we will make new generation product. (It will have dual GbE next version)