sipeed/NanoKVM #79: Feature Request: Zerotier

mangotek, 2024-10-03:
Simliar to Tailscale, alternative VPN is Zerotier a SD-WAN

Zepan, 2024-10-14:
Hi, what is the advantage of Zerotier, compares to Tailscale.
We will add support if it have enough advantages, otherwise you can port by yourself as it is opensource now.

amyc935, 2024-10-16:
> Hi, what is the advantage of Zerotier, compares to Tailscale. We will add support if it have enough advantages, otherwise you can port by yourself as it is opensource now.

The performance of SG2002 itself is not very powerful.
current Tailscale will take up too much performance, and zerotier will provide a better experience.

Pouletgrill, 2024-10-17:
I would personally only buy the NanoKVM if it can be integrated in my zero tier environement. I am sure im not the only one out there that use zero tier and dont want to use Tailscale in top of that. 
Thank you.

Zepan, 2024-10-22:
I see, we will consider it

shimco, 2024-12-21:
On my remote devices including pikvm's, which are often deployed hundreds of miles away, I use both Tailscale and Zerotier to provide redundancy if one service fails.
Adding multiple options like this would be beneficial for the NanoKVM project, so I hope you can implement this.

staberas, 2025-01-18:
---

## **Incomplete and **NOT** yet working Guide to Installing ZeroTier 1.8.1 on NanoKVM **

### **Steps**

#### 1. **Download ZeroTier 1.8.1 `.deb` File**
Use `wget` to fetch the ZeroTier 1.8.1 `.deb` file for `riscv64`:
```bash
mkdir zero
cd zero
wget https://download.zerotier.com/debian/bullseye/pool/main/z/zerotier-one/zerotier-one_1.8.1_riscv64.deb
```

#### 2. **Extract the `.deb` File**
Extract the `.deb` contents using `ar` and `xzcat`:
```bash
ar x zerotier-one_1.8.1_riscv64.deb
xzcat data.tar.xz | tar -xvf -
```

#### 3. **Move Files to Correct Locations**
Move the extracted binaries and files to their appropriate locations:
```bash
cp -r ./usr/sbin/zerotier-one /usr/sbin/
cp -r ./var/lib/zerotier-one /var/lib/
cp -r ./etc/init.d/zerotier-one /etc/init.d/
```

#### 4. **Verify Required Libraries (not sure if 1.8.1 requires them)**
Ensure required libraries like `libstdc++`, `libgcc_s`, `libc6`, and `libm` exist in `/lib` or `/usr/lib`. If missing:
1. Download and extract dependencies from [[Debian repositories](https://ftp.debian.org/debian/pool/main)](https://ftp.debian.org/debian/pool/main).
2. Example for `libc6`:
   ```bash
   wget http://ftp.debian.org/debian/pool/main/g/glibc/libc6_2.40-5_riscv64.deb
   ar x libc6_2.40-5_riscv64.deb
   xzcat data.tar.xz | tar -xvf -
   cp -r ./lib/* /lib/
   ```

ZeroTier requires has an extra dependency in order to fully run which is lsb, up until now i havent managed to fix that and im currently editing zerotier code to sidestep this issue , if anyone with more knowledge want to take on the challenge you can try

ciroiriarte, 2025-01-24:
would like to see this included, I extensively use Zerotier already.

AndrewMoryakov, 2025-01-31:
I also actively use Zirotier, it can be much more convenient.

GulinSS, 2025-02-09:
What to do if it stopped to load after reboot? I mean NanoKVM itself does not loading after these steps, seems I got it broken somehow. 😅

staberas, 2025-02-09:
> What to do if it stopped to boot after reboot? I mean NanoKVM itself does not loading after these steps, seems I got it broken somehow. 😅

it doesn't boot at all? or zerotier doesn't start at boot ?

GulinSS, 2025-02-09:
> it doesn't boot at all?

That one. KVM does switch on its display with IP address.

staberas, 2025-02-09:
you have to reflash the sd card, but before that you need to see if the device is available on the network and access it.

GulinSS, 2025-02-10:
Yes, reflasing internal sd card with fresh image fixed the device. Well, another turn to enable ZeroTier :-)

staberas, 2025-02-10:
> [@staberas](https://github.com/staberas) which versions did you use?

[image]

1.8.1 , its the only one i got working . (i started with 1.14.something added the libs on step 4 but it failed to work then went down to 1.8.1 which i know works on the licheeRV )

GulinSS, 2025-02-10:
`1.14` worked for me but I got a broken state after kvm reboot: it was unable to connect to network and turn on lcd.

staberas, 2025-02-10:
@GulinSS  
So apparently 
 
has been replaced by 
http://ftp.debian.org/debian/pool/main/g/glibc/libc6_2.40-6_riscv64.deb
(i have the old one)

regardless

post me what /usr/sbin/zerotier-one tells you

and do the following :
`strings /usr/sbin/zerotier-one | grep '\.so'  `

you should get the following :

```
/lib/ld-linux-riscv64-lp64d.so.1
libstdc++.so.6
libm.so.6
libgcc_s.so.1
libpthread.so.0
libc.so.6
ld-linux-riscv64-lp64d.so.1
/var/run/minissdpd.sock
```

After downloading it should do the following (confirming):

` ar x libc6_2.40-5_riscv64.deb`  or  ` ar x libc6_2.40-6_riscv64.deb `
 
//we dont need debian-binary
` rm control.tar.xz debian-binary 
`
//decompress and move the files 
```
 xzcat data.tar.xz | tar -xvf - 
 mv usr/lib/* /lib/ 

```
//remove the data.tar when done
`  rm -rf usr lib data.tar.xz `

Josephur, 2025-02-12:
Thanks a lot guys! I was able to get zerotier working on my nanovkm using latest firmware, and by using the directions above but using the latest libc from http://ftp.debian.org/debian/pool/main/g/glibc/libc6_2.41-1_riscv64.deb

However I'm not sure if I missed something in the instructions but for me the usr/lib/zerotier-cli and usr/binzerotier-idtool binaries that came in the deb file did not copy, I just copied them manually to /usr/sbin

This really needs to be a built in feature to choose between zerotier and/or tailscale.

Josephur, 2025-02-12:
@staberas I'm going to flash back to stock and make a YT video actually, as I see one does not yet exist.  I will provide the commands I use here as well.  I hope to get it done in the next few days.

yesm1ke, 2025-03-07:
Just preordered nanokvm and can't wait for delivery and zerotier support out of box!

Great product, thanks guys!

Msprg, 2025-06-29:
@Josephur 
Hey guys, have you came any closer to making a more complete guide/video about installing zerotier? I tried it once but failed pretty miserably, so I decided to wait since it looked like you were putting some instructions together.

GulinSS, 2025-06-30:
@Msprg, no, who got it to work and... well.. after that everything more is eaten by laziness.

staberas, 2025-06-30:
@Msprg which version did you tried to install ?

Msprg, 2025-06-30:
I can't remember anymore, however, I can try to reproduce it if you'd like

Msprg, 2025-06-30:
Okay so...
first of all:
```
wget https://download.zerotier.com/debian/bullseye/pool/main/z/zerotier-one/zerotier-one_1.8.1_riscv64.deb
wget: not an http or ftp url: https://download.zerotier.com/debian/bullseye/pool/main/z/zerotier-one/zerotier-one_1.8.1_riscv64.deb
```

This is fun, curl works...

the rest went technically well, I skipped the step 4 you said to skip

and then I ended up basically here:

```
/zero # /etc/init.d/zerotier-one start
/etc/init.d/zerotier-one: line 25: log_daemon_msg: not found
nice: can't execute '/usr/sbin/zerotier-one': No such file or directory
/etc/init.d/zerotier-one: line 27: log_end_msg: not found
/zero # ls /usr/sbin/zerotier-one 
/usr/sbin/zerotier-one
/zero # chmod +x /usr/sbin/zerotier-one 
/zero # /etc/init.d/zerotier-one start
/etc/init.d/zerotier-one: line 25: log_daemon_msg: not found
nice: can't execute '/usr/sbin/zerotier-one': No such file or directory
/etc/init.d/zerotier-one: line 27: log_end_msg: not found
```

And the last time I ended up here I couldn't make any further progress...

staberas, 2025-06-30:
log_daemon_msg , log_end_msg are all LSB dependency https://www.linuxfromscratch.org/lfs/view/development/scripts/apds02.html <- have you added the script ?   
also do an ls on /usr/sbin/ , does  "zerotier-one"  actually exists?

Msprg, 2025-06-30:
Of course I have added the script, and of course the zerotier binary exists, otherwise the chmod +x command you see above, would fail:

```
/zero # head /lib/lsb/init-functions
#!/bin/sh
########################################################################
#
# Begin /lib/lsb/init-funtions
#
# Description : Run Level Control Functions
#
# Authors     : Gerard Beekmans - gerard@linuxfromscratch.org
#             : DJ Lucas - dj@linuxfromscratch.org
# Update      : Bruce Dubbs - bdubbs@linuxfromscratch.org
/zero # ls /usr/sbin/zerotier*
/usr/sbin/zerotier-one
/zero # 
```

staberas, 2025-06-30:
ok will check my nanokvm later this evening , i might have done some editing in the zero-tier one to make LSB work

Msprg, 2025-06-30:
Thank you very much, I would greatly appreciate that~

staberas, 2025-07-01:
> And on that note, the link http://ftp.debian.org/debian/pool/main/g/glibc/libc6_2.40-6_riscv64.deb is no longer valid, and has been now superseded by http://ftp.debian.org/debian/pool/main/g/glibc/libc6_2.41-9_riscv64.deb

yes , it should be that thats missing now 
`# strings /usr/sbin/zerotier-cli | grep '\.so'
`
this will show you what is needed 
`/lib/ld-linux-riscv64-lp64d.so.1
libstdc++.so.6
libm.so.6
libgcc_s.so.1
libpthread.so.0
libc.so.6
ld-linux-riscv64-lp64d.so.1
/var/run/minissdpd.sock
`
here my version in case the new one doesnt work or smthg
[](https://github.com/staberas/libc6_2.40-5_riscv64_/blob/main/libc6_2.40-5_riscv64.deb
)
```
ar x libc6_2.40-5_riscv64.deb
xzcat data.tar.xz | tar -xvf -
cp -r ./lib/* /lib/

```

Ferohers, 2025-08-25:
> [#79 (comment)](https://github.com/sipeed/NanoKVM/issues/79#issuecomment-2599633763)
> 
> this line `wget http://ftp.debian.org/debian/pool/main/g/glibc/libc6_2.40-5_riscv64.deb` not work. The newest is: `wget http://ftp.debian.org/debian/pool/main/g/glibc/libc6_2.42-1_riscv64.deb` From this link: http://ftp.debian.org/debian/pool/main/g/glibc/
> 
> Like you should put it like this is better: `wget http://ftp.debian.org/debian/pool/main/g/glibc/libc6_<the lastest version num>_riscv64.deb`
> 
> And!!
> 
> the fun "log_daemon_msg" is not in the "init-functions" from https://www.linuxfromscratch.org/lfs/view/development/scripts/apds02.html
> 
> So I use the dummy fun instead.....like below:
> 
> ```
> log_daemon_msg () {
>     # Dummy function to be replaced by LSB library.
> 
>     echo $@
> }
> log_progress_msg() {
>    echo $@
> }
> log_end_msg () {
>     # Dummy function to be replaced by LSB library.
> 
>     if test "$1" != "0"; then
>       echo "Error with $DESCRIPTION: $NAME"
>     fi
>     return $1
> }
> ```
> 
> Come on.... linux world is full of this for another half century? Shall do many cmd things and dependcyxxx things by users? Just write some smart scripts to handle the install and config things.... Is our time not valuable?

So sad to see how this will break heart of new users...

staberas, 2025-08-26:
@cvocvo 
as far as i have , no , the only thing that might get broken is the boot order number [S97] and at that point you just need to set a new number

manueldev, 2025-11-02:
> [@staberas](https://github.com/staberas) Thank you for all the help, seems I've finally got it working then. I've put together an installation script and then put in in a repo so that it's easier to access and maintain it.
> 
> Repo here: https://github.com/Msprg/nanoKVM-ZeroTier-one
> 
> Or all-on-one release for this thread here: [nanokvm-zerotier.zip](https://github.com/user-attachments/files/20997614/nanokvm-zerotier.zip)

Installed and working without problems on version 2.2.9.

Msprg, 2025-11-02:
I'm very happy to hear that! 🥳