sipeed/NanoKVM #80: 4-6 second delay between key input and reflected in video

CRCinAU, 2024-10-03:
Hi all,

I just received my devices today, updated them to 2.0.9. After this, just using a terminal console on a proxmox server, I can measure between 4-6 seconds of lag between typing in a key on the web interface and it showing on the video output.

Any suggestions on how to make this a little more responsive?

polyzium, 2024-10-03:
Try disabling Frame Detect and lowering the JPEG quality to 50%.

CRCinAU, 2024-10-03:
Thanks - I did some tweaking, as well as rebooting each system with the IPKVM connected to the HDMI port, and things seem to be a little more sane now...

I'm curious - if I change the resolution on on the web UI - does this change the EDID information send to the OS?

It would be great if the modeset on the linux terminal gets set to the requested resolution on boot...

polyzium, 2024-10-03:
The resolution drop-down you see in the web UI only affects the streaming resolution and doesn't affect the EDID at all. Previous versions of the NanoKVM firmware didn't take resolutions other than 1080p, thankfully this is no longer the case. However, there is no custom EDID like on the PiKVM, but this seems to be planned. See https://github.com/sipeed/NanoKVM/issues/5#issuecomment-2343107386

closed 2024-10-03 by CRCinAU