sipeed/NanoKVM #82: Doc request: scripts

FlorianHeigl, 2024-10-04:
Hi,

at the moment the 'scripts' feature is not listed in the documentation.
I can see they're uploaded to `/etc/kvm/scripts/` but it's unclear to me what scope of things they are meant to do and if there's any interface to connect to for them.
i.e. are they expected to do things like
- 'chose iso file and trigger reboot, and press enter after 3 seconds'
- 'fetch the most recent iso images from our main repository'
- 'clone the scripts directory from github'
- 'upload a screen dump'
- 'set the cat pic from /bin as windows kernel boot image'

and whether anything exists to support such tasks.

patschi, 2024-10-08:
The source code has just been published a few hours ago, and it looks like scripts are just ran locally on the NanoKVM: https://github.com/sipeed/NanoKVM/blob/main/server/service/vm/script.go. For what use-cases they provide? Not sure...

Would be interesting to know if it could be used to automate sequences, like mounting a pre-loaded ISO and then auto-typing keyboard keys or controlling the mouse...

robot-o, 2024-11-06:
it would be great if we could use this feature to interact with the MCU pins on the NanoKVM lite. One usecase would be triggering buttons or interacting with the system programatically (like triggering the inputs on an usb switch/ hardware kvm switch to multiplex the nanokvm)