sipeed/NanoKVM #84: High load average without any peripheral attached

SuperKali, 2024-10-05:
Hi,

I don't think that's normal, but the load of the nanokvm is always high, even without any streaming video:

[image]

It's fixable?

Z2Z-GuGu, 2024-10-14:
Hi, the SG2002 used by NanoKVM only has one RISC-V (C906) running Linux, and its performance is not strong, so even small tasks will have a certain amount of CPU usage, which is normal.

jidckii, 2026-01-31:
In my view 1 GHz seems more than enough even on a single core.

It looks like the issue is simply that Linux is not optimized for the RISC-V architecture.
On similar ARMv7 hardware everything flies, even with more processes running.
Here, a 30 MB file takes 1.5 minutes to download with constant timeouts, the network barely works due to LA of 5. I can't imagine how long it would take to download a 5 GB ISO.

[image]

What's strange is there are no heavy processes, CPU shows 90% idle, no IO wait — yet LA stays at 3-5 for no apparent reason. The load average calculation seems broken or the kernel scheduler has issues on RISC-V.

<details>
<summary>Diagnostic info</summary>
```
<img width="1093" height="711" alt="Image" src="https://github.com/user-attachments/assets/ec128c5a-f0a1-48ee-ae40-70c933bc1e5b" />

<img width="1099" height="719" alt="Image" src="https://github.com/user-attachments/assets/c361cabd-af48-42f1-b00f-2b3b654aa162" />

<img width="340" height="144" alt="Image" src="https://github.com/user-attachments/assets/aea7a45d-aaa5-4fe4-810f-0fe50962b28d" />

<img width="1080" height="796" alt="Image" src="https://github.com/user-attachments/assets/8ecef6f4-2dcf-424b-a481-054b839dfcb1" />

<img width="1080" height="796" alt="Image" src="https://github.com/user-attachments/assets/7707852c-4322-400a-b421-022cdca386c6" />

</details>

dormancygrace, 2026-09-14:
Hi! I've included a fix for this in [NanoKVM OS](https://github.com/dormancygrace/NanoKVM-OS), my independent community firmware for SG2002-based NanoKVM Cube and PCIe.

The VI driver uses idle waits for its inactive event threads, so sleeping capture workers no longer inflate load average. This change was exercised with driver reload/video restart checks.

This fixes idle load accounting; it does not establish that every device freeze or slowdown has been fixed.

[Implementation / details](https://github.com/dormancygrace/NanoKVM-OS/blob/9f65f9554142db4a61ba866b4498363f4f21d711/firmware/osdrv/patches/0023-vi-idle-wait-accounting.patch).

If you're still experiencing this issue, you're welcome to try it and let me know whether it resolves the problem on your setup.

It's currently in beta. Development testing has been on PCIe/UXC; Cube testing is still pending. Installing from stock firmware requires flashing a full SD image, so please back up your card first.

[Latest release](https://github.com/dormancygrace/NanoKVM-OS/releases/latest) · [Installation guide](https://github.com/dormancygrace/NanoKVM-OS/blob/main/docs/INSTALL.md)

Please include your board revision and reproduction steps when reporting results. Thanks!