sipeed/NanoKVM #85: FR-CH keyboard - cannot send special characters

jcheger, 2024-10-05:
Quite an extension of #27 and #32.

FR-CH (Swiss-French) keyboard is a QWERTZ variant, using Alt-Gr for special characters. But I cannot find a way to send them.

[image]

At least, those characters are mandatory for remote maintenance (Windows and Linux):

| char | ascii hex |
|---|---|
| @ | 0x40 | 
| # | 0x23 | 
| < | 0x3c |
| > | 0x3e |
| \ | 0x5c |
| \| | 0x7c |

« Alt-Gr 1 » or « Alt-Gr 7 » for « | » (pipe) can be quite confusing, depending on the program for remote control, and the OS (Windows and Linux behave differently).

Versions:
- OS: 1.2.0
- NanoKVM: 2.0.9

Yeocti, 2024-10-06:
I can confirm this issue using a french keyboard (AZERTY).
Special character requiring the use of Alt-Gr are not sent.

IsayIsee, 2024-10-08:
double  ctrl, double prtsc also not sent 
some software requires special key to activate
recommended to implement standard keyboard functions and not to artificially filter certain keys.

azsde, 2025-02-03:
This is a huge issue, I'm very disappointed this still hasn't been addressed ...

v7o73, 2025-02-07:
I can confirm the same issue with DE-CH (Swiss-German) QWERTZ  Keyboard.
This is a huge issue for me to. I cant use it as it is now.

Anacondo, 2025-04-08:
Same issue here with a Spanish layout (ES) keyboard. Unable to send the "|" or "@" or anything related to the ALT GR key, which is quite limiting. Hoping for a solution soon, having to copy+paste them from somewhere else is a hassle.

azsde, 2025-04-08:
A workaround is to use ctrl + alt keys on the left of the keyboard, but this is really annoying and I don't understand what is taking so long for them to fix this bug