sipeed/NanoKVM #86: Screen not working

pabloalcantara, 2024-10-07:
Hi,
The display is not working anymore. It worked for more or less 2 days, then now it shows only an icon o a monitor with a cross in it. I'm using the 2.0.9 version
I know it's not the output of my pc, because taking the hdmi cable and putiing in a monitor, it works perfectly.
[image]

ep1cman, 2024-10-07:
I just had the same issue with a device. I reflashed the SD card, it still didn't work. Once I did an update though it came back alive.

pabloalcantara, 2024-10-07:
> I just had the same issue with a device. I reflashed the SD card, it still didn't work. Once I did an update though it came back alive.

I´ll try to reflash.

Igor-M-77, 2024-10-07:
I have a similar problem. HDMI stopped working after half a day. I have two nano kvm devices. The second one works if I use it. If you managed to solve the problem, then tell me where I can find instructions for reflashing?

ep1cman, 2024-10-07:
> I have a similar problem. HDMI stopped working after half a day. I have two nano kvm devices. The second one works if I use it. If you managed to solve the problem, then tell me where I can find instructions for reflashing?

Take the four screws at the bottom off and there is a micro sd card. You can find the image to flash to it on the release page of this GitHub repository. Bare in mind it’s compressed, I’d recommend a tool like “balena etcher” which can decompress it and write it to the sd card

Igor-M-77, 2024-10-07:
> Take the four screws at the bottom off and there is a micro sd card. You can find the image to flash to it on the release page of this GitHub repository. Bare in mind it’s compressed, I’d recommend [...]

It worked after reflashing and updating. I was already starting to worry that it was a hardware problem. Thank you.

pabloalcantara, 2024-10-08:
Reflashed. But still hit and miss...
Some times works, and some don´t..After a lot of reboot.

marshalleq, 2024-10-08:
Just had this happen too, it seems to come right by itself sometimes.  This time it isn't.  Refreshing is a bit extreme, we probably need to find out what is causing this....

patschi, 2024-10-08:
I have had the same issue in the past. Somehow a random mixture of power-resetting, re-attaching HID and HDMI fixed it. But only seen it once so far...

pabloalcantara, 2024-10-08:
I think I discovered the possible problem USB power. My device is the light one, so is using power from the USB, maybe, only maybe, the USB can´t delivere the necessary power.
Ony one know how to use an external power supply on it using the pin headers?
I know that it have 3 power pins: vsys 5v, vbus 5v and 3v3 and 2 ground pins. Some time ago i played with arduino, and I have an external power suply that can output 5v or 3.3v, and I have all the necessiry cables. But I don´t know witch power pin I can use.

haggis-supper, 2024-10-10:
> I think I discovered the possible problem USB power. My device is the light one, so is using power from the USB, maybe, only maybe, the USB can´t delivere the necessary power. Ony one know how to [...]

I had the same problem with my nanoKVM Lite. 
So i've done a couple of things
1. Reflashed the SD card ( went here and pulled down the 1.2.0 version [images](https://github.com/sipeed/NanoKVM/releases) as i noticed when i looked at the "About KVM " pulldown it showed image version 1.0.0 )
2. I then used the option in the KVM to "Check for Update" which updated me from Application version 1.0.0 to 2.0.9.
3. Got a shorter USB-A to USB-C cable ( 0.5m )

Now it seems to be working ok.

marshalleq, 2024-10-10:
I thought mine is plugged into an external power source, but I've just seen that it isn't.  So it could be something...  No idea why that makes it lose its marbles though.  I wonder if just external powering it is enough to sort it on its own.  Somehow I doubt it but I really don't want to have to pull it all apart again to get the card out....

kcastner, 2024-10-10:
My devices expand the view to the NanoKVM. If i set it to mirror, the NanoKVM wont show any HDMI output anymore. 
Same problem at changing screen resolution on host system.

I have to reboot the NanoKVM to get it working again.
It seems that the video input wont be reenabled after any input change. Replug wont work either.

Can anyone reproduce this problem? I think its related to it. 

My current workaround: open NanoKVM terminal via GUI and type "reboot"

Joshndroid, 2024-10-10:
I appear to have the same issue - NanoKVM Full - 2.0.9
I can set it up to work after a clean flash of the sdcard, everything works as expected and I can see the screen.
Once i shut down the PC I get the blank screen.
I can boot the PC, and it appears that the mouse movement moves ont he screen however the video is blank.
Rebooting the NanoKVM doesn't make any difference, unplugging and replugging the hdmi makes no difference, still blanks screen.
The only way I have been able to get it to work is by reflashing the sd-card again

marshalleq, 2024-10-10:
I am new to this project, but I assume there will be a way of providing logs before we re-flash the device so that we can help pin this down?  Anyone know or have a link for a procedure?  Or otherwise what logs might be needed?  I haven't refreshed mine yet in the hope I could provide some kind of details to help, rather than just do a workaround.

kcastner, 2024-10-11:
> otherwise what logs might be needed? I haven't refreshed mine yet

my 4 devices are ready to provide logs too - does anybody know where to find those?

Device: NanoKVM Full - GUI v.2.0.9

haggis-supper, 2024-10-11:
The only log file I can see is the /var/log/messages file, everything seems to be executable.So unless there is a command to dump logs from the running executables I don't know.Will scan the docs and see myself.

aka-charos, 2024-10-14:
I can confirm the issue myself. Not sure what triggered it , but after a week of usage, I get blank screen (with the "X"). 
I think developers needs to step in, identify and fix the issue. 
Re-flashing is not even a proper workaround. The whole idea of KVM is remote access. If I have to flash firmware everytime this happens, what's the point of having a KVM in the first place. Mine is plugged into a different country. I can only "fix" this in a month...this is ridiculous

Z2Z-GuGu, 2024-10-14:
Thanks for your feedback. I am currently trying to reproduce this issue. When you encounter this problem, you can try entering the command `/etc/init.d/S95nanokvm restart_stream`. If the problem still persists, please paste the log here. Thanks!

Joshndroid, 2024-10-14:
What log do you require and how do we obtain it?

Mine does it everytime I shut down my PC which renders the KVM worse than useless at present.

I will report back in a couple hours my result of the below command

> Thanks for your feedback. I am currently trying to reproduce this issue. When you encounter this problem, you can try entering the command `/etc/init.d/S95nanokvm restart_stream`. If the problem still persists, please paste the log here. Thanks!

Joshndroid, 2024-10-14:
@Z2Z-GuGu 

I came home to see my nanokvm had the monitor with the X in it.

I ran the command it it doesn't make a difference.

`/etc/init.d/S95nanokvm restart_stream
killall: kvm_stream: no process killed
OK

[KVM_stream]:start
HDMI res modification event
new res: 1920 * 1080
CSI res: 1920 * 1080
VI RES = 1920 * 1080
&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&
1920 * 1080
[SAMPLE_COMM_SNS_ParseIni]-2204: Parse /mnt/data/sensor_cfg.ini
[parse_source_devnum]-1797: devNum = 1
[parse_sensor_name]-1878: sensor = LONTIUM_LT6911_2M_60FPS_8BIT
[parse_sensor_busid]-1907: bus_id = 4
[parse_sensor_i2caddr]-1918: sns_i2c_addr = ff
[parse_sensor_mipidev]-1929: mipi_dev = 0
[parse_sensor_laneid]-1940: Lane_id = 2, 4, 3, 1, 0
[parse_sensor_pnswap]-1951: pn_swap = 0, 0, 0, 0, 0
[parse_sensor_mclken]-1973: mclk_en = 0
[parse_sensor_mclk]-1984: mclk = 0
try release sys ok
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^reinit_soph_vb
rmmod: can't unload module 'soph_vpss': Resource temporarily unavailable
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^reinit_soph_vb
insmod: can't insert '/mnt/system/ko/soph_base.ko': File exists
insmod: can't insert '/mnt/system/ko/soph_rtos_cmdqu.ko': File exists
insmod: can't insert '/mnt/system/ko/soph_fast_image.ko': File exists
insmod: can't insert '/mnt/system/ko/soph_mipi_rx.ko': File exists
insmod: can't insert '/mnt/system/ko/soph_snsr_i2c.ko': File exists
insmod: can't insert '/mnt/system/ko/soph_vi.ko': File exists
insmod: can't insert '/mnt/system/ko/soph_vpss.ko': File exists
HDMI res modification event
new res: 1920 * 1080
CSI res: 1920 * 1080
kvm_cfg.fps = 30
kvm_cfg.qlty = 60
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^reinit_soph_vb
[_mmf_init]-1142: maix multi-media version:67a90e2-musl_riscv64
[SAMPLE_COMM_SNS_ParseIni]-2204: Parse /mnt/data/sensor_cfg.ini
[parse_source_devnum]-1797: devNum = 1
[parse_sensor_name]-1878: sensor = LONTIUM_LT6911_2M_60FPS_8BIT
[parse_sensor_busid]-1907: bus_id = 4
[parse_sensor_i2caddr]-1918: sns_i2c_addr = ff
[parse_sensor_mipidev]-1929: mipi_dev = 0
[parse_sensor_laneid]-1940: Lane_id = 2, 4, 3, 1, 0
[parse_sensor_pnswap]-1951: pn_swap = 0, 0, 0, 0, 0
[parse_sensor_mclken]-1973: mclk_en = 0
[parse_sensor_mclk]-1984: mclk = 0
pstSize->u32Width = 1920
pstSize->u32Height = 1080
pstSize->u32Width = 1920
pstSize->u32Height = 1080
pstSize->u32Width = 1920
pstSize->u32Height = 1080
ISP Vipipe(0) Allocate pa(0x8b302000) va(0x0x3fd591f000) size(291424)
pstSize->u32Width = 1920
pstSize->u32Height = 1080
stSnsrMode.u16Width 1920 stSnsrMode.u16Height 1080 60.000000 wdrMode 0 pstSnsObj 0x3fd608b2e8
pstSize->u32Width = 1920
pstSize->u32Height = 1080
pstSize->u32Width = 1920
pstSize->u32Height = 1080
[SAMPLE_COMM_VI_StartMIPI]-494: sensor 0 stDevAttr.devno 0
[sensor_prob = lt6911_probe]
########### PinMux #######################################################################
########### PinMux End #######################################################################
[lt6911_probe] -189
[lt6911_probe] -195
[lt6911_i2c_write] : addr = ee. data = 1
[lt6911_i2c_read] : addr = 0
[lt6911_i2c_write] : addr = ee. data = 0
[lt6911_i2c_write] : addr = ee. data = 1
[lt6911_i2c_read] : addr = 1
[lt6911_i2c_write] : addr = ee. data = 0
[lt6911_probe] jump return nVal -198
[lt6911_probe] -201
data:ffffffff ffffffff
[lt6911_probe] jump return CVI_FAILURE -206
[lt6911_probe] -208 = exit success
pstSize->u32Width = 1920
pstSize->u32Height = 1080
20241014 06:36:00.657 1870 E isp AE_InitSensorFpsPara:5555 pfn_cmos_fps_set NULL!

AE_SetExposure inttime update NULL.
awbInit ver 6.9@2021500
0 R:1400 B:3100 CT:2850
1 R:1500 B:2500 CT:3900
2 R:2300 B:1600 CT:6500
Golden 1024 1024 1024
WB Quadratic:0
isWdr:0
20241014 06:36:00.667 1870 E isp setNoiseProfile:3460 Noise profile get fail. Please check

cvi_bin_isp message
gerritId: 97347 commitId: fc6ce647
md5: d6db2297ddfd44e8252c1f3f888f47b2
sensorNum 1
sensorName0 6911

PQBIN message
gerritId:
commitId: fc6ce647
md5: d6db2297ddfd44e8252c1f3f888f47b2
sensorNum 1
sensorName0 4653

author: lxowalle desc: gc4653 30fps
createTime: 2024-03-26 13:41:34version:
tool Version: v3.0.5.48 mode:

sensorName(0) mismatch, mwSns:6911 != pqBinSns:4653
[SAMPLE_COMM_ISP_Thread]-95: ISP Dev 0 running!
pstSize->u32Width = 1920
pstSize->u32Height = 1080
VI_SDK_IOC_S_CTRL - vi_sdk_enable_chn NG, Out of memory
HDMI res modification event
new res: 1920 * 1080
CSI res: 1920 * 1080
ISP Vipipe(0) Free pa(0x8b302000) va(0x0x3fd591f000)
-- [E] Trigger signal, code:SIGSEGV(11)!`

Joshndroid, 2024-10-14:
Heres some findings;

- After that failed command, I ran a 'reboot' within the nanokvm terminal and the screen came back.
- I was able to reboot my PC from within windows and the screen continued to function on the nanokvm
- I ran a shutdown and then utilised the nanokvm to turn the PC back on (as I have the full version) and the screen continued to function on the nanokvm
- I turned off my PC within windows and left it for about 8 minutes while I did something else. I turned OFF my primary monitor (to try and simulate what I want to be able to do when I am away) and used the nanokvm to turn it on (it is plugged into a second HDMI port on my graphics card). I get a black screen. Not the monitor with the little X in it, just blank screen. I turned ON my primary monitor and the nanokvm still shows a blank screen however the mouse appears to function on my primary monitor. I ran that earlier restart_stream command and the monitor within the nanokvm began to function once again.

Z2Z-GuGu, 2024-10-14:
@pabloalcantara Thank you for your reply. I'll try to fix this in a subsequent release

lovedayb, 2024-10-21:
@Z2Z-GuGu I'm having the same issue with mine, but when I try to run /etc/init.d/S95nanokvm restart_stream, I get the following error:

`[_mmf_sys_init]-930: system init failed with 0xffffffff
[_mmf_init]-1194: sys init failed. s32Ret: 0xffffffff !
maix multi-media init failed
terminate called after throwing an instance of 'maix::err::Exception'
  what():  : Runtime error: mmf init failed

-- [E] Trigger signal, code:SIGABRT(6)!`

Any ideas?

argakon, 2024-10-27:
Same problem, looks like the problem is in not default resolution and for some reason it never working until i reflash sdcard.
This happens to me, then nanokvm was 3000 km away, and when i need to use it it fail. I have to investigate a proble, when call a person and he reflashed the card.
Please make it high priority, when kvm doesn't show a picture, it is not a kvm.

marshalleq, 2024-10-27:
Yeah, I bought this so I can manage a New Zealand Sever from Colombia.  But presently it's not possible.  This is a pretty foundational issue that makes the device completely worthless.  And taking apart the full product that comes with case etc is far too fidly to be doing more than once a year or so too. Hopefully we can get a fix.  Even if there was something we could ssh into the box and do a temporary fix would be acceptable. @Z2Z-GuGu do you think we will get a fix this year?  Thanks.

haggis-supper, 2024-10-28:
I've got a NanoKVM lite and see that i get no response when i connect up, I always have to do a "Reset Hid", then log out and back in again to see the console and manage to access my console.

My KVM is connected to a linux host, if i dont do those steps i dont see anything on screen ( although no big X ).
Once i did get the big X and i did a couple of things, checked the HDMI cable and also reblast the sd card with 1.2.0.

fizzroy, 2024-10-30:
> What log do you require and how do we obtain it?
>
> Mine does it everytime I shut down my PC which renders the KVM worse than useless at present.
>
> I will report back in a couple hours my result of the [...]

This helped. Thanks. I've had this exact issue for a couple of days now. Reboots did nothing. I think it happened after power failure.

alexmoras, 2024-11-02:
Plus one on this issue, however the `/etc/init.d/S95nanokvm restart_stream` command seems to work for me.

aksultank, 2024-11-02:
Same issue here. I received it today, and it worked for only 40 minutes before it completely stopped. I’ve tried every fix posted here, but nothing has worked. This is incredibly disappointing, and it seems I have no choice but to wait for an update or another solution.

bolausson, 2024-11-02:
Same here - stopped showing the stream `/etc/init.d/S95nanokvm restart_stream` didn't help and did throw an error (I forgot to note it down though). Rebooting, power-cycling, etc - nothing helped.

I had to re-flash the image and update the app.

marshalleq, 2024-11-02:
I've never bothered to re-flash.  It's clear that it will just happen again so there's not point.  Right now this device is a doorstop, but can't even be used for that.  Would sure be nice to hear that someone is even looking at the problem.  I should have got the other pi kvm unit, at least that works.  Now I have no solution and will have to wait 2 months to come back and fix anything that goes wrong.

aka-charos, 2024-11-02:
just for the sake of input, here's the command stuck in a loop in my kvm:
```
# /etc/init.d/S95nanokvm restart_stream
OK
# [KVM_stream]:start
kvm_cfg.fps = 60
kvm_cfg.qlty = 90
kvm_cfg.width = 55596
kvm_cfg.height = 20
kvm_cfg.res = 1080
VI RES = 1920 * 1080 
&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&&
55596 * 20
[SAMPLE_COMM_SNS_ParseIni]-2204: Parse /mnt/data/sensor_cfg.ini
[parse_source_devnum]-1797: devNum =  1
[parse_sensor_name]-1878: sensor =  LONTIUM_LT6911_2M_60FPS_8BIT
[parse_sensor_busid]-1907: bus_id =  4
[parse_sensor_i2caddr]-1918: sns_i2c_addr =  ff
[parse_sensor_mipidev]-1929: mipi_dev =  0
[parse_sensor_laneid]-1940: Lane_id =  2, 4, 3, 1, 0
[parse_sensor_pnswap]-1951: pn_swap =  0, 0, 0, 0, 0
[parse_sensor_mclken]-1973: mclk_en =  0
[parse_sensor_mclk]-1984: mclk =  0
try release sys ok
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^reinit_soph_vb
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^reinit_soph_vb
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^reinit_soph_vb
[_mmf_init]-1142: maix multi-media version:67a90e2-musl_riscv64
[SAMPLE_COMM_SNS_ParseIni]-2204: Parse /mnt/data/sensor_cfg.ini
[parse_source_devnum]-1797: devNum =  1
[parse_sensor_name]-1878: sensor =  LONTIUM_LT6911_2M_60FPS_8BIT
[parse_sensor_busid]-1907: bus_id =  4
[parse_sensor_i2caddr]-1918: sns_i2c_addr =  ff
[parse_sensor_mipidev]-1929: mipi_dev =  0
[parse_sensor_laneid]-1940: Lane_id =  2, 4, 3, 1, 0
[parse_sensor_pnswap]-1951: pn_swap =  0, 0, 0, 0, 0
[parse_sensor_mclken]-1973: mclk_en =  0
[parse_sensor_mclk]-1984: mclk =  0
pstSize->u32Width = 55596
pstSize->u32Height = 20
alloc_buf_size(294912): 294912, phy_addr(8a293000): 0x8a293000, buffer_name: ISP_SHARED_BUFFER_0
ion_data.size:294912, ion_data.addr_p:0x8a293000, ion_data.name:ISP_SHARED_BUFFER_0
pstSize->u32Width = 55596
pstSize->u32Height = 20
pstSize->u32Width = 55596
pstSize->u32Height = 20
ISP Vipipe(0) Allocate pa(0x8a293000) va(0x0x3fbfae4000) size(291424)
pstSize->u32Width = 55596
pstSize->u32Height = 20
stSnsrMode.u16Width 55596 stSnsrMode.u16Height 20 60.000000 wdrMode 0 pstSnsObj 0x3fc02512e8
pstSize->u32Width = 55596
pstSize->u32Height = 20
pstSize->u32Width = 55596
pstSize->u32Height = 20
[SAMPLE_COMM_VI_StartMIPI]-494: sensor 0 stDevAttr.devno 0
[sensor_prob = lt6911_probe] 
###########   PinMux  #######################################################################
###########   PinMux End  #######################################################################
[lt6911_probe] -189 
[lt6911_probe] -195 
[lt6911_i2c_write] : addr = ee. data = 1 
[lt6911_i2c_read] : addr = 0 
[lt6911_i2c_write] : addr = ee. data = 0 
[lt6911_i2c_write] : addr = ee. data = 1 
[lt6911_i2c_read] : addr = 1 
[lt6911_i2c_write] : addr = ee. data = 0 
[lt6911_probe] jump return nVal -198 
[lt6911_probe] -201 
data:ffffffff ffffffff
[lt6911_probe] jump return CVI_FAILURE -206 
[... 139 lines cut ...]
VIInImgWidth            :55596
VIInImgHeight           :  20
VIDevFPS                :   0
VIFPS                   :   0
VISofCh0Cnt             :   0
VIPreFECh0Cnt           :   0
VIPreBECh0Cnt           :   0
VIPostCnt               :   0
VIDropCnt               :   0
VIDumpCnt               :   0
[VI ISP_PIPE_A Csi_Dbg_Info]
VICsiIntStatus0         :0x0
VICsiIntStatus1         :0x0
VICsiCh0Dbg             :0x0
VICsiCh1Dbg             :0x0
VICsiOverFlowCnt        :   0
VICsiCh0WidthGTCnt      :   0
VICsiCh0WidthLSCnt      :   0
VICsiCh0HeightGTCnt     :   0
VICsiCh0HeightLSCnt     :   0

```

This issue and the fact that keyboard cannot be used during boot (https://github.com/sipeed/NanoKVM/issues/25) , makes the device totally useless as a KVM.
Given the slow pace of any fixes in the project, I don't expect to be usable any time soon. I guess PiKVM purchase is inevitable after all ...

ZenoBell, 2024-11-05:
I just received this, and I'm experiencing the same issue. Could the team prioritize this and consider releasing a hotfix as soon as possible?

wj-xiao, 2024-11-08:
Please update to version 2.1.0 in the webpage. We have added support for H.264.

marshalleq, 2024-11-08:
Are you saying that updating to H.264 fixes the issue?

wj-xiao, 2024-11-08:
@marshalleq Yes. We have changed the method for obtaining HDMI output (using CGO). So this version will be more stable.

This version will fix some legacy hdmi issues. For example, a black screen after restarting the service, or black screen after running for a while.

Z2Z-GuGu, 2024-11-08:
@marshalleq Thank you for your attention! In the recent update, we have changed the original architecture and attempted to fix the issue of ion memory leakage while refactoring. At least for now, ion will not easily explode We will continue to improve the stability of the new architecture and look forward to your feedback.

pabloalcantara, 2024-11-08:
Now I have a new problem with the 2.1.0 versions:
Green screen
[image]

wj-xiao, 2024-11-08:
@pabloalcantara What is the remote host's resolution? Some resolutions are not supported. Try to use 1080p or 720p.

Or it may be a compatibility issue with h264. Try to use MJPEG mode to check whether the screen is normal.

pabloalcantara, 2024-11-08:
It´s a linux shell...
Changing to mjpeg resolves it.

> @pabloalcantara What is the remote host's resolution? Some resolutions are not supported. Try to use 1080p or 720p.
> 
> Or it may be a compatibility issue with h264. Try to use MJPEG mode to check whether the screen is normal.

wj-xiao, 2024-11-08:
@pabloalcantara So it should be a compatibility issue with h264.
What browser are you using? Can you try chrome or another browser?

pabloalcantara, 2024-11-08:
Using edge

wj-xiao, 2024-11-08:
> Using edge

So there may be compatibility issues with Edge.
I haven't tested on Edge and I will check it.

haggis-supper, 2024-11-09:
> Well, the x is gone from the display window, but for me, this still doesn't work after update. I think I will have to set up on another computer to keep testing. I didn't run the restart script above though, perhaps I need to do that as well.

I did see that once on my NanoKVM lite, i think i had disrupted the ribbon cable between the HDMI board and the main Lichee board.  Moving it a little bit resolved things again.

marshalleq, 2024-11-09:
I plugged into another computer and it worked.  So plugged it back into the original and that worked too.  Not sure why it did that, hopefully just a one off.  So this is great, I didn't have to re-flash or anything, just updated and done!  Thanks for fixing it @wj-xiao .

mcury1985, 2024-11-13:
Just reflashed with 1.20 firmware, updated to 2.11.0 and tried h264, for me it doesn't work.
There is a black square in the middle of the screen, changing to MJPEG fixes the issue.
Can't find any relevant logs in /var/log/messages. /etc/init.d/S95nanokvm restart_stream didn't help.
Tried Chrome and Firefox, with cookies cleared, 1080 and 720p, no joy.

sroemer, 2024-11-24:
I today received my NanoKVM light and wasn't able to get a picture of the screen to be shown. After checking the ribbon cable with the macro mode of my phones camera I am pretty sure that the cable is not connected properly on the side of the addon board. Now need to figure out how to get that fixed.

haggis-supper, 2024-11-24:
> I today received my NanoKVM light and wasn't able to get a picture of the screen to be shown. After checking the ribbon cable with the macro mode of my phones camera I am pretty sure that the cable [...]

Have you pushed the latest image onto your NanoKVM and updated to the latest application version? If you check under About KVM you should see these versions ( this is what i am running on my lite just now )
.
Image Versionv1.2.0
Application Version2.1.1

I had earlier versions of software on mine and it was unpredictable. I did reseat the hdmi capture board cable as well, but i'm certain it wasnt that now.

sroemer, 2024-11-25:
> Have you pushed the latest image onto your NanoKVM and updated to the latest application version? If you check under About KVM you should see these versions ( this is what i am running on my lite [...]

Thanks for the feedback, that gives me at least some hope that the issue is not hardware related, but I already tried the latest release version 1.3.0, 1.2.1 and now - since you mentioned specifically this version - also 1.2.0.

So far none of it worked, but the reason might be the Application version. With none of those images I was able to perform an update yet. Checking for an update in the GUI just fails with the message "GetVersion failed". I logged in via SSH and checked that this is not a network issue already.

Is there a way to perform this Application update manually?

sroemer, 2024-11-25:
```
I think I discovered the possible problem USB power. My device is the light one, so is using power from the USB, maybe, only maybe, the USB can´t delivere the necessary power.
```
This might have been the problem in my case too. I reflashed with nearly all available images from 1.1.0 to the latest 1.3.0 and none worked. Now I powered the NanoKVM from an USB-C power supply I had for an Raspberry Pi and after that I was able to perform the Update and now even see an image.
But I am also not 100% sure about that, since after using the power supply I still got the "GetVersion failed" error a few times.

Anyway I am happy that it isn't a real hardware issue, but disappointed with the software so far - but there is some potential I think.

marshalleq, 2024-12-05:
Warning don't install the latest update or the problem is back!  @!&^*^&!  I don't know what is up with these guys and their testing, but clearly the professionalism is yet to be installed.  Basic features of a product should be tested before deployment.  So frustrating, I literally hovered over the update and questioned it, should have listened to my gut.  Grrr.

Z2Z-GuGu, 2024-12-05:
@marshalleq I am very curious about what kind of host device you are using. We conduct detailed testing on each version of hardware and host before releasing updates, but unfortunately, we cannot test every host. If possible, please execute the `cat /proc/cvitek/vi_dbg` command to check if the VI part of the chip is working properly

marshalleq, 2024-12-05:
It's just a Threadripper 2950X with a P2000 card in it, Asus X399 motherboard.  When I first got the KVM though, I had a Lenovo P700 with same GPU.  In my frustration yesterday (and apologies for the vent above), I should really have mentioned it's not exactly the same error.  Same result, just not same error.  The web service appears not to be running at all, though the IP is still alive.  Reboots don't fix it.  So technically it's a new bug.  I will see if I can ssh into it once I figure out what that username password combo is.

duffysk, 2024-12-12:
@Z2Z-GuGu I'm also having issues with accessing WebUI after upgrading to 1.3.0.
When I ssh (it works) I'm seeing a lot of:
`cat: can't open '/proc/cvitek/vi_dbg': No such file or directory` 

when trying
`/etc/init.d/S95nanokvm restart`

I'm getting this: 

```
# /etc/init.d/S95nanokvm restart
killall: NanoKVM-Server: no process killed
OK
# killall: NanoKVM-Server: no process killed
rm: can't remove '/tmp/server': No such file or directory
beta
oled exist
192.168.1.18
[SAMPLE_COMM_SNS_ParseIni]-2204: Parse /mnt/data/sensor_cfg.ini
[parse_source_devnum]-1797: devNum =  1
[parse_sensor_name]-1878: sensor =  LONTIUM_LT6911_2M_60FPS_8BIT
[parse_sensor_busid]-1907: bus_id =  4
[parse_sensor_i2caddr]-1918: sns_i2c_addr =  ff
[parse_sensor_mipidev]-1929: mipi_dev =  0
[parse_sensor_laneid]-1940: Lane_id =  2, 4, 3, 1, 0
[parse_sensor_pnswap]-1951: pn_swap =  0, 0, 0, 0, 0
[parse_sensor_mclken]-1973: mclk_en =  0
[parse_sensor_mclk]-1984: mclk =  0
try release sys ok
[_mmf_init]-1142: maix multi-media version:67a90e2-musl_riscv64
[SAMPLE_COMM_SNS_ParseIni]-2204: Parse /mnt/data/sensor_cfg.ini
[parse_source_devnum]-1797: devNum =  1
[parse_sensor_name]-1878: sensor =  LONTIUM_LT6911_2M_60FPS_8BIT
[parse_sensor_busid]-1907: bus_id =  4
[parse_sensor_i2caddr]-1918: sns_i2c_addr =  ff
[parse_sensor_mipidev]-1929: mipi_dev =  0
[parse_sensor_laneid]-1940: Lane_id =  2, 4, 3, 1, 0
[parse_sensor_pnswap]-1951: pn_swap =  0, 0, 0, 0, 0
[parse_sensor_mclken]-1973: mclk_en =  0
[parse_sensor_mclk]-1984: mclk =  0
pstSize->u32Width = 1920
pstSize->u32Height = 1080
alloc_buf_size(294912): 294912, phy_addr(8ace8000): 0x8ace8000, buffer_name: ISP_SHARED_BUFFER_0
ion_data.size:294912, ion_data.addr_p:0x8ace8000, ion_data.name:ISP_SHARED_BUFFER_0
Cannot open '/dev/cvi-vi': 2, No such file or directory
VI open failed
: No such file or directory
Cannot open '/dev/cvi-vpss': 2, No such file or directory
VPSS open fail
: No such file or directory
Cannot open '/dev/cvi-vo': 2, No such file or directory
Cannot open '/dev/cvi-dwa': 2, No such file or directory
dwa open failed: No such file or directory
Cannot open '/dev/cvi-rgn': 2, No such file or directory
RGN open failed
: No such file or directory
SYS_IOC_S_CTRL - sys_set_vivpssmode NG
SYS_IOC_S_CTRL
: Invalid argument
SYS_IOC_S_CTRL - sys_set_vpssmodeex NG
SYS_IOC_S_CTRL
: Invalid argument
pstSize->u32Width = 1920
pstSize->u32Height = 1080
pstSize->u32Width = 1920
pstSize->u32Height = 1080
ISP Vipipe(0) Allocate pa(0x8ace8000) va(0x0x3fbbf56000) size(291424)
pstSize->u32Width = 1920
pstSize->u32Height = 1080
stSnsrMode.u16Width 1920 stSnsrMode.u16Height 1080 60.000000 wdrMode 0 pstSnsObj 0x3fbc6792e8
pstSize->u32Width = 1920
pstSize->u32Height = 1080
pstSize->u32Width = 1920
pstSize->u32Height = 1080
Cannot open '/dev/cvi-vi': 2, No such file or directory
VI open failed
: No such file or directory
[_mmf_init]-1200: vi init failed. s32Ret: 0xc00e800e !
[_mmf_init]-1201: Please try to check if the camera is working.
maix multi-media init failed
cat: can't open '/proc/cvitek/vi_dbg': No such file or directory
```

silajim, 2024-12-15:
I am also having this issue, here is the vi_debug 

 cat /proc/cvitek/vi_dbg
[VI BE_Dbg_Info]
VIPreBEDoneSts          :0x0            VIPreBEDmaIdleStatus    :0x0
[VI Post_Dbg_Info]
VIIspTopStatus          :0x0
[VI DMA_Dbg_Info]
VIWdma0ErrStatus        :0x0    VIWdma0IdleStatus       :0x0
VIWdma1ErrStatus        :0x0    VIWdma1IdleStatus       :0x0
VIRdmaErrStatus         :0x0    VIRdmaIdleStatus        :0x0
[VI ISP_PIPE_A FE_Dbg_Info]
VIPreFERawDbgSts        :0x0            VIPreFEDbgInfo          :0x0
[VI ISP_PIPE_A]
VIOutImgWidth           :1920
VIOutImgHeight          :1080
VIInImgWidth            :1920
VIInImgHeight           :1080
VIDevFPS                :   0
VIFPS                   :   0
VISofCh0Cnt             :   0
VIPreFECh0Cnt           :   0
VIPreBECh0Cnt           :   0
VIPostCnt               :   0
VIDropCnt               :   0
VIDumpCnt               :   0
[VI ISP_PIPE_A Csi_Dbg_Info]
VICsiIntStatus0         :0x0
VICsiIntStatus1         :0x0
VICsiCh0Dbg             :0x0
VICsiCh1Dbg             :0x0
VICsiOverFlowCnt        :   0
VICsiCh0WidthGTCnt      :   0
VICsiCh0WidthLSCnt      :   0
VICsiCh0HeightGTCnt     :   0
VICsiCh0HeightLSCnt     :   0

duffysk, 2024-12-16:
@silajim so what I managed to do so far is to bring it back.
1) I removed SD card
2) flashed 1.2.1 image
3) ran python upgrade script https://github.com/Bengt/NanoKVM-Updater/blob/main/nanokvm-updater.py
4) it's up an running

LED lights icon not available in WebUI (not sure if it's feature or bug though).
I'm planning to flash 1.3. image later.

**Learning:** case seem to be 3D printed, one screw simply does not hold after mounting case back. Not sure if @Z2Z-GuGu  is Sipeed representative, if so, I'd recommend to improve case quality or make SD card accessible so that no case dismantling is needed.

Z2Z-GuGu, 2024-12-16:
> **Learning:** case seem to be 3D printed, one screw simply does not hold after mounting case back. Not sure if @Z2Z-GuGu is Sipeed representative, if so, I'd recommend to improve case quality or make SD card accessible so that no case dismantling is needed.
We have now started producing the injection-molded version of the case, which will have better quality. If you prefer not to disassemble it, please refer to [this](https://wiki.sipeed.com/hardware/en/kvm/NanoKVM/system/flashing.html#USB-Update-TF-Card-Image) link to complete the programming.
I've recently received quite a few inquiries from non-Chinese users, likely due to DNS issues that prevent a complete update. I will do my best to create a 1.3.1 image this week to address the recently mentioned problems.

Z2Z-GuGu, 2024-12-16:
> I am also having this issue, here is the vi_debug

It seems that the HDMI is not recognized at all. Please open the webpage, then unplug and replug the HDMI. If the HDMI is detecting an image, VIDevFPS will not be '0'.

duffysk, 2024-12-16:
@Z2Z-GuGu Thanks!

How can I get case replacement as part of warranty? The reason I'm asking is that as my screw does not hold in, heatsink is not properly attached.

Z2Z-GuGu, 2024-12-16:
@marshalleq 
If you haven't set a new username and password, it should be root/root. After logging in, try executing 
```shell
python /etc/kvm/update-nanokvm.py
```
 If you encounter an error, maybe due to DNS issues, please add the following line to `/etc/resolv.conf`:
```shell
nameserver 119.29.29.29
nameserver 223.5.5.5
```
Then try again.

[11 more replies]