sipeed/NanoKVM #87: No Keyoard or Mouse

haggis-supper, 2024-10-07:
Image version 1.0.0 running Application version 2.0.9.

Connected to a non graphical linux host i can see the console on screen and login prompt but keyboard and mouse are not being seen.
If i log into the host and check USB devices it is not showing any keyboard or mice.

This has worked with previous application versions.

patschi, 2024-10-08:
Under the keyboard icon there is "Reset HID device". If you press this, does that somehow help solving it?

haggis-supper, 2024-10-08:
Alas no, resetting the hid did nothing. I've even moved the usb plug to a different usb port, Linux messages does not show any new usb connections.

RDunmow, 2024-10-08:
I also have the same issue, using the paste option to send commands to the remote device works, but this is not great.

haggis-supper, 2024-10-09:
ok, i have found a solution. I just got a shorter new USB-A ro USB-C cable and that is working fine. Keyboard and Mouse are working.

I bought the 0.5m version of this - [EBAY ](https://www.ebay.co.uk/itm/266341622032?var=566163822382)

haggis-supper, 2024-10-09:
> @haggis-supper And what cables were you using so far? The ones included with NanoKVM? Cause I'm also having issues with mouse/keyboard :/

I have the NanoKVM Lite, so you done get any cables with it. 
Just the barebones board that you need to add cables and an SD card too. 
I have 3d designed my own case for it [Thingiverse](https://www.thingiverse.com/thing:6773907)

fenio, 2024-10-10:
Actually I found some different cable in my "maybe I will need it some day store" and it worked!
So seems that this device is pretty sensitive when it comes to cable quality / lenght.

nwgat, 2024-11-10:
using HDMI 2.0 and USB 3.0 cables might help, as they are generally better shielded and are backwards compatible with older specs

wilsonnkwan, 2024-12-18:
The quality USB cable comment works. But that said - Manufacturer, please at least sell your kits with the sufficiently quality cables. Thanks