sipeed/NanoKVM #88: Feature request - PC-USB and HDMI connections - enable on demand with a tick option.

mcury1985, 2024-10-07:
I noticed that when you connect the HDMI cable to the host, sometimes the host's screen doesn't work correctly, blank screen in the host.
Same problem applies to the keyboard when the host's keyboard won't work, the only keyboard that works is the KVM's keyboard, usually during boot (in BIOS or LUKs screen, or unlocking the opal drive in sedutil).

Since we have an option for HID reset, I think a good option with a tick to enable or disable, for HDMI and PC-USB connections.
By having this, we would be able to connect to the KVM and use the (reset and power button), and enable the PC-USB and HDMI only if required.
This would also avoid a problem with the host's OS because it detects two monitors by default.

Thanks in advance, great product you guys have here.

Z2Z-GuGu, 2024-10-14:
Thank you for your feedback. Due to the hardware design of the NanoKVM Cube, the function of disconnecting HDMI cannot be achieved. This function will be provided in the future NanoKVM PCIe. In addition, we have not encountered any problem of the keyboard outside the NanoKVM keyboard being unusable in our testing. If you encounter such a problem, you can use `/etc/init.d/S03usbdev stop` to turn off USB and `/etc/init.d/S03usbdev start` to turn on USB. If this approach is effective, please let us know and we will consider providing such a switch in future versions.

yawor, 2025-01-10:
> Due to the hardware design of the NanoKVM Cube, the function of disconnecting HDMI cannot be achieved. This function will be provided in the future NanoKVM PCIe.

How do I use this feature on NanoKVM PCIe? I've received one today and I'd like to use on-demand HDMI, but I don't see any option to do that in the UI (upgraded to 2.1.4). I'd like to have HDMI disabled by default, so the PC doesn't see another monitor connected unless I connect to the UI or even enable it in the UI.

carlossless, 2025-02-10:
> Thank you for your feedback. Due to the hardware design of the NanoKVM Cube, the function of disconnecting HDMI cannot be achieved. This function will be provided in the future NanoKVM PCIe.

@Z2Z-GuGu is there a projected timeline for when this feature will be introduced? I'm very much looking forward to it.

dormancygrace, 2026-09-14:
Hi! [NanoKVM OS](https://github.com/dormancygrace/NanoKVM-OS), my independent community firmware for SG2002-based NanoKVM Cube and PCIe, includes functionality relevant to this request.

The Web UI has an overall USB enable/disable switch, and the PCIe version also has HDMI enable/disable controls. The USB switch applies an empty or enabled gadget composition; the PCIe HDMI path controls the receiver through the native capture service.

The HDMI control described here is specific to PCIe. I have not verified that it reproduces physical cable removal on your host or resolves the BIOS, local-keyboard and monitor-enumeration symptoms in this thread.

[Implementation 1](https://github.com/dormancygrace/NanoKVM-OS/blob/9f65f9554142db4a61ba866b4498363f4f21d711/web/src/pages/desktop/menu/settings/usb/index.tsx#L109) · [Implementation 2](https://github.com/dormancygrace/NanoKVM-OS/blob/9f65f9554142db4a61ba866b4498363f4f21d711/server/service/vm/hdmi.go#L45).

If this addresses your use case, you're welcome to try it and let me know how it works on your setup.

It's currently in beta. Development testing has been on PCIe/UXC; Cube testing is still pending. Installing from stock firmware requires flashing a full SD image, so please back up your card first.

[Latest release](https://github.com/dormancygrace/NanoKVM-OS/releases/latest) · [Installation guide](https://github.com/dormancygrace/NanoKVM-OS/blob/main/docs/INSTALL.md)

Please include your board revision and reproduction steps when reporting results. Thanks!