sipeed/NanoKVM #90: API Documentation

johncoles, 2024-10-08:
I see the code has now been shared so I may go through and do this myself, but it would be nice to have documentation for the HTTP API used.

I note so far that there is a request to `/api/auth/login` to perform the login, but the password is not sent as plaintext.  Subsequent requests to the API seem to use a token that is set as a cookie.

Other endpoints include:
GET `/api/vm/gpio/led` which returns the LED state:
`{"code":0,"msg":"success","data":{"pwr":false,"hdd":false}}`

GET `/api/storage/images/mounted`
`{"code":0,"msg":"success","data":{"file":""}}`

POST `/api/vm/gpio` with data like: `{"type":"power","duration":800}` will press the power button for 800ms.
Returns:
`{"code":0,"msg":"success","data":null}`

It would be great to have docs on how to use this to integrate in to other systems (I would like to add an integration in to Home Assistant for example) to control multiple machines.

wj-xiao, 2024-10-10:
Please refer to [these files](https://github.com/sipeed/NanoKVM/tree/main/server/proto) for now. A more detailed document will be provided later.

patschi, 2024-10-10:
I took some time reverse-engineering API calls and crafting a bash-curl-way of interacting with the API. Pretty surely also easy to adapt in other programming languages. 

Note: Some API paths seems to mismatch between "reality" (NanoKVM with latest 2.0.9 as of today) and today's available code on GitHub: For example, `/storage/image/mounted` fails, while `/storage/images/mounted` works. I believe the code is newer than most recent publicly available firmware.

I thought it might be useful for others, so here we go:

## AUTH
**Note:**
- I decided to go with `jq` to use it for urlencode and prettify JSON output. It has been proofen way easier than using just curl.
- The password secret key defaults to `nanokvm-sipeed-2024` and must be identical between frontend and backend (see [encrypt.go](https://github.com/sipeed/NanoKVM/blob/726cd915cc2a52347b775499f4935ff301c15df1/server/utils/encrypt.go#L9))
- If this looks painful complicated, then you can imagine how it felt figuring out below.

```bash
# Login to NanoKVM and receive token
HOSTNAME="192.168.0.1"
USERNAME="admin"
PASSWORD="admin"
PWSECKEY="nanokvm-sipeed-2024"

PASSENC=$(echo -n "$PASSWORD" | openssl enc -aes-256-cbc -base64 -salt -md md5 -pass pass:$PWSECKEY 2>/dev/null)
AUTHJSON=$(jq -crnM --arg u "$USERNAME" --arg p "$PASSENC" '{"username":$u|@uri,"password":$p|@uri}')
REQUEST=$(curl -s -X POST http://$HOSTNAME/api/auth/login -H 'Content-Type: application/json' --data-raw "$AUTHJSON")

TOKEN=$(echo $REQUEST | jq -r .data.token)
COOKIE="nano-kvm-token=$TOKEN"
```

After this, below commands can be used.

## GENERAL (named "VM" in code)

### Get device info

```bash
# curl -s -b $COOKIE http://$HOSTNAME/api/vm/info; echo
{"code":0,"msg":"success","data":{"ip":"<IP>","mdns":"kvm-<LAST 4 DIGITS OF MAC>.local","image":"2024-08-17-18-13-713161.img","firmware":"2.0.9","deviceKey":"<ID>"}}
```

### Get GPIO LED state

```bash
# curl -s -b $COOKIE http://$HOSTNAME/api/vm/gpio/led
{"code":0,"msg":"success","data":{"pwr":true,"hdd":false}}
```

### Trigger power switch
**Note:**
- `800` is probably 0.8s and is considered a short click, e.g. turning on or sending shutdown request to host. 
- `8000` is considered a long click, e.g. holding power button and letting the host crash.

```bash
# curl -s -X POST -b $COOKIE http://$HOSTNAME/api/vm/gpio -H "Content-Type: application/json" -d '{"type":"power","duration":800}'
{"code":0,"msg":"success","data":null}
```

## STORAGE

### Get available images

```bash
# curl -s -b $COOKIE http://$HOSTNAME/api/storage/images | jq -r
{
  "code": 0,
  "msg": "success",
  "data": {
    "files": [
      "/data/ubuntu-24.04-desktop-amd64.iso"
    ]
  }
}
```

### Change mounted image
**Note:** Path is absolute path to ISO file

```bash
# curl -s -X POST -b $COOKIE http://$HOSTNAME/api/storage/image/mount -H "Content-Type: application/json" -d '{"file": "/data/ubuntu-24.04-desktop-amd64.iso"}'
{"code":0,"msg":"success","data":null}
```

### Get currently mounted image

```bash
# curl -s -b $COOKIE http://$HOSTNAME/api/storage/images/mounted
{"code":0,"msg":"success","data":{"file":"/data/ubuntu-24.04-desktop-amd64.iso"}}
```

## VIRTUAL DEVICES

### Get current virtual devices state

```bash
# curl -s -b $COOKIE http://$HOSTNAME/api/vm/device/virtual; echo
{"code":0,"msg":"success","data":{"network":true,"disk":true}}
```

### Set/Toggle state of virtual device
**Note:** Use `network` or `disk` for `device` key

```
$ curl -s -X POST -b $COOKIE http://$HOSTNAME/api/vm/device/virtual -H "Content-Type: application/json" -d '{"device": "network"}'
{"code":0,"msg":"success","data":{"on":true}}

$ curl -s -X POST -b $COOKIE http://$HOSTNAME/api/vm/device/virtual -H "Content-Type: application/json" -d '{"device": "network"}'
{"code":0,"msg":"success","data":{"on":false}}
```

PatrickHuetter, 2024-10-16:
Nice! Thanks for your work and sharing this!

FlorianHeigl, 2024-10-22:
kudos for the tesseract example, more people should figure out stuff like this and do it!

FlorianHeigl, 2024-10-22:
@wj-xiao Hi! is there a non-authenticated local API endpoint for scripts running on the device? Would be easier for sharing, unless someone says it's bad for security and I just didn't notice a problem there.

wj-xiao, 2024-11-08:
@FlorianHeigl By default, APIs require authentication.

However, you can disable authentication by modifying the configuration file.
Add one line `authentication: disable` to the configuration file `/etc/kvm/server.yaml`. And restart the service by `/etc/init.d/S95nanokvm restart`.

Please note that this will make the device insecure.

adamponi, 2024-11-22:
Is there an API that will allow me to click? Also press keys

caffeine-mgn, 2025-04-08:
How can I perform a mouse click using an API?

Similarly, how can I press keyboard keys using an API?

kaspergrubbe, 2026-01-28:
This seems to be deprecated:

```
curl -s -b $COOKIE http://$HOSTNAME/api/vm/gpio/led
404 page not found
```

You can use this instead:

```
curl -s -b $COOKIE http://$HOSTNAME/api/vm/gpio
{"code":0,"msg":"success","data":{"pwr":false,"hdd":false}}
```