sipeed/NanoKVM #91: Each keyboard key in the bios settings is pressed twice

Igor-M-77, 2024-10-08:
Hello. In the BIOS of a computer on the GIGABYTE GA-J1800N-D2H motherboard, when connected using nano KVM, each keyboard key is pressed twice. Any key, including arrow keys. The problem is reproduced when using any browser and keyboard, including virtual keyboards. I updated the BIOS to the latest version, but the problem persists. Various BIOS settings do not help.
I have two NANO KVM FULL devices and two computers - one based on the GIGABYTE GA-J1800N-D2H motherboard, the second based on Zotac. Both KVM devices do not work correctly in the Gigabyte-based bios settings and both work correctly on the basis of Zotac. The problem is clearly incompatibility with GIGABYTE GA-J1800N-D2H. I understand that the board is old and uninteresting, but it is not a fact that a similar problem will not arise for more modern ones. I repeat that in the bios settings, each key is pressed twice, which is why I cannot correctly select the settings. At the same time, there is no problem when working in the operating system itself.

gustavolaureano, 2024-11-26:
Pinging here, I also have this exact same issue with the BIOS on all 2 of my hardwares, any possible solutions?

aptalca, 2024-12-04:
I just received my device today and experiencing the same thing in the old school bios of the older Thinkstation board.

In windows, keyboard works just fine.

chronically-late, 2025-04-19:
Ran into the same issue with a Cisco UCS C220 M3 server.