sipeed/NanoKVM #101: Black shows as grey in the web UI (no HDMI full color range?)

p4block, 2024-10-10:
I'm pretty sure the capture chip's EDID isn't signaling to the PC that it can use the full color range. 
This is often a setting called "HDMI Deep color" on the TV and sometimes the client (like a PS3 or other console)

It may not actually support this, just throwing around the idea. 

[image]
up top "black" from tty shown as gray, below true black in the webapp.

mdmitry, 2025-01-17:
Looks like color range is converted from range 0..255 to 16..235, and when MJPEG is used it is done two times, resulting in 30..218 range.

Here's how 256 different grays look when H.264 is used:

[image]

The same when MJPEG is used: (you can measure values, they will range from 30 to 218)

[image]

Here's a simple RGB image compared when viewed locally versus remote MJPEG:

[image]

All this is happening while HDMI is outputing RGB, and NOT YCbCr:

[image]

mdmitry, 2025-01-17:
Update: when I switch HDMI mode to YCbCr, H.264 colors become perfect, and MJPEG colors become 16..235 range.