sipeed/NanoKVM #98: Virtual network from LAN not passed through

kcastner, 2024-10-10:
The network provided via LAN wont be able to be provided to administrated host via virtual Network.

Terminal output for ``ìp a```
[image]

GUI
[image]

Re-enable wont fix this problem. What should look like the wireing for the virtual network use?

Windows and Linux shows "not connected"

Z2Z-GuGu, 2024-10-14:
Hi, virtual network refers to USB RNDIS. You can refer to [here](https://wiki.sipeed.com/hardware/en/kvm/NanoKVM/user_guide.html#RNDIS) for usage instructions

vldm, 2024-11-15:
@kcastner If i understand correctly RNDIS is mainly used for connecting to NanoKVM ssh from machine that you control.

If you want to bridge usb network to the eth, you can use one of the scripts below:

1. `nat-iptables` - if you just need access to the internet.
```sh
echo 1 > /proc/sys/net/ipv4/ip_forward
iptables -t nat -A POSTROUTING -o eth0 -j MASQUERADE
iptables -A INPUT -i usb0 -j ACCEPT
iptables -A INPUT -i eth0 -m state --state ESTABLISHED,RELATED -j ACCEPT
iptables -A OUTPUT -j ACCEPT
```
2. "pseudo-bridge-nat" - if you want to access server from local network trough NanoKVM

Replace 192.168.55.55 - with IP in your network, and check that RNDIS client is 10.54.72.100
```sh
echo 1 > /proc/sys/net/ipv4/ip_forward
ip addr add 192.168.55.55/24 dev eth0
iptables -t nat -A POSTROUTING -s 10.54.72.100 -o eth0 -j SNAT --to-source 192.168.55.55
iptables -t nat -A PREROUTING -d 192.168.55.55 -j DNAT --to-destination 10.54.72.100
# Allow incoming packets from eth0 to usb0
iptables -A FORWARD -i eth0 -d 10.54.72.100 -j ACCEPT
# Allow outgoing packets from usb0 to eth0
iptables -A FORWARD -o eth0 -s 10.54.72.100 -j ACCEPT
```

The network speed for me is around 50-80 Mbps

kcastner, 2024-11-15:
@vldm that Looks great. Thats what i looked for. Thank you!

@Z2Z-GuGu the description of the virtual network looks like sharing network of the NanoKVM. There was a misunderstanding regarding the description. Maybe this should be clarified here. What do you think?

ARI-DevOps, 2025-02-07:
Does this allow you to have both LAN and internet access all through the KVM bridge? I've been trying to set this up the way you have on the 2nd script and can't figure out how to get it working. I'm replacing _**192.168.55.55**_ with an IP on my network not in use or in the DHCP pool. Then replacing your _**10.54.72.100**_ with the IP I see from the usbnic in the output of "ip a". (This is 10.5.104.1/24). I'm on Windows11 using the Acer USB Ethernet/RNDIS Gadget driver from 1/13/2010. 
Any help is appreciated. Would love to remove duplicate ethernet cables if this allows the internet to go through the KVM.