sipeed/NanoKVM #99: Feature Request: WireGuard

patschi, 2024-10-10:
It would be great to have WireGuard implemented in UI and connection details saved in the configuration file.

nwgat, 2024-10-21:
love seeing this as well

lievin-christopher, 2025-01-15:
Hello,
You can paste your wireguard configuration file in /etc/wireguard/ and create a startup script in /etc/init.d/ (remove DNS line from configfile)
e.g : 
# cat /etc/wireguard/home.conf
```
[Interface]
PrivateKey = XXXXX
Address = YYYYY
MTU = 1420

[Peer]
PublicKey = XXXXY
AllowedIPs = YYYYX
Endpoint = XYXYX
PersistentKeepalive = 21
```
# cat /etc/init.d/S97wireguard

```
#!/bin/sh
DAEMON="wireguard"
PIDFILE="/var/run/$DAEMON.pid"

case "$1" in
	start)
		/usr/bin/wg-quick up home
                [ $? = 0 ] && echo "OK" || echo "FAIL"
                ;;
        stop)
                /usr/bin/wg-quick down home
                [ $? = 0 ] && echo "OK" || echo "FAIL"
                ;;
        restart|reload)
                $0 stop
                $0 start
                ;;
        *)
                echo "Usage: $0 {start|stop|restart}"
                exit 1
esac

exit 0

emmy008, 2025-03-06:
Followed your instructions, but getting an error:
stat: unrecognized option: c
BusyBox v1.36.1 () multi-call binary.

Usage: stat [-lt] FILE...

Display file status

        -L      Follow links
        -t      Terse display
stat: unrecognized option: c
BusyBox v1.36.1 () multi-call binary.

Usage: stat [-lt] FILE...

Display file status

        -L      Follow links
        -t      Terse display
/usr/bin/wg-quick: line 47: ((: ( &  & 0007) == 0: syntax error: operand expected (error token is "&  & 0007) == 0")

which stat returns /usr/bin/stat

lievin-christopher, 2025-03-06:
maybe do something like that :
> ln -sf /bin/true /bin/resolvconf

lievin-christopher, 2025-03-06:
and keep secret file in /etc/wireguard/ 
> chmod 600 /etc/wireguard/*.conf

emmy008, 2025-03-17:
@lievin-christopher, thank you for the tips, but for the life of me I cannot get it working. I can get use wg-quick up home, and I have no errors, but it just won't connect. I know the server side is good as I'm using other clients to connect to it.

Eszartek, 2025-04-18:
Instead of "ln -sf /bin/true /bin/resolvconf"

create a new file /bin/resolvconf containing:
```
#!/bin/sh

exit 0
```
and run "chmod 755 /bin/resolvconf" .

It's working fine for me, I just disregard the stat error.

dormancygrace, 2026-09-14:
Hi! I've included support for this in [NanoKVM OS](https://github.com/dormancygrace/NanoKVM-OS), my independent community firmware for SG2002-based NanoKVM Cube and PCIe.

WireGuard is managed from Settings → VPN, with persistent imported profiles, start/stop controls and a per-profile choice for installing AllowedIPs routes.

Route Allowed IPs is off by default, so enable it on a stopped profile if you need those routes.

[Implementation / details](https://github.com/dormancygrace/NanoKVM-OS/blob/9f65f9554142db4a61ba866b4498363f4f21d711/docs/VPN.md#wireguard).

If you're still experiencing this issue, you're welcome to try it and let me know whether it resolves the problem on your setup.

It's currently in beta. Development testing has been on PCIe/UXC; Cube testing is still pending. Installing from stock firmware requires flashing a full SD image, so please back up your card first.

[Latest release](https://github.com/dormancygrace/NanoKVM-OS/releases/latest) · [Installation guide](https://github.com/dormancygrace/NanoKVM-OS/blob/main/docs/INSTALL.md)

Please include your board revision and reproduction steps when reporting results. Thanks!