sipeed/NanoKVM #102: Adding OLED to nanoKVM Lite

haggis-supper, 2024-10-11:
I would like to be able to add the i2c OLED to the NanoKVM Lite.

Is there any information available out what you need, for example

- i2c address
- Size of OLED
- Where to get the power from ( i can see the i2c channels and GND )

Z2Z-GuGu, 2024-10-14:
@haggis-supper Hi, this is feasible and does not require any software modifications. Here is the open source schematic of the Full version OLED board for your reference `https://cn.dl.sipeed.com/shareURL/KVM/nanoKVM/HDK/02_Schematic` 30113. The OLED size is 0.96 inches, which is a quite common model.

haggis-supper, 2024-10-14:
Thank you for that pointer and information :)

SantyC93, 2024-10-14:
I can confirm that by moving the resistor the screen works correctly, but I agree it would be easier to change the config if possible.
I also have another question, is it possible to use an SPI screen instead of IIC?

Z2Z-GuGu, 2024-10-15:
> Is there any way to change the i2c address from 0x7a to 0x7b in code/config file. I can see a lot of the screens availble are 0x7b i2c address. i'd rather change code/config file than try to desolder and move the resitior on the back of the oled display.

Sorry, the different versions of our NanoKVM (Lite/Full/PCIe) identify different models by the address of the screen, so it is not convenient to customize the screen address

Z2Z-GuGu, 2024-10-15:
> I can confirm that by moving the resistor the screen works correctly, but I agree it would be easier to change the config if possible. I also have another question, is it possible to use an SPI screen instead of IIC?

NanoKVM uses LicheeRV Nano as the core, no extra SPI pins assigned to OLED (GPIO P port is assigned to wifi)
But then we will open source the OLED display part, you can customize your screen to a large extent

haggis-supper, 2024-10-18:
@SantyC93 Thanks for letting me know the OLED works, i've got  a couple and will modify the i2c address and fit it this weekend.

In the meantime i've made another NanoKVM Lite enclosure encorporating space for the OLED in the top if you need one .  I've added it to my existing 3d design for the non oled case here - [NanoKVM Lite Case](https://www.thingiverse.com/thing:6773907)

SantyC93, 2024-10-18:
@haggis-supper Thank you very much for sharing! I was planning on making an enclosure myself, you just saved me a lot of time :D

damtrx, 2024-11-07:
> @SantyC93 Thanks for letting me know the OLED works, i've got a couple and will modify the i2c address and fit it this weekend.
>
> In the meantime i've made another NanoKVM Lite enclosure [...]

Hi ,
I'm stuck with the display too. Were you able to get one with 0x7a address ? All mine have got different address (0x3c) or similar. 
It is a shame that the i2c address is hardcoded.

haggis-supper, 2024-11-07:
Nope, i still need to desolder/solder the connections on the back of my screen to swap it over.  
Kind of reverted back to pikvm's just now as the nanoKVM is just not doing what i want.

damtrx, 2024-11-07:
> Nope, i still need to desolder/solder the connections on the back of my screen to swap it over. Kind of reverted back to pikvm's just now as the nanoKVM is just not doing what i want.

Got it.
The ones from aliexpress that I got are using different address and can't be changed. 
I just wonder why there is no search function in the code like every other arduino code that I used before.

Using hard codes address is something that is basically poor coding for me. I'm not buying this as excuse to distinguish the platforms

SantyC93, 2024-11-07:
> Hi , I'm stuck with the display too. Were you able to get one with 0x7a address ? All mine have got different address (0x3c) or similar. It is a shame that the i2c address is hardcoded.

Hi, all my SSD1306 and SSD1315 displays have a very small resistor in the back (right under the pins) that let you switch the address, usually on mine the default is 0x78 and I can change it to 0x7A.
None of my displays have 0x3c so maybe you have a different model, I don't know if it could have a way to change the address or not.
I can confirm tho that on SSD1306 and SSD1315 changing the resistor to 0x7A makes them work with the NanoKVM Lite.
I agree it's kinda inconvenient but apparently the should open source the OLED display part so maybe this will change.

WaldperlachFabi, 2024-11-16:
> Hi, all my SSD1306 and SSD1315 displays have a very small resistor in the back (right under the pins) that let you switch the address, usually on mine the default is 0x78 and I can change it to [...]

Thank you for that information! Can you tell me which pins on the nanokvm Lite I need to connect to SDA / SCL on the Display? Do you have a picture for me? Thank you very much!

SantyC93, 2024-11-16:
> Thank you for that information! Can you tell me which pins on the nanokvm Lite I need to connect to SDA / SCL on the Display? Do you have a picture for me? Thank you very much!

Should be GPIO P19 for SCL and P21 for SDA :)
https://cn.dl.sipeed.com/shareURL/KVM/nanoKVM/HDK/02_Schematic

trothe, 2025-02-01:
This issue has been very helpful. The schematic confused me a bit. I finally made it work using **GPIO15** for **SCL** and **GPIO27** for **SDA** and 3.3V. My display has a little 0R resistor on the back that I needed to solder to **0x7A**.

[image]

deftdawg, 2025-02-03:
I was inspired and had a go, though I quickly realized my tools and skills aren't good enough for this tiny work, I'd imagine a hot air station would've make short work of this... the resistor didn't make it with my ironing "technique"... anyway off to order a replacement resistor and another LED display from Ali in case I've toasted this one. 😅

[image]

trothe, 2025-02-03:
@deftdawg sorry to hear about your troubles. I should have given better instructions. When soldering these, I use a wide tip that touches both solder points at the same time, along with a pointed stick or tweezers to hold things in place.

In my case, the resistor was marked ‘0,’ meaning it’s a 0-ohm resistor, which effectively has no resistance. It can be replaced by a simple piece of wire. PCBs often use these 0-ohm resistors as jumpers to bridge connections.

It might be different for your display, so I don’t want to give you the wrong advice.

trothe, 2025-02-03:
@deftdawg glad you managed to make it work. I think the 2 color display looks better than the original. The big question is now how to package this nicely. A case would need to include a way to screw mount this. Maybe even on the side?

WaldperlachFabi, 2025-02-03:
I designed a case with a display to 3D print. I’m gonna upload it to cults and share the link with you! There are already a few cases but no one worked with the OLED I used(0.96) one.

SergeyCr, 2025-03-28:
So the address on the bus should be 0x7A - that's what almost all highly avalilable cheap 0.96" oled modules with 4 contacts (gnd-vcc-sda-scl) support. You really should go this way. All the soldering work you should do if you buy one of these is move one resistor which is selecting address to another set of pads, and that is usually marked on the board. Then this strange pin mapping of GPIO15 for SCL and GPIO27 for SDA works. BTW you can solder VCC to 5V - these boards usually incorporate sot23-3 type "662k" 3-pin 3v3 buck linear regulator, so no need to further downvoltage screen ciruitry unless you want lesser consumption and brightness.
BUT - there are other oled 0.96 boards which are like with 7 pins that are more universal, can select between 3 pin spi, 4 pin spi and iic interfaces, but have no solder pads for an address change resistor. However, this pin is routed to the one of external connected ones, so it could be easily bodgewired to 3v3 to select the address. Likely that should work, but still no luck running these as for 4-pin schematics using data pins...

haggis-supper, 2025-06-05:
Check out this page, it has my 3d enclosure for an oled screen but i also show you pictures of my wiring along with a description of what wire goes where..

https://www.thingiverse.com/thing:7014568

deftdawg, 2025-06-06:
> Check out this page, it has my 3d enclosure for an oled screen but i also show you pictures of my wiring along with a description of what wire goes where..
> 
> [thingiverse.com/thing:7014568](https://www.thingiverse.com/thing:7014568)

Thanks, I made some tweaks to the lid in Bambu Studio (increase size by 0.2mm, add 4 posts 22mm apart for locating screen (dab of hot glue on each post makes it stick well) and cut a channel for the screen ribbon), the nanokvm board floats a bit in the bottom of the case.   Bottom has no modifications, just included as part of the 3MF because needed to print it, 3MF project attached.

[image]

[image]

[nanoKVM_case_led-3mf.tar.gz](https://github.com/user-attachments/files/20630176/nanoKVM_case_led-3mf.tar.gz)

trothe, 2025-06-06:
@haggis-supper  @deftdawg Thank you both for working on an enclosure. That's really great! I printed the initial version and one issue I have is that the lid won't stay on. I was to lazy to reprint it. But I think it would be cool to extend the 2 things on the side so they can be secured by a small screw. Maybe like this:

[image]

It would also be great if the other 2 "taps" would extend further to push down on the PCB. Mine is not really tight in place.... just a suggestion.

I wish the NanoKVM would support PoE to eliminate that usb cable....

zangaby, 2025-06-27:
I've extracted it using 7zip on Windows.