sipeed/NanoKVM #103: IP Not working on 1.2.0 & 1.2.1

BertQuodge, 2024-10-11:
Hi

I've just received 6 "full version" NanoKVMs. All but 1 are working.

On the problematic unit I'm unable to see any video output on the web interface, I just see a small X where the video output should be. The other NanoKVMs work correctly when connected to the same PC and cables etc. To try and resolve the issue I flashed v.1.2.0 onto the supplied microSD card. If I boot the nanoKVM, with 1.2.0, the led display just says "Couldn't get IP". The switch connected to the NanoKVM shows the port is active. I then flashed v.1.2.1 onto the microSD and the same error is displayed on the led display. I took a backup of the original microSD and if I install that IP works ok but, obviously, video does not.

I tired another microSD card, out of interest, but the issues are the same. I see the notes on v1.2.1 mentions "Fixed the eth cable detection issue", I don't know if this is related to my issue but neither 1.2.0 or 1.2.1 are working correctly, with regard to IP, on my unit. 

Thanks

kcastner, 2024-10-11:
Have you tried to set a static ip according to [this](https://wiki.sipeed.com/hardware/en/kvm/NanoKVM/network/static_ip.html)?

Does it make any difference?

Civil, 2024-10-11:
As far as I understand, latest firmware is 2.0.9 and your unit should probably be shipped with 2.0.3+ (I got 2.0.6 pre-installed on mine), but there are no images for those available.

Have you tried getting a dump of working FW and flashing that to non-working unit instead of using 1.2.1?

BertQuodge, 2024-10-11:
Hi

Thanks for the suggestion!

I mounted the sd card in a Linux box and set the NanoKVMs IP address manually, as per your note above. With firmware 1.2.1 I still receive the same "Couldn't get IP" address message on the NanoKVM's display. The odd thing is I don't see any other text on the display, no NanoKVM logo, or other info. The switch the NanoKVM is attached to says the Nano KVM port is up but it doesn't see any traffic, not even a MAC address. If I install the firmware the KVM unit shipped with DHCP works ok and I get an IP address.

I made some comparisons between a working KVM and the failed unit:

WorkingKVM:
If I power a kvm on with no ethernet cable attached the led displays normally, with a NanoKVM logo but with the IP displayed as --

BrokenKVM:
If I power a kvm on, with firmware 1.2.0 or 1.2.1, with no ethernet cable attached the led only says "Couldn't get IP" with no other information. The NanoKVM logo is also not displayed.

WorkingKVM:
If I power a kvm on with an ethernet cable attached but no DHCP server available the led displays normally, with a NanoKVM logo, but with the IP displayed as --

BrokenKVM:
If I power a kvm on, with firmware 1.2.0 or 1.2.1, with an ethernet cable attached but no DHCP server available the led only says "Couldn't get IP" with no other information. The NanoKVM logo is also not displayed.

WorkingKVM:
If I power a kvm on with an ethernet cable attached & a DHCP server available the led displays normally, with a NanoKVM logo and a DHCP assigned IP address displayed.

BrokenKVM:
If I power a kvm on, with firmware 1.2.0 or 1.2.1, with an ethernet cable attached & a DHCP server available the led says "Couldn't get IP" with no other information. The NanoKVM logo is also not displayed.

Now time to swap some bits around between working and non-working KVMs.

PreviouslyWorkingKVM Hardware:
if I take a freshly flashed v1.2.1 card and install into a previously working KVM I get the same "Couldn't get IP" with no other information. The NanoKVM logo is also not displayed.

PreviouslyBrokenKVM Hardware:
If I install the card from a previously working KVM (that has been updated to v1.2.0 via the NanoKVM built in installer) the unit boots correctly and obtains an IP. The original fault, where the web page just displays an X instead of video, remains.

PreviouslyBrokenKVM Hardware:
If I take an image, of the previously working KVM card (that has been updated to v1.2.0 via the NanoKVM built in installer), via dd, and put the image on another sandisk 32GB card, install in the broken KVM the KVM mostly works. I only see an X for video on the KVM's web page, but at least I get an IP and a functioning led display with IP address, kvm logo etc.

Now that I have a updated kvm I'll take a look at the HDMI issue.

This is far from scientific but I've ssh'd into both a working and the working but no video over web KVMs at the same time to compare to if there is a differences regarding HDMI hardware. Both units have 176 entries in /dev/, there is nothing much different in syslog either. i2cdetect also shows the same things listed.

If I connect a functioning HDMI device to a working kvm I see this in dmesg:

[266873.834325] LT6911--:-- INT!!##
[266874.009512] close channel no. 0
[266874.361004] vi_event_handler exit
[266874.612790] vpss_release
[266878.278080] LT6911--:-- INT!!##
[266878.322369] vb_get_pool_info:536():  NULL VB HANDLE
[266878.323420] vb_get_pool_info:536():  NULL VB HANDLE
[266878.324108] vpss_open
[266878.946379] open channel no. 0
[266878.946757] close channel no. 0
[266878.946823] open channel no. 0

However, if I connect the same HDMI device to the non-working kvm I only see this in dmesg:
[ 4745.566844] LT6911--:-- INT!!##
[ 4750.630931] LT6911--:-- INT!!##

I don't know what this means but something is different :-)

It feels like I have a broken HDMI input card on one of my KVMs. I could try swapping the HDMI input card with another KVM to prove, but I've run out of time today. I still think something isn't quite right with the online images though, or I'm doing something really silly! I've tried using both dd and Ubuntu disk imager tool, both yield the same results. I've tried the supplied 32GB card and a known good Sandisk 32GB card, again, with the same results.

Thanks

BertQuodge, 2024-10-11:
> As far as I understand, latest firmware is 2.0.9 and your unit should probably be shipped with 2.0.3+ (I got 2.0.6 pre-installed on mine), but there are no images for those available.
> 
> Have you tried getting a dump of working FW and flashing that to non-working unit instead of using 1.2.1?

Thanks for the suggestion. When I install v1.2.0 from GitHub the application version displayed on the web interface is 2.0.9.

Thanks

Z2Z-GuGu, 2024-10-14:
@BertQuodge Hello, based on your description, I believe it is a hardware malfunction. The pin connected to RJ45 may have detached from the solder pad during violent transportation. Could you please provide us with your address so that I can inquire if there is a possibility of returning it

BertQuodge, 2024-10-14:
> @BertQuodge Hello, based on your description, I believe it is a hardware malfunction. The pin connected to RJ45 may have detached from the solder pad during violent transportation. Could you please provide us with your address so that I can inquire if there is a possibility of returning it

Thanks for your help with this. I raised a question where I purchased the item and they are going to send a new unit out to me. I was asked to try and re-seat the HDMI ribbon cable and when I took the kvm apart I found the connector to be broken.