sipeed/NanoKVM #104: Feature Request: custo edid for keyboard and mouse

samascience, 2024-10-11:
need custom edid for keybaord and mouse

Z2Z-GuGu, 2024-10-14:
Thank you for your feedback. You can now modify the initialization parameters of the keyboard and mouse in `/etc/init.d/S03usbdev` (but it may be overwritten when updating the application). I will provide a more concise method in future updates

samascience, 2024-10-27:
@Z2Z-GuGu  - now I am curious how we can do it ! should we change the name at server/service/vm/virtual-device.go would that change the HID values ?

samascience, 2024-10-30:
@Z2Z-GuGu  - I am unable to find the file /etc/init.d/S03usbdev !!

samascience, 2025-02-28:
`start_usb_dev(){
    . /etc/profile
    echo "usb mode: device"
    cd /sys/kernel/config/usb_gadget

    mkdir g0
    echo 0x1009 > idProduct
    mkdir strings/0x409

    mkdir configs/c.1
    echo 0xE0 > configs/c.1/bmAttributes
    echo 120 > configs/c.1/MaxPower
    mkdir configs/c.1/strings/0x409
    echo "NanoKVM" >  configs/c.1/strings/0x409/configuration

    if [ -e /boot/usb.ncm ]
    then
        if [ -e /boot/usb.rndis0 ]
        then
    fi

    if [ ! -e /boot/disable_hid ]
    then
        # keyboard
        mkdir functions/hid.GS0
        if [ -e /boot/BIOS ]
        then
            echo 1 > functions/hid.GS0/subclass
            echo 1 > functions/hid.GS0/wakeup_on_write
        fi
        ln -s functions/hid.GS0 configs/c.1

        # mouse
        mkdir functions/hid.GS1
        if [ -e /boot/BIOS ]
        then
            echo 1 > functions/hid.GS1/subclass
        fi
        if [ ! -e /boot/usb.notwakeup ]
        then
            echo 1 > functions/hid.GS1/wakeup_on_write
        fi
        echo 2 > functions/hid.GS1/protocol
        echo -ne \\x34 > functions/hid.GS1/report_length
        ln -s functions/hid.GS1 configs/c.1

        # touchpad
        mkdir functions/hid.GS2
        if [ -e /boot/BIOS ]
        then
            echo 1 > functions/hid.GS2/subclass
        fi
        if [ ! -e /boot/usb.notwakeup ]
        then
            echo 1 > functions/hid.GS2/wakeup_on_write
        fi
        echo 2 > functions/hid.GS2/protocol
        echo 6 > functions/hid.GS2/report_length
        echo -ne \\x05\\x01\\x09\\x02\\xa1\\x01\\x09\\x01\\xa1\\x00\\x05\\x09\\x19\\x01\\x29\\x03\\x15\\x00\\x25\\x01\\x95\\x03\\x75\\x01\\x81\\x02\\x95\\x01\\x75\\x05\\x81\\x01\\x05\\x01\\x09\\x30\\x09\\x31\\x15\\x00\\x26\\xff\\x7f\\x35\\x00\\x46\\xff\\x7f\\x75\\x10\\x95\\x02\\x81\\x02\\x05\\x01\\x09\\x38\\x15\\x81\\x25\\x7f\\x35\\x00\\x45\\x00\\x75\\x08\\x95\\x01\\x81\\x06\\xc0\\xc0 > functions/hid.GS2/report_desc
        ln -s functions/hid.GS2 configs/c.1
    fi

    if [ -e /boot/usb.disk0 ]
    then
            mkdir functions/mass_storage.disk0
            ln -s functions/mass_storage.disk0 configs/c.1/
            echo 1 > functions/mass_storage.disk0/lun.0/removable
            if [ -e /boot/usb.disk0.ro ]
            then
                    echo 1 > functions/mass_storage.disk0/lun.0/ro
                    echo 0 > functions/mass_storage.disk0/lun.0/cdrom
            fi
            disk=$(cat /boot/usb.disk0)
            if [ -z "${disk}" ]
            then
                    # if [ ! -e /mnt/usbdisk.img ]
                    # then
                    #         fallocate -l 8G /mnt/usbdisk.img
                    #         mkfs.vfat /mnt/usbdisk.img
                    # fi
                    echo /dev/mmcblk0p3 > functions/mass_storage.disk0/lun.0/file
            else
                    cat /boot/usb.disk0 > functions/mass_storage.disk0/lun.0/file
            fi
    fi

    ls /sys/class/udc/ | cat > UDC
    echo device > /proc/cviusb/otg_role
}

start_usb_host(){
    echo '' > /sys/kernel/config/usb_gadget/g0/UDC
    echo host > /proc/cviusb/otg_role
}`

samascience, 2025-02-28:
@Z2Z-GuGu  - any idea where in particular we can change ?

CaFeXYZ, 2025-03-21:
> [@Z2Z-GuGu](https://github.com/Z2Z-GuGu) - any idea where in particular we can change ?

You're looking at the wrong file. It's the one below. It's pretty self-explanatory once you open the file.

```
/etc/init.d # ls -ltr | grep S03usbdev
-rwxr-xr-x    1 root     root          4728 Mar 19 05:29 S03usbdev
```