sipeed/NanoKVM #109: CDROM/DVD emulation does not work with Windows 11

xardomain, 2024-10-16:
Steps to reproduce:
1)	Download 1.20 official firmware from https://github.com/sipeed/NanoKVM/releases
2)	Flashed a microSD 32Gb (SanDisk Ultra)
3)	Inserted into NanoKVM Full
4)	Connected to a desktop:
HW:
Processor	Intel(R) Core(TM) i9-14900K   3.20 GHz
Installed RAM	48.0 GB (47.7 GB usable)
BaseBoard Product	TUF GAMING Z790-PRO WIFI

OS:
System type	64-bit operating system, x64-based processor
Pen and touch	No pen or touch input is available for this display
Edition	Windows 11 Pro
Version	23H2
Installed on	‎1/‎13/‎2024
OS build	22631.4317
Experience	Windows Feature Experience Pack 1000.22700.1041.0
5)	Network connected to a TP-Link switch, acquired an ip address using DHCP:
Image Version2024-08-08-19-44-bef2ca.img
Firmware Version1.0.0

6)	Downloaded the following ISO from the pc and copied into the virtual disk:
Fedora FC40 Network Install
https://download.fedoraproject.org/pub/fedora/linux/releases/40/Everything/x86_64/iso/Fedora-Everything-netinst-x86_64-40-1.14.iso
SHA256: 4d1c0a7dda6c1d21a1483acb6c7914193158921113f947c5a0519d26bcc548b2
MD5: f7d36e9c364bc56df7b27a63642dd652

7)	Verified MD5 on the USB disk, it matches
8)	Ejected the USB disk from windows 11
9)	Rebooted the NanoKVM using the reboot command from Terminal
10)	After reboot, Ejected the USB disk from windows 11
11)	Mounted the iso image but I get the error:
"You need to format the disk in drive f: before you can use it"
12)	Updated the firmware from 1.0.0 to 2.1, same results

It seems that the command issued for the USB gadget is more like mounting a partition in rw than a .iso file to emulate a CDROM/DVD in ro.

Z2Z-GuGu, 2024-10-18:
@xardomain Hi, by default, USB Gadget will mount the third partition in the TF card. After inserting it into the computer, the computer will recognize an ExFAT disk of about 23GB. When copying the .iso file and selecting the file for mounting on the webpage, Gadget will mount the iOS image separately. Then, when your computer is in boot state, it will recognize the ISO image disk and start installing. But the computer system has started, the mounted ISO image is generally not recognized by the system, so a formatting reminder will pop up

Z2Z-GuGu, 2024-10-18:
Just now, I burned the image you provided with a real USB flash drive (not NanoKVM) and found that it would also pop up a formatted window in Win11. The virtual USB flash drive function of NanoKVM only simulates a installable USB flash drive, which performs the same as a USB flash drive in many cases. If you want to simulate CDROM/DVD, maybe you need to modify the NanoKVM system:`/etc/init.d/S03usbdev`, which has the initialization option for USB Gedget

xardomain, 2024-10-18:
Hi,
many thanks. I fiddled with S03usbdev and I was able to get the iso working, in a way I needed (boot from it). I would like if possible:
1) mount an iso as it would mimic a CDROM/DVD from web interface (feature request) 
2) mount a partition (as it currently works) but decided if I want it read-only or read-write (feature request)
3) should it be a way to refresh/sense when in the images change/are added. If I understand correctly, you need a kvm reboot to sense a change in the images (feature request).

closed 2024-10-18 by xardomain

maximko, 2025-01-29:
> Hi, many thanks. I fiddled with S03usbdev and I was able to get the iso working, in a way I needed (boot from it). I would like How did you get isos working? What needs to be changed in S03usbdev?

xardomain, 2025-01-30:
Hi Maximko,
I don't remember but it was a parameter used in linux gadgets, I would have to search for it. I would rather to have a distinction between mounting an ISO as bootable disk and mounting a partition with a proper menu item (as I specified before). Another "feature" is than it does not sense the change and you would have to reboot to make it acknowledge the change.

xardomain, 2025-02-02:
Maximko,

this is what I did, I described here:

https://maixhub.com/discussion/100604

It is a little bit of time since I used both of my NanoKVM (usb one has a problem with RJ connector), PCIe is at the moment powered off.

I would like to have the option to use ISO files as bootable one within user menus, but I don't know how to change them in the source. Hopefully it will make into a firmware release, didn't check if any new release is available.

maximko, 2025-02-03:
Thank you, I made same changes to S03usbdev shorly after posting here. I had luck with some linux images but not win11 iso. That change you linked was sufficient for you to boot windows installer?