sipeed/NanoKVM #110: Shipping question

Datexcz, 2024-10-17:
Hi, on september 29 i preorder the Nanokvm. Here on the github page it says that the preorders should ship out mid october. But then i noticed that Aliexpress says the NanoKVM will be shipped on Nowember 30. So wheres the truth?

Reletiv, 2024-10-17:
Mine now says December 31..... so far mid October yes indeed

kiler129, 2024-10-18:
@Datexcz this is GitHub and not sales support. If you have questions about pre-sale orders you can e-mail them at `support@sipeed.com`.

Zepan, 2024-10-18:
Hi, the Aug&Sep order will send out in Oct; the Oct order will send in Nov&Dec.
The newest 5Kpcs will start ship this weekend, most order before 10.1 will finish ship before 10.31

jernejs, 2024-11-04:
I ordered mine in August, and received shipping notification yesterday (and according to tracking it's been shipped on Friday, and it's already in the country).

wj-xiao, 2024-11-08:
Estimated shipping time:

| Purchase Time         | Shipping Time          |
|-----------------------|------------------------|
| July ~ September 24   | Before October 31      |
| September 25 ~ October 25 | November 15 ~ November 30 |
| October 26 ~ Now      | December               |

closed 2024-11-08 by wj-xiao