sipeed/NanoKVM #111: Not working with raspberrry pi 4

andrasg, 2024-10-19:
Trying to get nanovkm working with my rpi4. HDMI works, can see the screen but can't get USB HID to work.

See this in the pi log repeating:

```
Oct 19 14:24:59 ha-rpi kernel: [  838.689264] usb 1-1.3: new high-speed USB device number 80 using xhci_hcd
Oct 19 14:25:00 ha-rpi kernel: [  838.981484] usb 1-1.3: device descriptor read/64, error -71
Oct 19 14:25:00 ha-rpi kernel: [  839.169884] usb 1-1.3: Device not responding to setup address.
Oct 19 14:25:00 ha-rpi kernel: [  839.377509] usb 1-1.3: Device not responding to setup address.
Oct 19 14:25:00 ha-rpi kernel: [  839.585301] usb 1-1.3: device not accepting address 80, error -71
Oct 19 14:25:01 ha-rpi kernel: [  839.873310] usb 1-1.3: new high-speed USB device number 81 using xhci_hcd
Oct 19 14:25:01 ha-rpi kernel: [  839.974428] usb 1-1.3: New USB device found, idVendor=3346, idProduct=1009, bcdDevice= 5.10
Oct 19 14:25:01 ha-rpi kernel: [  839.974460] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Oct 19 14:25:01 ha-rpi kernel: [  839.974479] usb 1-1.3: Product: licheervnano
Oct 19 14:25:01 ha-rpi kernel: [  839.974495] usb 1-1.3: Manufacturer: flyingrtx
Oct 19 14:25:01 ha-rpi kernel: [  839.974510] usb 1-1.3: SerialNumber: 0123456789ABCDEF
Oct 19 14:25:01 ha-rpi kernel: [  839.982673] input: flyingrtx licheervnano as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.3/1-1.3:1.0/0003:3346:1009.044D/input/input1102
Oct 19 14:25:01 ha-rpi kernel: [  840.042700] hid-generic 0003:3346:1009.044D: input,hidraw0: USB HID v1.01 Keyboard [flyingrtx licheervnano] on usb-0000:01:00.0-1.3/input0
Oct 19 14:25:01 ha-rpi kernel: [  840.046945] input: flyingrtx licheervnano as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.3/1-1.3:1.1/0003:3346:1009.044E/input/input1103
Oct 19 14:25:01 ha-rpi kernel: [  840.047739] hid-generic 0003:3346:1009.044E: input,hidraw1: USB HID v1.01 Mouse [flyingrtx licheervnano] on usb-0000:01:00.0-1.3/input1
Oct 19 14:25:01 ha-rpi kernel: [  840.052440] input: flyingrtx licheervnano as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.3/1-1.3:1.2/0003:3346:1009.044F/input/input1104
Oct 19 14:25:01 ha-rpi kernel: [  840.053151] hid-generic 0003:3346:1009.044F: input,hidraw2: USB HID v1.01 Mouse [flyingrtx licheervnano] on usb-0000:01:00.0-1.3/input2
Oct 19 14:25:01 ha-rpi mtp-probe: checking bus 1, device 81: "/sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.3"
Oct 19 14:25:01 ha-rpi mtp-probe: bus: 1, device: 81 was not an MTP device
Oct 19 14:25:01 ha-rpi systemd-udevd[14577]: mouse1: Process '/usr/sbin/th-cmd --socket /var/run/thd.socket --passfd --udev' failed with exit code 1.
Oct 19 14:25:01 ha-rpi kernel: [  840.256322] usb 1-1.3: USB disconnect, device number 81
Oct 19 14:25:01 ha-rpi systemd-udevd[14571]: event2: Process '/usr/sbin/th-cmd --socket /var/run/thd.socket --passfd --udev' failed with exit code 1.
Oct 19 14:25:01 ha-rpi systemd-udevd[14578]: mouse0: Process '/usr/sbin/th-cmd --socket /var/run/thd.socket --passfd --udev' failed with exit code 1.
Oct 19 14:25:01 ha-rpi systemd-udevd[14566]: event4: Process '/usr/sbin/th-cmd --socket /var/run/thd.socket --passfd --udev' failed with exit code 1.
Oct 19 14:25:01 ha-rpi systemd-udevd[14565]: event3: Process '/usr/sbin/th-cmd --socket /var/run/thd.socket --passfd --udev' failed with exit code 1.
Oct 19 14:25:01 ha-rpi kernel: [  840.689278] usb 1-1.3: new high-speed USB device number 82 using xhci_hcd
Oct 19 14:25:02 ha-rpi kernel: [  840.790084] usb 1-1.3: New USB device found, idVendor=3346, idProduct=1009, bcdDevice= 5.10
Oct 19 14:25:02 ha-rpi kernel: [  840.790100] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Oct 19 14:25:02 ha-rpi kernel: [  840.790108] usb 1-1.3: Product: licheervnano
Oct 19 14:25:02 ha-rpi kernel: [  840.790114] usb 1-1.3: Manufacturer: flyingrtx
Oct 19 14:25:02 ha-rpi kernel: [  840.790120] usb 1-1.3: SerialNumber: 0123456789ABCDEF
Oct 19 14:25:02 ha-rpi kernel: [  840.795355] input: flyingrtx licheervnano as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.3/1-1.3:1.0/0003:3346:1009.0450/input/input1105
Oct 19 14:25:02 ha-rpi kernel: [  840.854026] hid-generic 0003:3346:1009.0450: input,hidraw0: USB HID v1.01 Keyboard [flyingrtx licheervnano] on usb-0000:01:00.0-1.3/input0
Oct 19 14:25:02 ha-rpi kernel: [  840.856523] input: flyingrtx licheervnano as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.3/1-1.3:1.1/0003:3346:1009.0451/input/input1106
Oct 19 14:25:02 ha-rpi kernel: [  840.856821] hid-generic 0003:3346:1009.0451: input,hidraw1: USB HID v1.01 Mouse [flyingrtx licheervnano] on usb-0000:01:00.0-1.3/input1
Oct 19 14:25:02 ha-rpi kernel: [  840.859115] input: flyingrtx licheervnano as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.3/1-1.3:1.2/0003:3346:1009.0452/input/input1107
Oct 19 14:25:02 ha-rpi kernel: [  840.859503] hid-generic 0003:3346:1009.0452: input,hidraw2: USB HID v1.01 Mouse [flyingrtx licheervnano] on usb-0000:01:00.0-1.3/input2
Oct 19 14:25:02 ha-rpi mtp-probe: checking bus 1, device 82: "/sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.3"
Oct 19 14:25:02 ha-rpi mtp-probe: bus: 1, device: 82 was not an MTP device
Oct 19 14:25:02 ha-rpi mtp-probe: checking bus 1, device 82: "/sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.3"
Oct 19 14:25:02 ha-rpi mtp-probe: bus: 1, device: 82 was not an MTP device
Oct 19 14:25:02 ha-rpi kernel: [  841.024416] usb 1-1.3: USB disconnect, device number 82
```

Tried both USB cables that were supplied with the nanokvm. Any ideas?

Z2Z-GuGu, 2024-10-22:
It seems that there is a problem of frequent disconnection, and it is most likely caused by poor contact of the USB cable, have you tried other cables?

andrasg, 2024-10-22:
> It seems that there is a problem of frequent disconnection, and it is most likely caused by poor contact of the USB cable, have you tried other cables?

I have tried both cables supplied in the box. I have tried all 4 ports on the pi. Will try to get another cable and test again then.

tatar-adrian, 2024-11-03:
I'm having the same issue as @andrasg 

[dmesg-pi4.txt](https://github.com/user-attachments/files/17609872/dmesg-pi4.txt)

LeonStraathof, 2025-06-09:
This is caused by a hardware problem. The HID connector backfeeds +5V into the Pi USB connector. The Pi does not like that. Most PC motherboards do not care, which is why you see less complaints there. I have both a Pi4 and a PC motherboard that don't like the backfeed. The cube version can be modified by removing a zero ohm resistor and replacing it with a diode. The PCI version you can remove the +5V wire from the 4 wire cable that connects the lower and upper pcb. When you don't have backfeed all works fine

Z2Z-GuGu, 2025-06-13:
It may be related to the USB mode. You can try updating to the latest version and enabling HID Only mode, as this mode has the best compatibility. In testing, many hosts with HID issues were fixed by enabling HID Only mode.

closed 2025-06-13 by Z2Z-GuGu

connorfryar, 2026-09-02:
Currently working on this issue as well. 
nanokvm version 
2026-06-10-1_4_3.img
```
rpi information:
PRETTY_NAME="Debian GNU/Linux 13 (trixie)"
NAME="Debian GNU/Linux"
VERSION_ID="13"
VERSION="13 (trixie)"
VERSION_CODENAME=trixie
DEBIAN_VERSION_FULL=13.5
ID=debian
```

Power and cables. 
- AUX is plugged directly into a wall socket
- HID is the stock cable modified so that the 5v has been cut.

Log output with the cut 5v
```
Sep 02 09:06:03 raspberrypi kernel: logitech-hidpp-device 0003:046D:404D.0008: input,hidraw1: USB HID v1.11 Keyboard [Logitech K400 Plus] on usb-0000:01:00.0-1.1/input2:1
Sep 02 09:06:03 raspberrypi kernel: usb 1-1-port2: Cannot enable. Maybe the USB cable is bad?
Sep 02 09:06:04 raspberrypi kernel: usb 1-1-port2: Cannot enable. Maybe the USB cable is bad?
Sep 02 09:06:04 raspberrypi kernel: usb 1-1-port2: attempt power cycle
Sep 02 09:06:05 raspberrypi kernel: usb 1-1-port2: Cannot enable. Maybe the USB cable is bad?
Sep 02 09:06:06 raspberrypi kernel: usb 1-1-port2: Cannot enable. Maybe the USB cable is bad?
Sep 02 09:06:06 raspberrypi kernel: usb 1-1-port2: unable to enumerate USB device
Sep 02 09:06:06 raspberrypi mtp-probe[1716]: checking bus 1, device 20: "/sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.1"
Sep 02 09:06:06 raspberrypi mtp-probe[1716]: bus: 1, device: 20 was not an MTP device
Sep 02 09:06:06 raspberrypi mtp-probe[1721]: checking bus 1, device 20: "/sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.1"
Sep 02 09:06:06 raspberrypi mtp-probe[1721]: bus: 1, device: 20 was not an MTP device
Sep 02 09:06:06 raspberrypi systemd-logind[822]: Watching system buttons on /dev/input/event4 (Logitech K400 Plus)
Sep 02 09:06:18 raspberrypi kernel: logitech-hidpp-device 0003:046D:404D.0008: HID++ 4.1 device connected.
Sep 02 09:09:17 raspberrypi kernel: usb 1-1-port2: Cannot enable. Maybe the USB cable is bad?
Sep 02 09:09:18 raspberrypi kernel: usb 1-1-port2: Cannot enable. Maybe the USB cable is bad?
Sep 02 09:09:18 raspberrypi kernel: usb 1-1-port2: attempt power cycle
Sep 02 09:09:19 raspberrypi kernel: usb 1-1-port2: Cannot enable. Maybe the USB cable is bad?
Sep 02 09:09:20 raspberrypi kernel: usb 1-1-port2: Cannot enable. Maybe the USB cable is bad?
Sep 02 09:09:20 raspberrypi kernel: usb 1-1-port2: unable to enumerate USB device
Sep 02 09:10:26 raspberrypi kernel: usb 1-1-port2: Cannot enable. Maybe the USB cable is bad?
Sep 02 09:10:27 raspberrypi kernel: usb 1-1-port2: Cannot enable. Maybe the USB cable is bad?
Sep 02 09:10:27 raspberrypi kernel: usb 1-1-port2: attempt power cycle
Sep 02 09:10:28 raspberrypi kernel: usb 1-1-port2: Cannot enable. Maybe the USB cable is bad?
Sep 02 09:10:29 raspberrypi kernel: usb 1-1-port2: Cannot enable. Maybe the USB cable is bad?
Sep 02 09:10:29 raspberrypi kernel: usb 1-1-port2: unable to enumerate USB device
```

Log output with a non cut 5v usb a to usb c

```
usb 1-1.4: new high-speed USB device number 64 using xhci_hcd
usb 1-1.4: Device not responding to setup address.
usb 1-1.4: Device not responding to setup address.
usb 1-1.4: device not accepting address 64, error -71
usb 1-1-port4: unable to enumerate USB device
```

I am continuing to hunt for a solution.
Next -- HID only mode.
- https://github.com/sipeed/NanoKVM/issues/397

Emilien-ma, 2026-09-15:
> This is caused by a hardware problem. The HID connector backfeeds +5V into the Pi USB connector. The Pi does not like that. Most PC motherboards do not care, which is why you see less complaints [...]

Thanks to you, I tried something : 
I put a piece of electrical tape on top of the 5V pin inside the USB-A connector of my USB wire (the cable to connect the HID USB-C port and the USB-A port of the Raspberry). And it seems to work : I have no more the warning "overcurrent" message on my Raspberry PI desktop.