sipeed/NanoKVM #115: [Security] EMBA report of 20240809_NanoKVM_Rev1_2_0.img

lwbt, 2024-10-25:
Out of curiosity I ran [emba](https://github.com/e-m-b-a/emba) today on the latest image. Below you can find an excerpt. I removed the components which had 0 CVEs to not make the list look much longer than necessary.

Don't just look the the CVEs. Stack canaries, RELRO, PIE and others are also important.

From what I can tell the images are simply attached to a release here on GitHub, there is no automated build pipeline and workflow to build, test and verify. I understand that one has to start somewhere to get a product out of the door before improving tooling. But the product is also advertised as being superior to traditional IPMI/KVM solutions, and to be honest, this report is devastating and discloses ignorance of industry standards going back 10 years and more.

If there is something like a board support supplier involved Sipeed should also talk to them and get as many issues resolved as possible. Shipping an LTS Linux kernel from a few years ago and a few not so fresh dependencies may pass. Shipping **1188 High rated CVEs** is not acceptable.

I can provide the full HTML log to everyone who is interested and doesn't want to spin up Kali Linux with EMBA for a 2-3 hours on 16 cores themselves.

```
[+] Final aggregator

The main aggregator module compiles and summarizes results from various analysis modules into a comprehensive overview by processing and logging detailed information from each identified element.

[+] Tested firmware: /home/kali/Downloads/emba/20240809_NanoKVM_Rev1_2_0.img.xz
[+] EMBA start command: ./emba -l /home/kali/log -f ./20240809_NanoKVM_Rev1_2_0.img.xz -p ./scan-profiles/default-scan.emba
[+] Detected architecture and endianness (verified): RISCV / EL
[+] Operating system detected (verified): Linux / v4.4.0 / v5.10.4
[+] Linux distribution detected: Buildroot 2023.11.2
[+] Linux distribution detected: buildroot 2023.11.2

[+] 25022 files and 5082 directories detected.
[+] Entropy analysis of binary firmware is: 7.999996 bits per byte.
[+] Entropy analysis of binary firmware is available: /logs/firmware_entropy.png
[+] Found 159 issues in 94 shell scripts.
[+] Found 16879 vulnerabilities in 3094 python files.

[+] Found the following configuration issues:
    Found 1 authentication issues.
    Found 107 password related details.
    Found 243 password related details via STACS.
    Found 15 outdated certificates and 6 expiring certificates in 145 certificate files and in a total of 703 certificates.
    Found 184 kernel modules with 0 licensing issues.
    Found 3 interesting files and 4 files that could be useful for post-exploitation.

[+] Found 397 (36%) binaries without enabled stack canaries in 1111 binaries.
[+] Found 190 (17%) binaries without enabled RELRO in 1111 binaries.
[+] Found 185 (17%) binaries without enabled NX in 1111 binaries.
[+] Found 66 (6%) binaries without enabled PIE in 1111 binaries.
[+] Found 916 (82%) stripped binaries without symbols in 1111 binaries.

[*] Identified the following software inventory, vulnerabilities and exploits:
[+] Found version details:      certifi             :   2023.7.22      :   CVEs: 1         :   Exploits: 0    :   Source: unknown
[+] Found version details:      babel               :   2.13.1         :   CVEs: 1         :   Exploits: 1    :   Source: unknown
[+] Found version details:      libcurl             :   8.5.0          :   CVEs: 1         :   Exploits: 0    :   Source: STAT
[+] Found version details:      libarchive          :   3.7.2          :   CVEs: 3         :   Exploits: 0    :   Source: STAT
[+] Found version details:      dnsmasq             :   2.89           :   CVEs: 2         :   Exploits: 0    :   Source: STAT
[+] Found version details:      django              :   4.2.7          :   CVEs: 7         :   Exploits: 0    :   Source: unknown
[+] Found version details:      pathtools           :   0.1.2          :   CVEs: 1         :   Exploits: 0    :   Source: unknown
[+] Found version details:      pip                 :   22.3.1         :   CVEs: 2         :   Exploits: 0    :   Source: unknown
[+] Found version details:      pillow              :   10.0.1         :   CVEs: 1         :   Exploits: 1    :   Source: unknown
[+] Found version details:      selenium            :   4.9.1          :   CVEs: 1         :   Exploits: 1    :   Source: unknown
[+] Found version details:      sqlparse            :   0.4.3          :   CVEs: 1         :   Exploits: 0    :   Source: unknown
[+] Found version details:      tornado             :   6.2            :   CVEs: 1         :   Exploits: 0    :   Source: unknown
[+] Found version details:      trio                :   0.22.0         :   CVEs: 1         :   Exploits: 1    :   Source: unknown
[+] Found version details:      werkzeug            :   0.0.0          :   CVEs: 8         :   Exploits: 1    :   Source: unknown        
[+] Found version details:      busybox             :   1.36.1         :   CVEs: 4         :   Exploits: 0    :   Source: STAT
[+] Found version details:      binutils            :   2.39.50        :   CVEs: 7         :   Exploits: 1    :   Source: STAT           
[+] Found version details:      libexpat            :   2.6.0          :   CVEs: 3         :   Exploits: 0    :   Source: STAT           
[+] Found version details:      sqlite              :   3.43.1         :   CVEs: 1         :   Exploits: 0    :   Source: STAT
[+] Found version details:      wpa_supplicant      :   2.10           :   CVEs: 1         :   Exploits: 0    :   Source: STAT
[+] Found version details:      systemd             :   3.2.14         :   CVEs: 25        :   Exploits: 8    :   Source: STAT
[+] Found version details:      zlib                :   1.2.11         :   CVEs: 3         :   Exploits: 0    :   Source: STAT
[+] Found version details:      zlib                :   1.2.13         :   CVEs: 1         :   Exploits: 0    :   Source: STAT
[+] Found version details:      zlib                :   1.3            :   CVEs: 1         :   Exploits: 0    :   Source: STAT
[+] Found version details:      libtiff             :   4.6.0          :   CVEs: 1         :   Exploits: 0    :   Source: STAT
[+] Found version details:      openssl             :   3.1.4          :   CVEs: 3         :   Exploits: 0    :   Source: STAT
[+] Found version details:      vim                 :   9.0            :   CVEs: 88        :   Exploits: 10   :   Source: STAT
[+] Found version details:      linux_kernel        :   5.10.4         :   CVEs: 1366      :   Exploits: 40   :   Source: STAT
[+] Found version details:      linux_kernel        :   4.4.0          :   CVEs: 1767      :   Exploits: 90   :   Source: STAT

[+] Identified a SBOM including 201 software components with version details.

[+] Identified 3306 CVE entries.
    Identified 1188 High rated CVE entries / Exploits: 97
    Identified 1975 Medium rated CVE entries / Exploits: 49
    Identified 143 Low rated CVE entries /Exploits: 8
    154 possible exploits available (17 Metasploit modules).
    Remote exploits: 0 / Local exploits: 34 / DoS exploits: 4 / Github PoCs: 0 / Known exploited vulnerabilities: 10 / Verified Exploits: 0
```

Zepan, 2024-10-28:
Hi, thank you for your feedback!
Most CVEs are introduced by old linux kernel. we have report it to sophgo months ago, they said they are pushing moving to kernel 6.6. when the offical SDK upgrade to linux 6.x, we will upgrade and clean the most CVEs.
In fact it is common problem in commercial products, billions IPCs are run on linux 5.x or older.
Even we upgrade to linux 6.x, few years latter, it also will become "unsafe"...

CRCinAU, 2024-10-28:
Context is needed here. How many are exploitable from remote? How many require an authenticated user? How many affect the web UI?

lwbt, 2024-10-28:
> Context is needed here. [...]

Thanks.

On second thought, it's mostly HTML, so I just uploaded it to a repo: https://github.com/lwbt/emba_20240809_NanoKVM_Rev1_2_0.img

Unfortunately it has to be downloaded before it can be viewed in a browser.

scpcom, 2024-10-29:
Most can be solved by updating buildroot and because linux 5.10.y is LTS you can merge the fixes from kernel.org.

It tested it with an image build my updated repository:
https://github.com/scpcom/LicheeSG-Nano-Build/tree/develop

```
The main aggregator module compiles and summarizes results from various analysis modules into a comprehensive overview by processing and logging detailed information from each identified element.

[+] Tested firmware: /home/kali/2024-10-28-03-02-17ccfa-nanokvm.img
[+] EMBA start command: ./emba -l /home/kali/log -f /home/kali/2024-10-28-03-02-17ccfa-nanokvm.img -p ./scan-profiles/default-scan.emba
[+] Detected architecture and endianness (verified): RISCV / EL
[+] Operating system detected (verified): Linux / v5.10.226
[+] Linux distribution detected: Buildroot 2024.05.3
[+] Linux distribution detected: buildroot 2024.05.3

-----------------------------------------------------------------

[+] 25153 files and 5210 directories detected.
[+] Entropy analysis of binary firmware is: 4.005469 bits per byte.
[+] Entropy analysis of binary firmware is available: /logs/firmware_entropy.png
[+] Found 154 issues in 94 shell scripts.
[+] Found 17105 vulnerabilities in 3108 python files.

-----------------------------------------------------------------

[+] Found the following configuration issues:
    Found 1 authentication issues.
    Found 107 password related details.
    Found 241 password related details via STACS.
    Found 9 outdated certificates and 6 expiring certificates in 145 certificate files and in a total of 709 certificates.                                
    Found 57 kernel modules with 0 licensing issues.
    Found 3 interesting files and 5 files that could be useful for post-exploitation.                                                                     

-----------------------------------------------------------------

[+] Found 293 (29%) binaries without enabled stack canaries in 1019 binaries.
[+] Found 59 (6%) binaries without enabled RELRO in 1019 binaries.
[+] Found 58 (6%) binaries without enabled NX in 1019 binaries.
[+] Found 94 (9%) binaries without enabled PIE in 1019 binaries.
[+] Found 954 (94%) stripped binaries without symbols in 1019 binaries.

-----------------------------------------------------------------

[*] Identified the following software inventory, vulnerabilities and exploits:
[+] Found version details:      babel               :   2.14.0         :   CVEs: 1         :   Exploits: 1    :   Source: unknown                         
[+] Found version details:      django              :   5.0.8          :   CVEs: 2         :   Exploits: 0    :   Source: unknown                         
[+] Found version details:      libcurl             :   8.9.0          :   CVEs: 1         :   Exploits: 0    :   Source: STAT                            
[+] Found version details:      libarchive          :   3.7.4          :   CVEs: 2         :   Exploits: 0    :   Source: STAT                            
[+] Found version details:      pathtools           :   0.1.2          :   CVEs: 1         :   Exploits: 0    :   Source: unknown                         
[+] Found version details:      pip                 :   23.3.2         :   CVEs: 1         :   Exploits: 0    :   Source: unknown                         
[+] Found version details:      tornado             :   6.2            :   CVEs: 1         :   Exploits: 0    :   Source: unknown                         
[+] Found version details:      trio                :   0.22.0         :   CVEs: 1         :   Exploits: 1    :   Source: unknown                         
[+] Found version details:      busybox             :   1.36.1         :   CVEs: 4         :   Exploits: 0    :   Source: STAT                            
[+] Found version details:      u-boot              :   2021.07        :   CVEs: 4         :   Exploits: 0    :   Source: STAT                            
[+] Found version details:      libexpat            :   2.6.2          :   CVEs: 3         :   Exploits: 0    :   Source: STAT                            
[+] Found version details:      binutils            :   2.39.50        :   CVEs: 7         :   Exploits: 1    :   Source: STAT                            
[+] Found version details:      wpa_supplicant      :   2.10           :   CVEs: 1         :   Exploits: 0    :   Source: STAT                            
[+] Found version details:      systemd             :   3.2.14         :   CVEs: 25        :   Exploits: 8    :   Source: STAT                            
[+] Found version details:      libtiff             :   4.6.0          :   CVEs: 1         :   Exploits: 0    :   Source: STAT                            
[+] Found version details:      vim                 :   9.1            :   CVEs: 2         :   Exploits: 2    :   Source: STAT                            
[+] Found version details:      linux_kernel        :   5.10.226       :   CVEs: 789       :   Exploits: 16   :   Source: STAT                            

[+] Identified a SBOM including 197 software components with version details.

[+] Identified 846 CVE entries.
    Identified 232 High rated CVE entries / Exploits: 18
    Identified 595 Medium rated CVE entries / Exploits: 11
    Identified 19 Low rated CVE entries /Exploits: 0
    29 possible exploits available.
    Remote exploits: 0 / Local exploits: 5 / DoS exploits: 0 / Github PoCs: 0 / Known exploited vulnerabilities: 2 / Verified Exploits: 0                 

```