sipeed/NanoKVM #116: Permanently disable tailscale service

FingerlessGlov3s, 2024-10-28:
I would like to keep the tailscale service stopped between reboots as it keeps a concurrent connection to tailscale servers, when it's a feature I'm not going to use, all access will be done locally or over S2S VPNs of physical firewalls.

For now I've just done the below, but wondered if there's a better way, other than just moving the file somewhere else.
```
 /etc/init.d/S98tailscaled stop
```

FingerlessGlov3s, 2024-10-29:
> ```
> chmod -x  /etc/init.d/S98tailscaled
> ```
> 
> doesn't work?

I'm pretty sure doing that won't do anything due to the filesystem being fat32 and has no concept of permissions, everything is rwx in the filesystem.

Deleting the file would stop it happening, but I don't want it coming back when I upgrade and forget I deleted it. This maybe more a feature feature, if there's not offical way to disable the starting of a service, like you can do with systemd or normal other init systems.

dulasau, 2024-10-29:
```/dev/mmcblk0p2 on / type ext4 (rw,relatime)```

FingerlessGlov3s, 2024-10-29:
> `/dev/mmcblk0p2 on / type ext4 (rw,relatime)`

Ah my bad, either confused with an old security post I was reading about it, or I just dreamt that. 😅

Question is will that stick between upgrades, assuming the upgrade won't revert the permission. If the updates are only delta's of what has changed file wise on the filesystem, it maybe fine until they update the tailscale init.d file.

I'm unlikily to upgrade it constantly, so I'll probably just add it ot my internal documentation to run the below after each upgrade to ensure it's got no execute perms, rather than just deleting the file.

```
ssh root@ip 'chmod -x /etc/init.d/S98tailscaled && reboot'
```

wj-xiao, 2024-11-08:
There are currently two ways to disable Tailscale:
- deleting or moving the Tailscale file `/usr/bin/tailscale` and `/usr/sbin/tailscaled` (you can reinstall Tailscale in the web page).
- deleting or removing the Tailscale startup script `/etc/init.d/S98tailscaled` (you should back up this file just like @FingerlessGlov3s ).

We will be adding a switch in the web page to allow users to decide whether Tailscale should run at startup.

wj-xiao, 2025-01-16:
In version 2.1.5, you can stop tailscale and disable automatic start.

closed 2025-01-16 by wj-xiao