sipeed/NanoKVM #117: Image doesn't get updated in BIOS or Linux console

dulasau, 2024-10-28:
Version 2.0.9
Everything works fine when an OS loaded (tried on different machines with Debian/Ubuntu), but whenever I enter BIOS or a machine loaded into Linux console image become static: image is not getting updated, only page refresh updates the image.

* Sometimes the image can get refreshed automatically after about 60 secs
* Frame Detect is disabled
* OLED Display shows 30FPS
* Tried different resolutions/FPS/Quality - the same result
* In a browser dev tool I can see that request /api/stream/mjpeg is working fine 
 
It almost looks like "Frame Detect" is't "fully" disabled

dulasau, 2024-10-29:
I've been playing with it entire day yesterday and it wasn't working, re-flashing didn't help either....
But I just turned it on and It magically started working!!!
I'll keep the issue open while keeping an eye on the device for a couple of more days just to be sure.

dulasau, 2024-10-30:
It stopped working again (((((

dulasau, 2024-10-30:
The problem seems to intermediate, most of the times it doesn't work, sometimes it does. Got the same problems once with Ubuntu desktop as well (((

nwgat, 2024-10-31:
how did you connect power?

dulasau, 2024-10-31:
Tried both options: through USB from the host machine and thought AUX USB port - the same behavior.

dulasau, 2024-11-05:
Alrighty, after a lot of testing turned out that the problem seems to be related to my Unify (pro 48 poe) switch ..... Have no idea how/what/why:
* nanokvm gets an IP
* I can access Web UI, but no image update
* Works flawless with my Mikrotik routers/switches (even with CRS317 through SPF+ ethernet adapter)

If i connect nanokvm to unifi switch and boot it - most of the times it doesn't work (sometimes it does), re-connect it (without powering off) to a mikrotik switch - it starts working, connect it back to unifi switch (without powering off) - it's working.... Mystery

closed 2024-11-06 by dulasau