sipeed/NanoKVM #119: nanoKVM doesnt work with Mac mini

haggis-supper, 2024-10-31:
I've plugged both a NanoKVM lite and a NanoKVM full into a MAC mini, they both show the screen but the mouse and keyboard dont work.

MetalSamurai99, 2024-10-31:
I have 3 NanoKVM (full) attached to Mac minis. 

For 1 running MacOS 10.12.6 the screen updates correctly, the keyboard works, but the mouse movements are very unpredictable and basically useless

The other two are running XCP-NG (Linux, I think kernel 5.x) with text console and they work just fine. 

I did test with an old Raspberry Pi 1B and again video capture works, but in that case I never got keyboard input working at all. 

All are set up with POE splitters for power, not drawing power from the host USB port.

haggis-supper, 2024-10-31:
Yep as you said, i'm running Mac Os 12.7.6, like youve said the screen updates but i dont get any life from Keyboard or Mouse. 

Linux hosts work fine, I will check my Mac mini with my PiKVM ( and see how that goes )

pfrench42, 2024-10-31:
Mine just arrived today.
Hooked it up to a 2012 Mac Mini.
Keyboard and Mouse worked , even through a reboot of the Mini.
After about 20 minutes, keyboard and mouse are unresponsive on the mini
I rebooted the NanoKVM and key/mouse came back up on the Mini

jasonhouston, 2024-11-04:
I am noticing that mine intermittently stop responding to keyboard/mouse on Linux hosts, a reboot generally fixes it but I have a couple that are persistently failing to respond to keyboard/mouse control.

haggis-supper, 2024-11-12:
New code now works with Mac minis and seems to be very stable.

vipavlovic, 2025-01-31:
Mine works with my M3 Macbook Pro, but it has problems with CMD+anything.  E.g., both CMD+C and CMD+V result in an infinite repetition of the same command, until I press ESC.  Any way to fix this?

tjcomserv, 2025-03-05:
How are you accessing the KVM? Via a Mac device? Using Safari/Chrome?

vipavlovic, 2025-04-08:
> How are you accessing the KVM? Via a Mac device? Using Safari/Chrome?

Yes, via a Mac device.  Tried several browsers, including Safari and Chrome, both result in the same behavior - repeated keystrokes.

tjcomserv, 2025-04-17:
I have just had a similar issue that I resolved by ensuring that the HID USB cable AND the power USB cable were both connected to the PC/Mac I wanted to control. If KVM is powered by another PC or wall plug then neither the keyboard and mouse work.

Myke, 2025-04-26:
HID-Only mode seems to do the trick on the MacMini I've been testing with.