sipeed/NanoKVM #124: Displayport to HDMI converter doesn't work

pojlFDlxCOvZ4Kg8y1l4, 2024-11-04:
Hello, 

The NanoKVM arrived a few days ago. 
It works fine with directly attached HDMI devices. 
I am having some problems with a DisplayPort to HDMI converter.  
There is no screen output recognized with the NanoKVM when using the DP to HDMI converter.
Other devices (HDMI monitors and a DIY PiKVM) work with the converter. 
I've swapped HDMI cables, tried old and newer cables. Also bought a new HDMI cable certified for 8k resolution. Sadly still no screen output with the NanoKVM.

Is there a config edit that I could do to fix this or is this a bug?

System details:
GPU: Nvidia T400 2GB
Converter: Startech.com MDP2HDW, Mini Displayport to HDMI converter.

nwgat, 2024-11-07:
my Active Club3D CAC-1070 DisplayPort 1.2 to HDMI 2.0 works,  Is the one you have active?

pojlFDlxCOvZ4Kg8y1l4, 2024-11-07:
The ones i have are passive. I'll look into getting an active converter. Is the one you mentioned working well?

nwgat, 2024-11-07:
its been stable for the few days ive tested with it, its a active adapter so it will output a proper hdmi signal nor some wierdly combined one like with passive adapters

Z2Z-GuGu, 2024-11-12:
@pojlFDlxCOvZ4Kg8y1l4  Hello, we have tested DP to HDMI conversion internally and it can be used normally. Please try switching the resolution on the controlled host. NanoKVM may have image loss issues in rare cases due to inability to recognize the resolution. Here is the problem description: https://wiki.sipeed.com/hardware/zh/kvm/NanoKVM/development.html#%E8%8E%B7%E5%8F%96%E6%8E%A8%E6%B5%81%E7%9B%B8%E5%85%B3%E6%95%B0%E6%8D%AE

pojlFDlxCOvZ4Kg8y1l4, 2024-11-13:
Thank you both for your replies. 
I have ordered an active Displayport to HDMI converter which arrived today.  
The HDMI input works well with the active adapter, so I'll put those passive converters in the drawer off stuff one maybe has use for in the future =]   

@Z2Z-GuGu The link to the wiki - I can't read that. Switching to the English language ends with a HTTP error 404.  

The issue is solved now, thanks!

closed 2024-11-13 by pojlFDlxCOvZ4Kg8y1l4

ScottESanDiego, 2024-11-15:
FWIW, the cheap passive adapter from Amazon https://www.amazon.com/gp/product/B09MTKHFKR/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&th=1 works for me (as does the Club 3D CAC-2070 active adapter https://www.amazon.com/gp/product/B0775Z57H9/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1 ).