sipeed/NanoKVM #125: RNDIS Driver install failed win10ltsc, usb device uncorrect

tom03dev, 2024-11-04:
Hi,
I've some issue with the rndis driver install on Win 10 ltsc. Any idea which is the correct driver ?
Out of that (I think) the stream is not shown correctly. Apps, shortcuts are not shown but mouse position on the target is correctly working and you could eg. apps.
Thanks Tom

dateno1, 2024-11-05:
It work test both lite/full on LTSC 2021

Which version you tested?

tom03dev, 2024-11-06:
I tested the full version. Firmware is newest version. However that has nothing to do with the target PC rndis driver.
I read the wiki...OOB driver ?

dateno1, 2024-11-06:
> I tested the full version. Firmware is newest version. However that has nothing to do with the target PC rndis driver. I read the wiki...OOB driver ?

https://wiki.sipeed.com/hardware/en/lichee/RV_Nano/5_peripheral.html#usb-rndis-%E7%BD%91%E5%8F%A3