sipeed/NanoKVM #126: Latest software update 2.0.9 not available on GitHub?

icedream, 2024-11-04:
I have just received the device and flashed the latest beta version available in GitHub releases to microSD, and noticed that it wasn't working (MJPEG server would not work).

Checking for software updates, it wanted to install version 2.0.9 which confused me because that version is not tagged or released on this repository. However I decided to install it anyways and that version indeed fixes the issue.

I wanted to check why that is and if I can find the source code for this version on GitHub. The Go binary of `NanoKVM-Server` included in the image has the following build metadata:

```
        build   vcs=git
        build   vcs.revision=879edf5caa44e2d0ac18b37fe8053d1ee434aa29
        build   vcs.time=2024-09-18T09:44:59Z
        build   vcs.modified=false
```

However when looking up commit `879edf5caa44e2d0ac18b37fe8053d1ee434aa29` on GitHub I can't find it - https://github.com/sipeed/NanoKVM/commit/879edf5caa44e2d0ac18b37fe8053d1ee434aa29 just returns a 404 page.

Do you have information on why that is? Is the source code tree hanging behind or not in sync with the releases for some reason?

wj-xiao, 2024-11-07:
NanoKVM consists of two components: the system image and application software.
The system image(v1.2.0) doesn't change frequently. And we won't release new versions very often. 
The application(v2.0.9) is frequently updated. So we use an online update mode.

Before the release of version 2.0.9, the project was closed source. And you could find all the activities in [this repository](https://github.com/sipeed/NanoKVM-Server/commit/879edf5caa44e2d0ac18b37fe8053d1ee434aa29). 
After 2.0.9, we made the code open source and migrated to this repository.

michaelsage, 2024-11-07:
> The system image(v1.2.0) doesn't change frequently. And we won't release new versions very often.
> The application(v2.0.9) is frequently updated. So we use an online update mode.

The application firmware is updated via the web interface. How do you update the system image? Do you have to reflash?

wj-xiao, 2024-11-08:
@michaelsage Yes. To update the image, you must take out the SD card and reflash it.
You can refer to this [document](https://wiki.sipeed.com/hardware/en/kvm/NanoKVM/system/flashing.html) for guidance.

icedream, 2024-11-09:
Thanks for the answers!

icedream, 2024-11-09:
Maybe this could be its own ticket on the respective repo but it may help to tag the releases there for organization. Not strictly needed if the commits can be discovered as described though.

closed 2024-11-09 by icedream