sipeed/NanoKVM #130: /dev/mmcblk0p3 not formatted, not mounting 23GB partition on /data

rolf-tx, 2024-11-05:
Just received NanoKVM with version app version 2.0.9, is this different from the git tag?

/dev/mmcblk0p3 was not formatted even tough /etc/S01fs appears to be intending to do this.
Result was that the 23GB partition did not get mounted on /data.

I manually formatted /dev/mmcblk0p3 from the terminal and rebooted and now 23GB is mounted on /data.

rolf-tx, 2024-11-05:
I tried flashing the 1.2.0 release on git and it formatted /dev/mmcblk0p3 and everything appears correct.
It is having the 1.0.0 version of the app and you download the latest 2.0.9 app version after first boot.

Would the developer please check the manufacturing image used?

Z2Z-GuGu, 2024-11-07:
Please refer [here](https://wiki.sipeed.com/hardware/en/kvm/NanoKVM/system/introduction.html) github 1.2.0/1.2.1 is the Image version, while 2.1.0/2.0.9 on the webpage is the APP version
Are you using NanoKVM Full? This may be related to issues in the production process, and I will resolve this problem as soon as possible.

rolf-tx, 2024-11-07:
Yes, I’m using the NanoKVM Full version

Z2Z-GuGu, 2024-11-12:
Last week, I tested the newly produced product continuously and did not find any issue with the third partition not being formatted correctly. This may be a minor bug, and I will conduct random testing in the future. I will temporarily close this issue, and once the cause of the problem is identified, I will follow up on it here. Thank you for your feedback.

closed 2024-11-12 by Z2Z-GuGu