sipeed/NanoKVM #134: 1.0 -> 2.1 update failed to launch web interface

mortenmoulder, 2024-11-06:
Got my NanoKVM Lite running with the latest v1.2.0 release. I can get into the web interface and sign in. When I search for an update, I see that I am at version v1.0.0 and that the latest is v2.1.0. I just updated and the device gets an IP but I cannot access the web interface. Not sure why. I can SSH as well.

Why did my brand new installation of v1.2.0 report as 1.0.0, and why isn't the latest release on GitHub under releases?

wj-xiao, 2024-11-07:
Could you try restarting the KVM and then check if the webpage is accessible?

Version 1.2.0 is the system image version, which does not update frequently.
Version 2.1.0 is the software version, and it updates more frequently. Therefore, it uses an online update mode.

wj-xiao, 2024-11-07:
Looks like the service is not running.
Please execute this command `/etc/init.d/S95nanokvm restart` to restart the service. And then check the log output.

mortenmoulder, 2024-11-07:
@wj-xiao Where would I find the log output? Here's the output of the command:

```
# /etc/init.d/S95nanokvm restart
killall: kvm_system: no process killed
killall: NanoKVM-Server: no process killed
rm: can't remove '/tmp/kvm_system': No such file or directory
rm: can't remove '/tmp/server': No such file or directory
cp: can't stat '/kvmapp/kvm_system': No error information
/etc/init.d/S95nanokvm: line 76: /tmp/kvm_system/kvm_system: not found
cp: can't stat '/kvmapp/server': No error information
/etc/init.d/S95nanokvm: line 85: /tmp/server/NanoKVM-Server: not found
OK
```

wj-xiao, 2024-11-07:
It's strange. The executable files have disappeared. It seems like the update failure caused this.
Could you try to follow [these steps](https://github.com/sipeed/NanoKVM/issues/42#issuecomment-2339549916) to update again?

mortenmoulder, 2024-11-07:
@wj-xiao Here's the output:

```
# python update-nanokvm-py
stop service...
killall: kvm_system: no process killed
killall: NanoKVM-Server: no process killed
rm: can't remove '/tmp/kvm_system': No such file or directory
rm: can't remove '/tmp/server': No such file or directory
OK
start update......
create temporary directory /root/nanokvm-cache/
download firmware...
download from https://cdn.sipeed.com/nanokvm/latest.zip?n=1730972997
download firmware done
download lib...
download lib done
update failed
[Errno 117] No error information: '/kvmapp'
killall: kvm_system: no process killed
killall: NanoKVM-Server: no process killed
rm: can't remove '/tmp/kvm_system': No such file or directory
rm: can't remove '/tmp/server': No such file or directory
cp: can't stat '/kvmapp/kvm_system': No error information
/etc/init.d/S95nanokvm: line 76: /tmp/kvm_system/kvm_system: not found
cp: can't stat '/kvmapp/server': No error information
/etc/init.d/S95nanokvm: line 85: /tmp/server/NanoKVM-Server: not found
OK
```

I tried this yesterday twice as well and got the same result. Is it possible you can try on a new NanoKVM Lite? Install the latest release on an SD card and boot it up, then update to v2.1.0 via web interface

Z2Z-GuGu, 2024-11-07:
> I tried this yesterday twice as well and got the same result. Is it possible you can try on a new NanoKVM Lite? Install the latest release on an SD card and boot it up, then update to v2.1.0 via web interface. I tried three times now and with two different SD cards (good ones).

Hi, I just tried using the new NanoKVM Lite and re burned the 1.2.0 image(with BalenaEtcher). After logging in and updating the software, I found that there were no issues

mortenmoulder, 2024-11-07:
> Hi, I just tried using the new NanoKVM Lite and re burned the 1.2.0 image(with BalenaEtcher). After logging in and updating the software, I found that there were no issues

Hmm that's very odd. I tried multiple SD cards. I do have a few NanoKVM Lite, but only tried on one of them.

Is there a log I can run `tail -f` on before pressing update, then see where it fails perhaps?

wj-xiao, 2024-11-07:
It seems to be a permissions issue. Moving `/kvmapp` directory failed.
You could try deleting the `/kvmapp` directory first to see if it can be deleted properly.
Then, run the python script again to check if the update works correctly.

Alternatively, check if you have the permissions to create files in the `/` root directory.

wj-xiao, 2024-11-07:
Another possibility is that the installation package is corrupted.
You could [download it](https://cdn.sipeed.com/nanokvm/latest.zip?n=1730972997) and extract it to check if everything is intact.

mortenmoulder, 2024-11-07:
@wj-xiao Trying to do anything with /kvmapp simply says `can't stat '/kvmapp': No error information` so that's quite weird.

If I make a brand new SD card with the v1.2.0 release, is there a manual update process? Would that be the Python script? If that's the case, then I can monitor the output of the command and see what fails

wj-xiao, 2024-11-07:
Are you using root/root to ssh into kvm? Looks like there is no permission to modify kvmapp.
The python script is a manual process. It is failed when trying to delete the old /kvmapp directory.

mortenmoulder, 2024-11-07:
> Are you using root/root to ssh into kvm? Looks like there is no permission to modify kvmapp. The python script is a manual process. It is failed when trying to delete the old /kvmapp directory.

Yes I'm using root/root. So if I go back to the release from Github, I should be able to update manually using the Python script?

wj-xiao, 2024-11-07:
Yes. You can also try the prerelease version 1.2.1.

mortenmoulder, 2024-11-07:
@wj-xiao That seems to be working. Very odd. I grabbed the 1.2.1 release, SSH'd into the KVM, and then ran the update-nanokvm.py script. I got quite a lot of errors, but I am now on 2.1.0 based on the UI :+1:

```
# pwd
/root
# mkdir update
# cd update
# vim update-nanokvm.py
# python update-nanokvm.py
stop service...
killall: kvm_stream: no process killed
rm: can't remove '/tmp/kvm_stream': No such file or directory
OK
start update......
create temporary directory /root/nanokvm-cache/
download firmware...
download from https://cdn.sipeed.com/nanokvm/latest.zip?n=1730983406
download firmware done
download lib...
download lib done
change permissions done
update to 2.1.0 success.
restart service
the nanokvm will reboot
killall: kvm_system: no process killed
killall: kvm_stream: no process killed
killall: NanoKVM-Server: no process killed
rm: can't remove '/tmp/kvm_system': No such file or directory
rm: can't remove '/tmp/kvm_stream': No such file or directory
rm: can't remove '/tmp/server': No such file or directory
OK
Error loading shared library libmaixcam_lib.so: No such file or directory (needed by /tmp/server/dl_lib/libkvm.so)
rm: can't remove '/etc/init.d/S02udisk': No such file or directory
Error relocating /tmp/server/dl_lib/libkvm.so: _Z11mmf_vi_initv: symbol not found
Error relocating /tmp/server/dl_lib/libkvm.so: _Z24mmf_invert_format_to_mmfi: symbol not found
Error relocating /tmp/server/dl_lib/libkvm.so: _Z16mmf_vi_frame_popiPPvPiS1_S1_S1_: symbol not found
Error relocating /tmp/server/dl_lib/libkvm.so: _ZNSt12experimental10filesystem2v118create_directoriesERKNS1_7__cxx114pathE: symbol not found
Error relocating /tmp/server/dl_lib/libkvm.so: _Z12mmf_venc_popiP12mmf_stream_t: symbol not found
Error relocating /tmp/server/dl_lib/libkvm.so: _Z15mmf_set_wb_modeii: symbol not found
Error relocating /tmp/server/dl_lib/libkvm.so: _Z15mmf_set_exptimeij: symbol not found
Error relocating /tmp/server/dl_lib/libkvm.so: _Z13mmf_venc_pushiPhiii: symbol not found
Error relocating /tmp/server/dl_lib/libkvm.so: _Z14mmf_try_deinitb: symbol not found
Error relocating /tmp/server/dl_lib/libkvm.so: _Z13mmf_set_againij: symbol not found
Error relocating /tmp/server/dl_lib/libkvm.so: _ZNSt12experimental10filesystem2v17__cxx114path14_M_split_cmptsEv: symbol not found
Error relocating /tmp/server/dl_lib/libkvm.so: _Z18mmf_set_vi_hmirrorib: symbol not found
Error relocating /tmp/server/dl_lib/libkvm.so: _Z29mmf_enc_jpg_push_with_qualityiPhiiii: symbol not found
Error relocating /tmp/server/dl_lib/libkvm.so: _Z15mmf_enc_jpg_popiPPhPi: symbol not found
Error relocating /tmp/server/dl_lib/libkvm.so: _ZNSt12experimental10filesystem2v17__cxx1128recursive_directory_iteratorD1Ev: symbol not found
Error relocating /tmp/server/dl_lib/libkvm.so: _Z12mmf_set_lumaij: symbol not found
Error relocating /tmp/server/dl_lib/libkvm.so: _Z16mmf_get_exp_modei: symbol not found
Error relocating /tmp/server/dl_lib/libkvm.so: _Z15mmf_get_exptimeiPj: symbol not found
Error relocating /tmp/server/dl_lib/libkvm.so: _ZNSt12experimental10filesystem2v19file_sizeERKNS1_7__cxx114pathE: symbol not found
Error relocating /tmp/server/dl_lib/libkvm.so: _ZNSt12experimental10filesystem2v17__cxx1128recursive_directory_iteratorppEv: symbol not found
Error relocating /tmp/server/dl_lib/libkvm.so: _Z12mmf_get_lumaiPj: symbol not found
Error relocating /tmp/server/dl_lib/libkvm.so: _Z13mmf_get_againiPj: symbol not found
Error relocating /tmp/server/dl_lib/libkvm.so: _Z16mmf_enc_jpg_freei: symbol not found
Error relocating /tmp/server/dl_lib/libkvm.so: _Z16mmf_get_vi_vflipiPb: symbol not found
Error relocating /tmp/server/dl_lib/libkvm.so: _Z18mmf_vi_chn_is_openi: symbol not found
Error relocating /tmp/server/dl_lib/libkvm.so: _Z13mmf_venc_freei: symbol not found
Error relocating /tmp/server/dl_lib/libkvm.so: _Z13mmf_vi_deinitv: symbol not found
Error relocating /tmp/server/dl_lib/libkvm.so: _Z20mmf_del_venc_channeli: symbol not found
Error relocating /tmp/server/dl_lib/libkvm.so: _Z8mmf_initv: symbol not found
Error relocating /tmp/server/dl_lib/libkvm.so: _ZNSt12experimental10filesystem2v18absoluteERKNS1_7__cxx114pathES5_: symbol not found
Error relocating /tmp/server/dl_lib/libkvm.so: _ZNKSt12experimental10filesystem2v17__cxx1118directory_iteratordeEv: symbol not found
Error relocating /tmp/server/dl_lib/libkvm.so: _ZNSt12experimental10filesystem2v114symlink_statusERKNS1_7__cxx114pathE: symbol not found
Error relocating /tmp/server/dl_lib/libkvm.so: _ZNSt12experimental10filesystem2v17__cxx1118directory_iteratorppEv: symbol not found
Error relocating /tmp/server/dl_lib/libkvm.so: _ZNSt12experimental10filesystem2v17__cxx1128recursive_directory_iteratorC1ERKNS2_4pathENS1_17directory_optionsEPSt10error_code: symbol not found
Error relocating /tmp/server/dl_lib/libkvm.so: _ZNKSt12experimental10filesystem2v17__cxx1128recursive_directory_iteratordeEv: symbol not found
Error relocating /tmp/server/dl_lib/libkvm.so: _ZNSt12experimental10filesystem2v116create_directoryERKNS1_7__cxx114pathE: symbol not found
Error relocating /tmp/server/dl_lib/libkvm.so: _Z16mmf_set_vi_vflipib: symbol not found
[... 75 lines cut ...]
Error relocating /tmp/server/dl_lib/libkvm.so: _ZNKSt12experimental10filesystem2v17__cxx114path18has_root_directoryEv: symbol not found
Error relocating /tmp/server/dl_lib/libkvm.so: _ZNSt12experimental10filesystem2v16removeERKNS1_7__cxx114pathE: symbol not found
[... cut ...]
Error relocating /tmp/server/dl_lib/libkvm.so: _ZNSt12experimental10filesystem2v19canonicalERKNS1_7__cxx114pathES5_: symbol not found
Error relocating /tmp/server/dl_lib/libkvm.so: _ZNSt12experimental10filesystem2v110remove_allERKNS1_7__cxx114pathE: symbol not found
Error relocating /tmp/server/dl_lib/libkvm.so: _ZNSt12experimental10filesystem2v112current_pathB5cxx11Ev: symbol not found
mkdir: can't create directory '/etc/kvm': File exists
# beta
-- [E] write failed
oled not exist
192.168.1.156
kvm_sys_state.res = 720
kvm_sys_state.type = 2
kvm_sys_state.qlty = 2
Connection to 192.168.1.156 closed by remote host.
```

wj-xiao, 2024-11-08:
Glad to see that.
The log is normal. Just ignore it.
These error messages appear on the version 2.1.0 first run because 2.1.0 introduced the so libraries.

closed 2024-11-08 by wj-xiao