sipeed/NanoKVM #136: HDD Light - not working.

victordk, 2024-11-06:
Hi 

Do anybody have the HDD activity light working on the web interface?

I run the full version updated with 2.10 firmware and KVM-A passthrough with the HDD light working on the PC case.

Z2Z-GuGu, 2024-11-07:
Thanks for your question. Due to the limited number of ATX wire cores, only PWR_LED, PWR_SW, and RST_SW signals can be processed in NanoKVM, and the status of HDD LED cannot be received by NanoKVM.And I found that there are missing parts in the description on the wiki, and I will supplement them as soon as possible. Thank you again.

closed 2024-11-07 by Z2Z-GuGu

stuartm, 2025-01-12:
The answer doesn't explain why, if HDD Activity LED cannot be received there is an icon for it in the GUI labelled "HDD LED"? Shouldn't this be removed from the UI if it cannot show anything?

stuartm, 2025-01-12:
> Shouldn't this be removed from the UI if it cannot show anything?

My bad, I see this has already been removed in the latest firmware.