sipeed/NanoKVM #137: HDMI input not working after update to 2.1.0

remie1529, 2024-11-06:
I can't get the HDMI input working anymore after update to 2.1.0

2024-08-17-18-13-713161.img

What I did try:
- Reboot with HDMI connected.
- Reboot without HDMI connected.
- New HDMI cable
- connect to other device.

wj-xiao, 2024-11-07:
It seems like H.264 does not work. Could you try switching to MJPEG to see if there's any video output?

[image]

wj-xiao, 2024-11-07:
It might also be a browser compatibility issue. Could you please provide your browser information?

remie1529, 2024-11-07:
When selecting MJPEG it works on chrome.
I use chrome: Version 130.0.6723.117 (Official Build) (64-bit)
I did also try it on Firefox, there it works perfectly with H.264 and MJPEG Version 132.0.1 (64-bits)

So i'm happy for now, thank you for your help!

wj-xiao, 2024-11-07:
Glad to hear that. 
Chrome should be working fine. I guess it was caused by the browser cache. You can also try clearing the browser cache or using incognito mode to open the page in chrome.

remie1529, 2024-11-07:
It's indeed the browser cache, incognito mode it did also work great.

So thank you again!!

wj-xiao, 2024-11-08:
Reopen it temporarily for people who are also encountering this issue.

tmysl-pton, 2024-11-12:
im having this issue in safari/Firefox and chrome, running 2.1.1 on using multiple different cables and devices

tmysl-pton, 2024-11-12:
```
Nov 12 15:43:32 kvm-6832 local5.err : [VPSS-ERR] cvi_vpss.c:911:CVI_VPSS_GetChnFrame(): Grp(0) Chn(0) get chn frame fail
Nov 12 15:43:32 kvm-6832 kern.err kernel: [  576.899946] base_get_chn_buffer:262(): Mod(VPSS) Grp(0) Chn(0), jobs wait(0) work(0) done(0)
Nov 12 15:43:33 kvm-6832 local5.err : [VPSS-ERR] cvi_vpss.c:911:CVI_VPSS_GetChnFrame(): Grp(0) Chn(0) get chn frame fail
Nov 12 15:43:33 kvm-6832 kern.err kernel: [  577.924003] base_get_chn_buffer:262(): Mod(VPSS) Grp(0) Chn(0), jobs wait(0) work(0) done(0)
Nov 12 15:43:35 kvm-6832 local5.err : [VPSS-ERR] cvi_vpss.c:911:CVI_VPSS_GetChnFrame(): Grp(0) Chn(0) get chn frame fail
Nov 12 15:43:35 kvm-6832 kern.err kernel: [  578.948017] base_get_chn_buffer:262(): Mod(VPSS) Grp(0) Chn(0), jobs wait(0) work(0) done(0)
Nov 12 15:43:36 kvm-6832 local5.err : [VPSS-ERR] cvi_vpss.c:911:CVI_VPSS_GetChnFrame(): Grp(0) Chn(0) get chn frame fail
Nov 12 15:43:36 kvm-6832 kern.err kernel: [  579.972069] base_get_chn_buffer:262(): Mod(VPSS) Grp(0) Chn(0), jobs wait(0) work(0) done(0)
Nov 12 15:43:37 kvm-6832 local5.err : [VPSS-ERR] cvi_vpss.c:911:CVI_VPSS_GetChnFrame(): Grp(0) Chn(0) get chn frame fail
Nov 12 15:43:37 kvm-6832 kern.err kernel: [  580.996098] base_get_chn_buffer:262(): Mod(VPSS) Grp(0) Chn(0), jobs wait(0) work(0) done(0)
Nov 12 15:43:38 kvm-6832 local5.err : [VPSS-ERR] cvi_vpss.c:911:CVI_VPSS_GetChnFrame(): Grp(0) Chn(0) get chn frame fail
Nov 12 15:43:38 kvm-6832 kern.err kernel: [  582.020189] base_get_chn_buffer:262(): Mod(VPSS) Grp(0) Chn(0), jobs wait(0) work(0) done(0)
Nov 12 15:43:39 kvm-6832 local5.err : [VPSS-ERR] cvi_vpss.c:911:CVI_VPSS_GetChnFrame(): Grp(0) Chn(0) get chn frame fail
Nov 12 15:43:39 kvm-6832 kern.err kernel: [  583.044183] base_get_chn_buffer:262(): Mod(VPSS) Grp(0) Chn(0), jobs wait(0) work(0) done(0)
Nov 12 15:43:40 kvm-6832 local5.err : [VPSS-ERR] cvi_vpss.c:911:CVI_VPSS_GetChnFrame(): Grp(0) Chn(0) get chn frame fail
Nov 12 15:43:40 kvm-6832 kern.err kernel: [  584.068216] base_get_chn_buffer:262(): Mod(VPSS) Grp(0) Chn(0), jobs wait(0) work(0) done(0)
Nov 12 15:43:41 kvm-6832 local5.err : [VPSS-ERR] cvi_vpss.c:911:CVI_VPSS_GetChnFrame(): Grp(0) Chn(0) get chn frame fail
Nov 12 15:43:41 kvm-6832 kern.err kernel: [  585.092277] base_get_chn_buffer:262(): Mod(VPSS) Grp(0) Chn(0), jobs wait(0) work(0) done(0)
Nov 12 15:43:42 kvm-6832 local5.err : [VPSS-ERR] cvi_vpss.c:911:CVI_VPSS_GetChnFrame(): Grp(0) Chn(0) get chn frame fail
Nov 12 15:43:42 kvm-6832 kern.err kernel: [  586.116313] base_get_chn_buffer:262(): Mod(VPSS) Grp(0) Chn(0), jobs wait(0) work(0) done(0)
Nov 12 15:43:43 kvm-6832 local5.err : [VPSS-ERR] cvi_vpss.c:911:CVI_VPSS_GetChnFrame(): Grp(0) Chn(0) get chn frame fail
Nov 12 15:43:43 kvm-6832 kern.err kernel: [  587.140345] base_get_chn_buffer:262(): Mod(VPSS) Grp(0) Chn(0), jobs wait(0) work(0) done(0)
Nov 12 15:43:44 kvm-6832 local5.err : [VPSS-ERR] cvi_vpss.c:911:CVI_VPSS_GetChnFrame(): Grp(0) Chn(0) get chn frame fail
Nov 12 15:43:44 kvm-6832 kern.err kernel: [  588.164397] base_get_chn_buffer:262(): Mod(VPSS) Grp(0) Chn(0), jobs wait(0) work(0) done(0)
Nov 12 15:43:45 kvm-6832 local5.err : [VPSS-ERR] cvi_vpss.c:911:CVI_VPSS_GetChnFrame(): Grp(0) Chn(0) get chn frame fail
Nov 12 15:43:45 kvm-6832 kern.err kernel: [  589.188438] base_get_chn_buffer:262(): Mod(VPSS) Grp(0) Chn(0), jobs wait(0) work(0) done(0)
Nov 12 15:43:46 kvm-6832 local5.err : [VPSS-ERR] cvi_vpss.c:911:CVI_VPSS_GetChnFrame(): Grp(0) Chn(0) get chn frame fail
Nov 12 15:43:46 kvm-6832 kern.err kernel: [  590.212492] base_get_chn_buffer:262(): Mod(VPSS) Grp(0) Chn(0), jobs wait(0) work(0) done(0)
# 
```

output from` var/log/message`

Blackheat45, 2024-11-12:
> im having this issue in safari/Firefox and chrome, running 2.1.1 on using multiple different cables and devices tested with:
> 
> M4 mac MIni M1 mac Studio Raspberry Pi 4 Lenovo x13 laptop
> 
> all with the same behavior, video modes don't seem to fix or change it, re-flashing card has not fixed it either.

I'm also having this issue. Works for a few seconds then it freezes. Have to restart the KVM. One was working fine before update and now not working. Another has never worked without freezing.

mouie, 2024-12-16:
Chiming in with my anecdotal experience. I was not having this issue with a 3 meter ethernet cat6 cable (Folishine brand from Amazon). After changing to a 1 meter ethernet cat 6 cable (also Folishine brand from Amazon), this same issue arose. 
That is, with 1 meter eth, H.264 freezes up and is impossible to use; but, with 3 meter eth, H.264 works fine. For clarity, MJPEG works with both lenghts.

For context, I am facing a separate issue with my NanoKVM's ethernet connection sporadically dropping, and was testing the shorter ethernet cable per suggestion here: https://github.com/sipeed/NanoKVM/issues/7#issuecomment-2298274176