sipeed/NanoKVM #138: Add suppot for forwarding serial console over telnet

sidcha, 2024-11-07:
Having the serial console in a tmux pane is far more convenient than the web interface. For now, I ssh into NanoKVM and start picocom manually but it would be nice to have something like [set2net](https://linux.die.net/man/8/ser2net).

Z2Z-GuGu, 2024-11-07:
[ser2net.zip](https://github.com/user-attachments/files/17655697/ser2net.zip)

@sidcha Hello, I have compiled the risc-v version of ser2net. Please unzip it in the `/var/sbin/ ` directory and grant it executable permission. If you have free time, please test it in your environment, after passing the test, I will include this tool in the next version of the image.

sidcha, 2024-11-07:
Thanks @Z2Z-GuGu. Any chance you can give me a static binary?

```
# sha256sum /var/sbin/ser2net 
4eeb8824ab77d9d3dca85e3b66cb9fe0cf54cf40247d1c1fadb399bd82c9c85f  /var/sbin/ser2net
# ls -la /var/sbin/ser2net 
-rwx--x--x    1 root     root        107200 Nov  7 16:17 /var/sbin/ser2net
# /var/sbin/ser2net 
Error loading shared library libgensio.so.0: No such file or directory (needed by /var/sbin/ser2net)
Error relocating /var/sbin/ser2net: sel_alloc_selector_thread: symbol not found
Error relocating /var/sbin/ser2net: gensio_str_in_auxdata: symbol not found
Error relocating /var/sbin/ser2net: gensio_acc_get_type: symbol not found
Error relocating /var/sbin/ser2net: gensio_to_sergensio: symbol not found
Error relocating /var/sbin/ser2net: gensio_close: symbol not found
Error relocating /var/sbin/ser2net: sergensio_dtr: symbol not found
[...]
```

If not, please pack all the shared objects that ser2net depends on.

Z2Z-GuGu, 2024-11-12:
It seems that there are some missing dependency libraries. I will try compiling these missing dependencies at a later time

mangelajo, 2025-11-25:
I am interested in this.