sipeed/NanoKVM #139: Paste Content needs to support all symbols and letters

nwgat, 2024-11-07:
trying to copy some powershell code like this
`$product_name = (Get-WmiObject Win32_BaseBoard).Product; echo "Product Name: $product_name";`
but it ends up very differently on the other machine  (see notepad in screenshot)
[image]

wj-xiao, 2024-11-08:
Could you please provide information about the input method on your remote host?

nwgat, 2024-11-08:
> Could you please provide information about the input method on your remote host?

i almost figured it out, my other system was set to united states instead of united kingdom so it i set them to both united kingdom and norwegian bokmål as the keyboard layout it seem to copy the correct text, but the keys for `@` dont seem to work under nanokvm, if i connect via rustdesk i can use `alt gr + 2` for `@`

closed 2024-11-08 by nwgat