sipeed/NanoKVM #141: Headscale Support

m4us1ne, 2024-11-07:
Hey, I have setup a [Headscale](https://github.com/juanfont/headscale) instance and was wondering if it could be supported it via the Tailscale Integration.

This could be possible via changing the tailscale login command in https://github.com/sipeed/NanoKVM/blob/74a303bd5cbb58f9d8ddd81abaaf4919dbbfb71b/server/service/network/tailscale.go#L145 to support the --login-server= argument.

wj-xiao, 2024-11-08:
Yes. PR is welcome.
For this request, I think you can execute the command manually.

```shell
# ssh into kvm

# restart tailscale
/etc/init.d/S98tailscaled restart

# login
tailscale login --login-server=argument
```

m4us1ne, 2024-11-08:
I already did that :)

PatrickHuetter, 2024-11-09:
+1 for that!