sipeed/NanoKVM #144: Add control to change hostname

jtenniswood, 2024-11-08:
This would be useful as it's what shows up on Tailscale and could be useful for local dns lookups

wilsonnkwan, 2024-12-18:
I am not sure if this will help. Within the /etc/init.d/, there is a file call S10uuid. There are a few lines of the code which determines if some file in the boot directory exist, one of whichc is the /boot/hostname. If the file does not exist, then the hostname will follow kvm-1st4CharacterGeneratedWithSHA512Sum. This /boot/hostname file does not exist by default.

BUT, if you create one, and write within the file only in alphanumeric no space small caps like "helloitsme", it will change the hostname to the desired "helloitsme" everytime when boot.

jtenniswood, 2025-01-06:
Awesome thanks, yes that works. Would be nice to have that within the UI, but this is a good work around.

stvorl, 2025-01-10:
There also would be good to have an option to advertise hostname to DHCP server (so, it can register device in DNS, for example).
I did workaround for me, changed line 40 of /etc/init.d/S30eth:
```
    else
        #(udhcpc -i eth0 -t 10 -T 1 -A 5 -b -p /run/udhcpc.eth0.pid) &
        (udhcpc -i eth0 -t 10 -T 1 -A 5 -b -x hostname:$(cat /etc/hostname) -p /run/udhcpc.eth0.pid) &
    fi
```
But, of course, it is crude solution.
Maybe to include some file in /boot, like "/boot/dhcp-extra-options", that would be included here, in udhcpc command line, OEM?

AJolly, 2025-01-24:
oh thanks, I was changing /etc/hostname and being annoyed it would not save

Thehoney4you, 2025-04-04:
I found the file wilsonnkwan is referring to and changing it only survives 1 reboot. :(

wj-xiao, 2025-05-28:
Added in version [2.2.6](https://github.com/sipeed/NanoKVM/blob/main/CHANGELOG.md#226-c83dc55-2025-04-17).