sipeed/NanoKVM #149: Tailscale authentication is lost by updating the NanoKVM application

Bengt, 2024-11-12:
When I have Tailscale connected and update the NanoKVM application via the web interface, Tailscale is no longer connected:

```# tailscale status
failed to connect to local tailscaled; it doesn't appear to be running
```

When I reauthenticate using the wizard in the web interface, Tailscale works again.

nwgat, 2024-11-12:
i also ran into this issue

jasonafriedman, 2024-11-13:
I am running into the issue where authentication is being lost without updating after a couple hours. I am using nanokvm 2.1.1.

alexoughton, 2024-11-13:
I'm having the same issues with both the authentication and the oom-kill. It looks like it's important to keep tailscale at version 1.70.0 and not update it to 1.76.6 in order to avoid this issue. I was also able to downgrade back to 1.70.0 on a KVM experiencing this problem and it went away.

mindstorms6, 2024-11-13:
@alexoughton how did you force Tailscale back down to 1.70.0?

alexoughton, 2024-11-13:
@mindstorms6 Some of these steps may not have been necessary, but this is what I had done while investigating the problem:

1. I moved `/etc/init.d/S98tailscaled` away to `~` to disable it, and rebooted the KVM.
2. I moved `/usr/bin/tailscale` to `/usr/bin/tailscale-old`, and moved `/usr/sbin/tailscaled-old` to `/usr/bin/tailscaled-old`.
3. I went to the tailscale option in the web interface. It told me it wasn't installed, so I hit the "install" button. This shows an error but...
4. ...the correct versions of `/usr/bin/tailscale` and `/usr/sbin/tailscaled` were now in place. I moved `/etc/init.d/S98tailscaled` back into place and ran `/etc/init.d/S98tailscaled restart`.
5. Back in the web interface, I was now able to re-register to tailscale, and the admin interface for that showed I was back at 1.70.0.

As I said, this may have been more steps than actually needed, but stumbling through this as above got me into the right place.

mindstorms6, 2024-11-13:
@alexoughton you're the best - that also worked for me - thank you!

jasonafriedman, 2024-11-13:
For me removing the old packages and reinstalling through webui worked.
`rm /usr/bin/tailscale` and `rm /usr/sbin/tailscaled`

crustovsky, 2024-11-15:
This worked @alexoughton thanks, there seems to be mistake in step 2 though, i think it should state to "and moved `/usr/sbin/tailscaled` to `/usr/sbin/tailscaled-old`"

alexoughton, 2024-12-04:
Note that you will also need to run `tailscale set --auto-update=false` to prevent auto-updates from occurring again until this is fixed.

ironicbadger, 2024-12-05:
@alexoughton where does the fix need to come from? sipeed side? or Tailscale?

alexoughton, 2024-12-05:
@ironicbadger I haven't at all investigated why tailscale is getting oom-killed, only found this workaround. I'd say sipeed need to do that initial investigation, and then feed back to tailscale if it's just a memory leak on their side.

ironicbadger, 2024-12-05:
Lmk as I am a Tailscale employee

wj-xiao, 2025-01-14:
Most issues are caused by insufficient memory.

Tailscale's memory usage grows with longer connection times and more connected clients. This can eventually lead to OOM errors, causing the Tailscale process to terminate and potentially lose login credentials.

- In image version 1.3.0, we increased the memory allocation from 128MB to 158MB.
- In application version 2.1.5, we optimized memory usage by setting GOMEMLIMIT.

These two changes significantly reduce the occurrence of OOM errors. Please try to update to the latest version.

We are actively working on further solutions, such as using swap, persisting login information, providing more software support, etc.

wj-xiao, 2025-01-15:
@alexoughton 
check these two options in the settings:
[image]

And check this [document](https://wiki.sipeed.com/hardware/en/kvm/NanoKVM/system/flashing.html) about flashing image.

ItzDerock, 2025-01-15:
Hi, just updated to this version. Is the default memory limit supposed to be 1024MiB? Seems unreasonable given the device only has 158MB to begin with, so just wanted to check if this was intentional or not.

```
# /etc/init.d/S98tailscaled start
GOMEMLIMIT set to 1024MiB
Starting tailscaled[1.70.0]: OK
```

That limit is defined in the `/etc/init.d/S98tailscaled` file:
```
                if [ -f /etc/kvm/GOMEMLIMIT ]; then
                     value=$(cat /etc/kvm/GOMEMLIMIT)
                     export GOMEMLIMIT="${value}MiB"
                else
                     export GOMEMLIMIT=1024MiB
                fi
```

wj-xiao, 2025-01-15:
@ItzDerock 
When this feature is disabled, the default value is set to 1024M. This value may have some negative impacts, but there's no difference compared to older versions.

We'll monitor it for a while. In future versions, we may enable this feature by default when Tailscale startup and set the value to 50M.

alexoughton, 2025-01-15:
@wj-xiao My image version is listed as `2024-08-17-18-13-713161.img`. If updating the image can't be done via the web UI or via SSH then unfortunately I'm not going to be able to do so (one of my KVMs is thousands of miles away from me so hands-on access is impossible).

With that said, so far the one KVM I updated/configured as above is stable so far. Remaining connected to Tailscale, about 49MB memory free, and no oom-kill in dmesg.

diamondiron, 2025-01-16:
Updated to latest version and now I cannot view screen. Seems NanoKVM is completely broken. When I login to web ui and I'm also in SSH, the console just fills with this message repeatedly.

```
# cd /etc/kvm
# python update-nanokvm.py ^C

# python --version
Python 3.11.6
# python ./update-nanokvm.py 
stop service...
killall: kvm_system: no process killed
killall: NanoKVM-Server: no process killed
rm: can't remove '/tmp/kvm_system': No such file or directory
rm: can't remove '/tmp/server': No such file or directory
OK
start update......
create temporary directory /root/.kvm-cache/
download firmware...
download from https://cdn.sipeed.com/nanokvm/latest.zip?n=1737019612
download firmware done
download lib...
download lib done
change permissions done
update to 2.1.5 success.
restart service
the nanokvm will reboot
killall: kvm_system: no process killed
killall: NanoKVM-Server: no process killed
rm: can't remove '/tmp/kvm_system': No such file or directory
rm: can't remove '/tmp/server': No such file or directory
OK
rm: can't remove '/etc/init.d/S02udisk': No such file or directory
rm: can't remove '/etc/init.d/S30wifi': No such file or directory
mkdir: can't create directory '/kvmapp/kvm': File exists
mkdir: can't create directory '/etc/kvm': File exists
/etc/kvm/hw exist
hw = beta!
# sh: write error: Resource busy
sh: write error: Resource busy
sh: write error: Resource busy
sh: write error: Resource busy
sh: write error: Resource busy
rmmod: can't unload module 'i2c_gpio': Resource temporarily unavailable
rmmod: can't unload module 'i2c_algo_bit': Resource temporarily unavailable
insmod: can't insert '/mnt/system/ko/i2c-algo-bit.ko': File exists
insmod: can't insert '/mnt/system/ko/i2c-gpio.ko': File exists
killall: NanoKVM-Server: no process killed
rm: can't remove '/tmp/server': No such file or directory
OK
killall: NanoKVM-Server: no process killed
rm: can't remove '/tmp/server': No such file or directory
beta
oled exist
192.168.1.39
rm: can't remove '/etc/kvm/wifi_exist': No such file or directory
[SAMPLE_COMM_SNS_ParseIni]-2204: Parse /mnt/data/sensor_cfg.ini
[parse_source_devnum]-1797: devNum =  1
[parse_sensor_name]-1878: sensor =  LONTIUM_LT6911_2M_60FPS_8BIT
[parse_sensor_busid]-1907: bus_id =  4
[parse_sensor_i2caddr]-1918: sns_i2c_addr =  ff
[parse_sensor_mipidev]-1929: mipi_dev =  0
[parse_sensor_laneid]-1940: Lane_id =  2, 4, 3, 1, 0
[parse_sensor_pnswap]-1951: pn_swap =  0, 0, 0, 0, 0
[parse_sensor_mclken]-1973: mclk_en =  0
[parse_sensor_mclk]-1984: mclk =  0
try release sys ok
[_mmf_init]-1142: maix multi-media version:67a90e2-musl_riscv64
[SAMPLE_COMM_SNS_ParseIni]-2204: Parse /mnt/data/sensor_cfg.ini
[parse_source_devnum]-1797: devNum =  1
[parse_sensor_name]-1878: sensor =  LONTIUM_LT6911_2M_60FPS_8BIT
[... 194 lines cut ...]
[2025-01-16 09:27:48] [error] [cgo.go:93] failed to read kvm image: -1
[2025-01-16 09:27:49] [error] [cgo.go:93] failed to read kvm image: -1
[2025-01-16 09:27:50] [error] [cgo.go:93] failed to read kvm image: -1
[2025-01-16 09:27:51] [error] [cgo.go:93] failed to read kvm image: -1
[2025-01-16 09:27:52] [error] [cgo.go:93] failed to read kvm image: -1
[2025-01-16 09:27:53] [error] [cgo.go:93] failed to read kvm image: -1
^C

# [2025-01-16 09:27:55] [error] [cgo.go:93] failed to read kvm image: -1
[2025-01-16 09:27:56] [error] [cgo.go:93] failed to read kvm image: -1
/etc/[2025-01-16 09:27:57] [error] [cgo.go:93] failed to read kvm image: -1
init.d/[2025-01-16 09:27:58] [error] [cgo.go:93] failed to read kvm image: -1
# /etc/init.d/S95nanokvm sto[2025-01-16 09:27:59] [error] [cgo.go:93] failed to read kvm image: -1
p

Received signal: terminated
OK
# [2025-01-16 09:28:00] [error] [cgo.go:93] failed to read kvm image: -1
[2025-01-16 09:28:01] [error] [cgo.go:93] failed to read kvm image: -1
maix multi-media driver destroyed.
ISP Vipipe(0) Free pa(0x8ace8000) va(0x0x3ff3626000)
-- [E] Trigger signal, code:SIGSEGV(11)!
```

@wj-xiao

wj-xiao, 2025-01-17:
@diamondiron Please check this [FAQ](https://wiki.sipeed.com/hardware/en/kvm/NanoKVM/faq.html#No-Display-After-Logging-into-the-Browser-Interface).

alexoughton, 2025-01-21:
The device I updated has remained stable and online since the update. I have extended testing to three more devices.

shepherdjerred, 2025-01-21:
I've also had no issues after the latest firmware update

alexoughton, 2025-01-24:
All the devices I have updated as above have remained stable and online. Looks like this may be resolved.

alexoughton, 2025-03-02:
It appears this problem has returned. The most recent update is no longer applying proper memory limits to tailscale.

WibowoS26, 2025-07-04:
> It appears that issues staying connected to tailscale may have returned with the latest update. I'm not certain, but it may be that the memory limits are no longer being applied.

Look like i had same issues, nanokvm version 2.2.9 getting kicked from tailscale network, only when i restart the device and reinstall tailscale it can be connected to tailscalr network again