sipeed/NanoKVM #152: Absolute mouse mode with multiple monitors

TheOnlyMatty, 2024-11-14:
For my use case I am using 2 NanoKVM's connected to the same system. This allows for having two monitors open at the same time.

This leads to a problem when using absolute mouse mode, when moving the mouse on the 2nd monitor it will move the mouse on the primary monitor as opposed to the monitor the KVM is displaying.

When using relative mouse movement the mouse can be moved freely between screens however this comes with the disadvantage of input delay.

As the KVM is unaware of any other monitors a possible solution would be to allow the user to input a offset so the absolute position is offset by the dimensions of the primary monitor.

dormancygrace, 2026-10-11:
NanoKVM OS is an independent community firmware for the SG2002 NanoKVM (Cube, Lite, PCIe): a full SD image on Alpine Linux, updated with apk. The current release is an alpha; PCIe is the validated platform, the Cube and Lite profiles are included but not yet validated on hardware. I maintain it, so take this as a biased pointer: https://github.com/dormancygrace/NanoKVM-OS

boeltjen, 2026-10-29:
FYI, this is a Windows thing, not a kvm one.  There's no possibility of an offset in absolute because it's only relative to the primary display.  It seems that Windows itself relies on relative/movement mode to get across it's displays.