sipeed/NanoKVM #154: Feature Request: Update Hostname in DHCP Records

ShoelaceMan, 2024-11-14:
First off, thank you for creating the NanoKVM—it's a fantastic product! 

I've noticed that the NanoKVM doesn’t update its hostname in DHCP after obtaining an address, which would be a helpful feature, especially in network environments where hostname tracking is essential. 

The device uses `udhcpc` as the DHCP client, which includes an option (`-F`) for updating the hostname in the DHCP server's records. Here is a command that successfully achieves this:

```bash
udhcpc -i eth0 -t 3 -T 1 -A 5 -b -p /run/udhcpc.eth0.pid -F ${HOSTNAME}
```

Currently, `udhcpc` runs with these options: `-i eth0 -t 3 -T 1 -A 5 -b -p /run/udhcpc.eth0.pid`

But I tested and noted that it works as expected with the additional `-F ${HOSTNAME}` flags:
```bash
# ps aux | grep [d]hcpc
868 root     udhcpc -i eth0 -t 3 -T 1 -A 5 -b -p /run/udhcpc.eth0.pid -F kvm-b6c9
```

By setting the `-F ${HOSTNAME}` flag dynamically, the device will automatically update its hostname in the DHCP records, improving its integration in network environments without needing further configuration.

**Suggested Implementation**:
Please consider adding `-F ${HOSTNAME}` to the `udhcpc` command in the initialization files for NanoKVM Full. This should allow each device to reflect its hostname in the DHCP server. Here are some examples that apply the changes, which I've deployed on all of my NanoKVM's:
`sed -i 's#(udhcpc -i eth0 -t 10 -T 1 -A 5 -b -p /run/udhcpc.eth0.pid)#(udhcpc -i eth0 -t 10 -T 1 -A 5 -b -p /run/udhcpc.eth0.pid -F ${HOSTNAME})#' /etc/init.d/S30eth`
`sed -i 's#udhcpc -i eth0 -t 3 -T 1 -A 5 -b -p /run/udhcpc.eth0.pid#udhcpc -i eth0 -t 3 -T 1 -A 5 -b -p /run/udhcpc.eth0.pid -F ${HOSTNAME}#' /etc/init.d/S30eth`

Now, if this gets implemented, it might be a good time to look at this feature request as well:
https://github.com/sipeed/NanoKVM/issues/144

Which seems to be as simple as echoing the inputted hostname to `/boot/hostname`

Thanks again for your attention to this request, and please let me know if I can provide further details!

wilsonnkwan, 2024-12-18:
This is assuming that the DHCP server has an assigned name within which the suggestion that you gave will import that set hostname into the system?

ShoelaceMan, 2024-12-18:
Hey,

I am not sure if I fully understand the question, but this is simply assuming that the DHCP server in question supports RFC 2136. I don’t have an exhaustive list of which DHCP servers wouldn’t support this, nor can I say definitely if this would reduce compatibility on DHCP servers that don’t support it, but it works on the DHCP server that I tested it with, which is a standard configuration windows DHCP server.

archont94, 2024-12-26:
`-F` option doesn't work with Mikrotik DHCP server. Working alternative is `udhcpc -i eth0 -t 10 -T 1 -A 5 -b -p /run/udhcpc.eth0.pid -x hostname:$HOSTNAME` instead of `udhcpc -i eth0 -t 10 -T 1 -A 5 -b -p /run/udhcpc.eth0.pid`

dormancygrace, 2026-09-19:
Hi! [NanoKVM OS](https://github.com/dormancygrace/NanoKVM-OS/releases/tag/v2.0-a2), my independent community firmware for SG2002-based NanoKVM devices, includes a change relevant to this request.

Ethernet and Wi-Fi DHCP startup now read and validate the configured hostname and pass both -x hostname:<name> and -F <name> to udhcpc. This covers hostname option 12 as well as FQDN option 81, including the distinction raised in the MikroTik comment.

[Implementation 1](https://github.com/dormancygrace/NanoKVM-OS/blob/bf7d726f502526abb127f687ca3a868487f39b25/kvmapp/system/init.d/S30eth) · [Implementation 2](https://github.com/dormancygrace/NanoKVM-OS/blob/bf7d726f502526abb127f687ca3a868487f39b25/kvmapp/system/init.d/S30wifi).

You're welcome to try **NanoKVM OS v2.0-a2** and let me know how it works on your setup!

[Release and installation details](https://github.com/dormancygrace/NanoKVM-OS/releases/tag/v2.0-a2)

Please include your board model, OS version and reproduction steps when reporting results. Thanks!