sipeed/NanoKVM #158: [FEATURE REQUEST] Turn on Mouse Jiggler when Idle

dw5, 2024-11-20:
If user has enabled Mouse jiggler feature, It will allow to monitor the remotely connected device, by preventing it from getting into sleep mode or locking itself (plus would not require installation of additional software or purchase of USB jiggler)

PiKVM has an implementation, and it would be a great addition to NanoKVM. 
It will also allow me to buy more NanoKVMs thanks to not needing to buy a mouse jiggler too.

alexislefebvre, 2024-11-25:
As a workaround, can you disable sleep mode on the monitored device?

Snazzie, 2024-11-26:
This man just wants to auto jiggle his work laptop so his teams doesnt rat on him lmao

lwbt, 2024-12-07:
> This man just wants to auto jiggle his work laptop so teams doesnt rat on him lmao

He could get that for a fraction of the price of a NanoKVM if it was that important to him.
Why trash talk a good idea and a useful feature?

Snazzie, 2024-12-08:
> > This man just wants to auto jiggle his work laptop so teams doesnt rat on him lmao
> 
> He could get that for a fraction of the price of a NanoKVM if it was that important to him. Why trash talk a good idea and a useful feature?

take a joke, party pooper.

dw5, 2024-12-16:
shout out to the spammers above

closed 2024-12-16 by dw5

RobbieL811, 2025-05-05:
Is there any way to disable this god forsaken feature? I tried disabling it in the gui. It's screwing with my games on my gaming desktop. Keeps taking over control from my controller. Disabling in the gui did nothing. Mouse continued to jiggle. I do not want this. I want my computer to sleep when it wants to and I can wake it when I need it. Hoping theres a command that can do away with this nonsense.

civictd04, 2025-05-06:
> Is there any way to disable this god forsaken feature? I tried disabling it in the gui. It's screwing with my games on my gaming desktop. Keeps taking over control from my controller. Disabling in [...]

Since this is a closed issue you might want to report it as a new issue.