sipeed/NanoKVM #159: v1.3.0 update failed

eiqnepm, 2024-11-20:
Using the UI I'm unable to update from 1.0 to 2.1.1, it just says "Update failed. Please retry." instantly each time I click confirm.

mcury1985, 2024-11-21:
To update the firmware, you need to remove the SD card and burn it manually.
more info at [https://wiki.sipeed.com/hardware/en/kvm/NanoKVM/system/flashing.html](https://wiki.sipeed.com/hardware/en/kvm/NanoKVM/system/flashing.html)

eiqnepm, 2024-11-22:
> To update the firmware, you need to remove the SD card and burn it manually. more info at https://wiki.sipeed.com/hardware/en/kvm/NanoKVM/system/flashing.html

That's what I did. v1.3.0 flash was a success but I can't update the web component using the built in update GUI.

mcury1985, 2024-11-22:
It should auto update to 2.1.1 automatically during first boot, at least mine did.
I got that to work by enabling DHCP for the network that the KVM is connected to, once it booted, the update was automatic.

eiqnepm, 2024-11-26:
Never mind, was the result of a failing microSD card.