sipeed/NanoKVM #162: Stuck on auto-update v1.3.0

paolopg, 2024-11-21:
Flashed the SD card using Balena Etcher to v1.3.0. The unit is stuck on Auto-Update screen. Tried power cycling it, no success.

mcury1985, 2024-11-22:
Make sure you have it connected to the network during first boot after update, DHCP and Internet, both need to be working on that network.

schlypel, 2024-11-26:
I have the same problem and network ist definitely connected. it says "Please connect ..." and after a second gets the DHCP IP, which it starts showing, while it says "Auto updating"
After a while it reboots and the cycle starts again.

Eisaichen, 2024-11-27:
Same here, flashed to 1.3.0 then stuck on auto-update. Why don't ship the update with the image?

paolopg, 2024-11-28:
I found a solution that worked for me. I created a separate VLAN without any geoblocking. The unit was able to "dial home" and get the update. Once updated, it works fine without any connection to the internet.

diamondiron, 2024-12-04:
I'm having the same issue, I've disabled any and all private DNS on my network but it is still happening. I just received a fresh batch of the full package. I've noticed since when I received my devices, the devices are NOT able to ping any domain names. There is no DNS server located in /etc/resolv.conf. It is able to ping IP's but not able to resolve domains. And it looks like its pulling the autoupdate image from cdn.sispeed.com, but cannot resolve it. I try adding nameserver 1.1.1.1 to /etc/resolv.conf but it removes it.

This thing is just looping, what can one do?

Eisaichen, 2024-12-05:
@diamondiron Maybe put this line under `/etc/hosts` as a temporary workaround?
```
3.33.139.32    cdn.sispeed.com
```

marshalleq, 2024-12-05:
I think that's what I did originally.  If I recall correctly, the available software to download did not include the latest available version by quite some margin and so that had to be updated via online source once it was installed.  It took me ages to get the latest version on because I think of this same problem.  I'm pretty sure I either used /etc/hosts file like suggested above OR managed to get an update file from somewhere else and placed it onto the sd card.

The real fix for this (if it's still the same issue), would be to make the latest stable version available to download from the GitHub page.

NefilimPL, 2024-12-12:
my kvm just arrived, first time running = no IP adress 
ok so mayby update will help...
after update stuck on auto updating (with IP)... And i don't have custom DNS on all devices, only on a few

schlypel, 2024-12-12:
I was able to get mine updated to 1.3 or so. But it really was a hassle.
I wasn't able to get the intended updater to work, so I had to dig into the scripts, find that they generate a device ID which get send to a different url, to retrieve a module (some .so file). So basically downloading and including a binary blob?!?

Also I had to mangle some http headers to be able to download all required files. In the end I patched the bash script that runs on auto update to not try downloading stuff, since, you know, it's too broken to work. Then I just pointed the parts of the updater script to the pre downloaded files and it worked.

All you need to know to replicate the process of the update can be read in the updater file (some basic understanding of http required).

Hope this helps.
Also hope they just go to generate the required artifacts through github actions and have the releases be built here in the open. Then the updater could just check the releases link and download...

schlypel, 2024-12-12:
I would really love to see nanoKVM be easy to deploy and maybe even manage in quantities, but right now its an alpha experience.