sipeed/NanoKVM #170: HP KVM Switch no keyboard mouse

Reletiv, 2024-11-27:
Bought a nanokvm to control a HP kvm switch and make it "smart" But this doesn't work.
Screen does work using a expensive VGA to HDMI converter. But keyboard and mouse do simply not work, 
they do simply not work.... I have no clue how to fix this. I tried 2 different types of kvm switches but they still dont work..

markuman, 2024-12-06:
https://github.com/sipeed/NanoKVM/issues/135#issuecomment-2460483092  

> The cable that got delivered with the nanoKVM are not working. Once i've tried a USB-C cable that i got here the keyboard works just fine so your cable you deliver are bad...

Reletiv, 2024-12-06:
I have tried about 5 different cables, they all work when used on a different computer

DiogoConstantino, 2024-12-19:
Changing cable also worked for me.

Reletiv, 2025-02-17:
Still no keyboard mouse, screen works

yodatak, 2025-10-31:
Same for me too ...