sipeed/NanoKVM #174: Can't update from version 2.0.9

Filadeus, 2024-11-30:
I'm trying to use a WebUI to update my nanoKVM to the latest release, but get a `Get version failed` error when doing so.
[image]
nanoKVM is able to reach the Internet which I am able to confirm by pinning `cdn.sipeed.com`:
```
# ping cdn.sipeed.com
PING onlyv4.overseacdnweb.qiniu.com.w.kunlunsl.com (163.181.92.180) 56(84) bytes of data.
64 bytes from 163.181.92.180 (163.181.92.180): icmp_seq=1 ttl=55 time=24.8 ms
64 bytes from 163.181.92.180 (163.181.92.180): icmp_seq=2 ttl=55 time=25.9 ms
64 bytes from 163.181.92.180 (163.181.92.180): icmp_seq=3 ttl=55 time=26.6 ms
64 bytes from 163.181.92.180 (163.181.92.180): icmp_seq=4 ttl=55 time=25.2 ms
^C
--- onlyv4.overseacdnweb.qiniu.com.w.kunlunsl.com ping statistics ---
4 packets transmitted, 4 received, 0% packet loss, time 3005ms
rtt min/avg/max/mdev = 24.838/25.643/26.640/0.684 ms
```

doesntremindm3, 2024-12-01:
I have the same issue.  It appears to have started on all my nanoKVMs at the same time, maybe a week ago.  It's possible I'm on 2.0.9 as well.

doesntremindm3, 2024-12-01:
On a new nanoKVM that was not affected by this, I was on (I believe) 2.1.1 (certainly higher than 2.0.9) and updates seemed to be working properly.  I could see my version number using the webUI.  I then changed to a static IP and started seeing this error.  

I followed the steps at https://wiki.sipeed.com/hardware/en/kvm/NanoKVM/network/static_ip.html   I did set a gateway per these instructions.

Do you have a static IP on your nanoKVM(s) affected by this issue?

Filadeus, 2024-12-01:
Yes, I also had my nanoKVM configured to have a static IP. I tried to check for an update just now and it seems to be available, will try and run it now.
[image]

Filadeus, 2024-12-01:
Can confirm that it updated successfully, probably was just some temporary issue.
[image]

doesntremindm3, 2024-12-01:
Issue is ongoing for me, I just tested. Has been for weeks, probably a month+.

Any changes? Power cycles, changes to eth.nodhcp file, anything?

Filadeus, 2024-12-01:
Oh since I did not do any changes to the config of the device I just assumed that it was, perhaps, an issue on sipeed side. I have my nanoKVM static IP configured on the router via address reservation, so I don't have a eth.nodhcp file.

Filadeus, 2024-12-01:
Did you try updating it manually as described here ?
https://github.com/sipeed/NanoKVM/issues/42#issuecomment-2339549916

mcury1985, 2024-12-05:
Just as note, I performed some steps that corrected this, perhaps it works for you.

1- Followed the documentation at: [https://wiki.sipeed.com/hardware/en/kvm/NanoKVM/network/static_ip.html](https://wiki.sipeed.com/hardware/en/kvm/NanoKVM/network/static_ip.html) to set a static IP.

2 - Set DNS-  Created the file /etc/init.d/S99dns   (Change 192.168.30.1 for the DNS you want to use).
echo 'nameserver 192.168.30.1 # eth0' > /etc/resolv.conf
then run  chmod 755 /etc/init.d/S99dns

3- Edited the file /etc/profile (just add export TZ=America/Sao_Paulo in the same location as I did, but use your timezone instead.
And added my timezone:

cat /etc/profile
```
export PATH="/usr/bin:/usr/sbin"

if [ "$PS1" ]; then
        if [ "`id -u`" -eq 0 ]; then
                export PS1='# '
        else
                export PS1='$ '
        fi
fi

export EDITOR='/bin/vi'
export TZ=America/Sao_Paulo

# Source configuration files from /etc/profile.d
for i in /etc/profile.d/*.sh ; do
        if [ -r "$i" ]; then
                . $i
        fi
done
unset i
```

After doing these steps, everything just works now. Using 1.30 firmware with latest application version.

doesntremindm3, 2024-12-09:
This looks like fantastic information and I'll assume it's accurate, but with 10 of these in production (and 5 more in the mail) I'm not sure this is the fix for me. Also, I understood these words, but if something went wrong or it caused problems down the road, I'd be in trouble.

Seems it is related to wanting a static ip though? For something that would reasonably be expected to be given one? And we have a fix? Can this be put into a future version? I expect to buy more.

I don't know if there's a step after a forum post to request a feature or a bug report or whatever.

L00maca, 2024-12-19:
Had the same issue, in my case it was just the ntp not working, setting the date manually allowed the update to work (until reboot). Still haven't managed to get ntp to actually work tho.