sipeed/NanoKVM #177: NanoKVM Lite - External Power

marino-mrc, 2024-12-02:
Hello,
I cannot understand if I want provide 5V to the NanoKVM Lite through pins and not from the USB C port.

The problem is that I'm trying to manage a SBC and when I reboot the SBC, the NanoKVM Lite loses the power and I don't see the initial steps in the boot process.

Thank you.
Regards,
Marco

x-magic, 2024-12-03:
There are pads to the back of USB Type-C port which can be used to power up the board: 
[image]

Two of the pin headers are also on 5V rail, but those, alongside GPIO A14 and the 3V3 pins, are being occupied to fix the HDMI board in place, thus unusable.

tzahari, 2024-12-04:
@x-magic is there a risk that this will inject the power into the PC via the USB-C cable?

x-magic, 2024-12-04:
> is there a risk that this will inject the power into the PC via the USB-C cable?

It absolutely will 🤣

I personally won’t rely on NanoKVM Lite for POST stage as I found the way it handles HID causing 2/3 of my devices not to recognise keyboard on boot. So it doesn’t really matter if the board is powered externally or not.

marino-mrc, 2024-12-04:
so I don't think the proposed solution is valid @x-magic 
Is there any protection we can apply? 
What's the purpose of using NanoKVM lite if you cannot access the bootmenu of a Linux box and select the right Kernel or access the BIOS?

x-magic, 2024-12-04:
@marino-mrc just FYI - I'm not affiliated with Sipeed and just an end user like you. 

I'm saying that because I have gone through their documentation before purchasing and having a realistic expectation of pros and cons. NanoKVM full version (final version, not beta) has proper backfeed-protected aux power. 

According to the official FAQ, you can decouple the USB VCC and board VCC by removing the 0R on the 5V pad next to the microSD card slot, and power on the board separately from the USB host.

x-magic, 2024-12-05:
The official documentation is linked on this very repo's README file. 

I can't tell how 0R pad works exactly as I can't find this thing on the schematics, or maybe I'm just suck at reading schematics 🙃

If you want more information, it's worth it to at one of the repo maintainers.

silajim, 2024-12-17:
Are there any updates on this? I'd like to power my Lite from external power but I don't need the atx controls nor the screen

silajim, 2024-12-18:
I did look at the schematics, but without any pcb with traces I can't understand what is going on

adamz01h, 2025-01-05:
I designed a case around splitting out the USBC power and data lines using a small breakout board.

https://www.printables.com/model/1075051-nanokvm-lite-usb-power-splitter

I have also designed and manufactured a ATX power control PCB that works well with minimal investment.
No case at this point. 
https://github.com/adamz01h/nano_kvm_lite_atx_pwr

If people are interested in a bulk buy for cheap let me know.

madmurl0c, 2025-01-22:
How did you proceed @marino-mrc ? I'm thinking about connecting external power too but idk how. I didn't go for the full version since I don't need an oled screen but external power would be pretty neat

kolossboss, 2025-01-23:
Maybe a USB Y-cable could help, but I’m not sure if it’s a good idea to power it from a different source??

https://amzn.eu/d/5Q6ktY9

madmurl0c, 2025-01-23:
The Pro can be powered via a separate USB C cable. I thought the light could be powered via the GPIO pins or something.

I thought about just cutting open a USB A to C cable. Wire 5V and ground to a separate plug and put it in a phone charger. That should keep the data connection alive and electrically isolate the KVM from the Device it's attached to, right?

adamz01h, 2025-01-27:
guys, yall trying way too hard, this is solved. I made the part!

https://media.printables.com/media/prints/1075051/images/8131720_868112c7-1624-4095-a1be-629718c703f8_02218922-37b0-4797-b1ee-e3fed777b617/thumbs/inside/1280x960/jpg/photo_2024-11-15_23-02-17.webp

silajim, 2025-01-27:
> guys, yall trying way too hard, this is solved. I made the part!
> 
> https://media.printables.com/media/prints/1075051/images/8131720_868112c7-1624-4095-a1be-629718c703f8_02218922-37b0-4797-b1ee-e3fed777b617/thumbs/inside/1280x960/jpg/photo_2024-11-15_23-02-17.webp

Not every one has a 3D printer to make it, and if you are going to send it to a company to be printed then you are better off with a Y splitter for the power (if it ends up working)

adamz01h, 2025-01-27:
Thats not the point, its the wiring, as you can see, if you want to make your own cable you can. with common grns, one set for tx/rx/ground then one for power / ground. The usb breakouts just make it easy. I dont expect you to print the case.