sipeed/NanoKVM #179: Device is extremely unreliable. Almost always no video

Rainz3, 2024-12-03:
It seems I can get this device to work less than 50% of the time. I'm able to navigate and log in but I just get a black screen. I've tried h264 and mpjeg, multiple browsers, multiple computers, nothing works. It seems i have to constantly unplug it and replug in everything in hopes that it may work. Currently on version 2.1.1. I have tried with a dedicated 5v power and just power from the computer. I've tried it connected to multiple computers. Is there anything else I can try?

Snazzie, 2024-12-04:
firmware version != UI Version. have you tried flashing your nanokvm with new firmware.

tzahari, 2024-12-04:
@Snazzie I have the latest firmware 1.3.0 & UI version 2.1.1

marshalleq, 2024-12-05:
Don't know if this is latest firmware, but yet again I have no video, perhaps no nothing.  This is extremely disappointing.  The very purpose of what this device is for, simply does not work.  I am beyond dissappoointed given the recent fix to use h264 is not working, though this may be the latest update.  I am off overseas in less than 1 week so I assume this device will be a waste of money and time.  Guys, I get this is beta / open source or whatever, I get that things are not perfect,  but you gotta prioritise the basics for the use case.  I can't buy your product and then worry about if the basics are going to work.  Do your testing better.  PLEASE.  We will love you forever if you get this one part right so that we don't go backwards.  Thankyou.

Snazzie, 2024-12-05:
mine has been working quite decently fortunately and have another on its way.

have you ruled out bad hdmi cable?

bimusiek, 2024-12-06:
Same here. I upgraded to newest image, did the auto update script, rebooted and no video. It was there for a moment though. I am cancelling additional 3 orders of KVMFull only because it DOES NOT WORK.

How the hell can you ship something so unreliable?

Rainz3, 2024-12-08:
Just to update, I've tried reflashing and got the same thing. Tried yet another computer and same results. Took replugging in everything a dozen times before I finally got video. It was still virtually unusable as it would take a full minute for a single button press to go through. I checked today and there was an update 2.1.2 and so far its become usable again. Still laggy but it is actually usable. No clue if I put it back to the actual computer I want to control if it'll stay working.

marshalleq, 2024-12-08:
I ended up refreshing (and put to a new SD Card because the brand Kioxia gives no inspiration and the kernel logs of my host were filled with messages about it unmounting.  It now works again, I won't be updating it again unless I'm right beside it.  It's not trivial even when you are.

lemonderon, 2024-12-17:
@Rainz3 Did your device shows a blinking IP-address periodically? I mean the video only works if there is a static display output regarding the IP-address, when it blinks sometimes - it does not have a video output for me. #208 and #26. I think it's a grounding issue.

danbriant, 2025-01-02:
I'm also having the same issue
App version 2.1.3

SSH Console output

[2025-01-02 09:06:59] [error] [cgo.go:93] failed to read kvm image: -1

Snazzie, 2025-01-02:
> I'm also having the same issue App version 2.1.3
> 
> SSH Console output
> 
> [2025-01-02 09:06:59] [error] [cgo.go:93] failed to read kvm image: -1

have you tried reflashing with firmware 1.3.0?

danbriant, 2025-01-02:
> > I'm also having the same issue App version 2.1.3
> > SSH Console output
> > [2025-01-02 09:06:59] [error] [cgo.go:93] failed to read kvm image: -1
> 
> have you tried reflashing with firmware 1.3.0? maybe image is corrupt.

Thats what I'm thinking. Strange thing is this was a brand new device.

I will be taking it apart removing the card and re-flashing everything. Just waiting on my SD adaptor to arrive tomorrow.

kuhnla, 2025-01-09:
Hi,

call me an idiot. I bought 5 NanoKVM for my home-servers and none of them display anything. They are booting and they obtain an IP-Address via DHCP and I can login using ssh. Even the Web-Interface is working. But....

...no console output. The screen remains black in MPEG and in H.264 mode. I tried everey resolution and had a look in the debug-output of the HDMI-grabber.

They are completly unusable. Who wants them? ;-)

What kind of business is this? Sending devices to the customers which are unable to provide the very basic functionality?

patschi, 2025-01-09:
Have you tried re-flashing, different HDMI cable, check if another monitor has some output, a different browser or machine? Are there any errors in developer tools? As it probably works for most and you have 5 KVMs, it could be a more specific than a general issue.

kuhnla, 2025-01-09:
The NanoKVMs are booting and I see no reason to assume an error in the image of the SD-Card. Why should I reflash then?
I tried several HDMI-Cables ( I have a box full of them) and tested with different systems. All of my servers are Linux based and I would love to see a console. But the screen on Web-UI remains black. 

Let's see. Tomorrow I will open one of them and reflash the SD-card. But I am not very optimistic though.

bimusiek, 2025-01-09:
Ya, this product is a joke. We have just returned 4of those and got refund from Aliexpress for all of them. Then went to JetKVM and ordered those. Let’s hope they do good for their customers.

lemonderon, 2025-01-09:
Shipping this one back from where I live would cost the same as buying a new one...

Also waiting for my JetKVM right now. Hope it would be more competently executed.

marshalleq, 2025-01-09:
> The NanoKVMs are booting and I see no reason to assume an error in the image of the SD-Card. Why should I reflash then?
>
> I tried several HDMI-Cables ( I have a box full of them) and tested with [...]

The reason to reflate them is they come with old firmware which has known display issues. And unfortunately it's not easy to figure out how to uodate from that old version, but there is a process others have documented on here. After that they're quite good.

AJolly, 2025-01-10:
I've got a new one and not having any luck with video, I flashed it with the latest firmware.

marshalleq, 2025-01-10:
> @marshalleq How do we check for the current firmware version? There is separate UI version that can be seen in the device GUI by following the provided IP Address.

I actually thought that was the same thing. I'm not by a computer right now but I have just been using what's in the gui.

Snazzie, 2025-01-12:
Could their be hdmi version mismatch between your device and nanokvm? Couldnt find that info on their site. 2/2 of my nanokvm works fine in terms of video.

Snazzie, 2025-01-12:
> @marshalleq How do we check for the current firmware version? There is separate UI version that can be seen in the device GUI by following the provided IP Address.

image version via about nanokvm. 
[image]

kuhnla, 2025-01-12:
Thank you all for your thoughts and your your suggestions.

Meanwhile I reflashed the SD-Cards with the provided image 
20241120_NanoKVM_Rev1_3_0.img.

With this image the HDMI-Output of my servers are showed up in the 
Web-GUI of the NanoKVMs. Thank you for motivating me to do this.

Nevertheless the system setup of the NanoKVMs need some adjustments to 
work properly in a private LAN behind a firewall.

1. I changed the hostname and created /boot/hostname

2. I deleted /boot/hostname.prefix

3. I fiddled around with the DNS-Settings in /etc/resolv.conf. If I 
change this file or if I change /boot/resolv.conf the system always 
overwrites my changes at the next reboot. Then I used the steamroller 
method to overcome this behaviour and created a init script which copies 
/etc/resolv.conf.dhcp to /etc/resolv.conf if there is one.

Who the hell at sipeed had the idea to nail down this weird nameserver 
setup in resolv.conf and make it such sticky? My firewall needs 
recreational holiday after this ;-)

My currents quick and dirty setup runs and the NanoKVMs use DHCP for 
setup. If they proof reliable, I will setup them for using fixed 
IP-Addresses. It's always a bad idea to use DHCP at the device which 
controls the DHCP-Server.

So far the news from me

Markus

AJolly, 2025-01-13:
[image]
But I still haven't had any luck I've tried it with the standard desktop computer, as well as with an odroid N2+.
 I check that the HDMI cable works correctly and I've even tested with an HDMI splitter so I could watch the monitor and see that video output was happening.

AJolly, 2025-01-23:
Yeah, theyd ahve to respond to emails...

papou-github, 2025-03-15:
Hi,
My 2 cents. I had black screen with Image Version V1.4.0 and application Version 2.2.2. 
cat /proc/cvitek/vi_dbg was returning VIDevFPS = 0. I tried many HDMI cable but no way to have any display (X264 or MPEG)
I bought a 30cm HDMI cable and now it work like a charm !
So you may should try.

luisxkimo, 2025-03-27:
Same here. Lot of issues related with video in the web UI:

- Change HDMI cable not works. Sometimes works in same device and cable, sometimes no
- Update to new versions not works. It seems that more stable is 2.2.0 version

I'm trying to figure out which could be the issue, but after some tests changing between connected device and cables, I cannot give a pattern where fails recurrently

Snazzie, 2025-04-04:
Fyi, it appears video feed can go to sleep. pressing space would awake the video feed.

Also by any chance are you guys using firefox? firefox video feed doesnt work, but chrome and edge does

0ut1awStar, 2025-07-07:
I also face occasional no video issues. Removing and re-plugging the hdmi cable will get the video feed to show, but input controls still do not work, which I believe is the cause of the no video issue in the first place. Rebooting the nanoKVM always resolves the issue, so it really seems to be software related.

[image]